keyboardio/Kaleidoscope #603: OneShot is broken

algernon, 2019-03-15:
OneShots have been broken for a while, possibly since the last refactor. They require one more tap to do their thing: two taps to activate, three to become sticky.

gedankenexperimenter, 2019-03-15:
This seems like either uninitialized memory, or accessing the wrong memory somewhere, because changes to other (unused) parts of the code lead to different results for me. When I change the definition of `OneShot::inject()` I get what seems like normal behaviour, but when I remove it entirely (nothing in the standard sketch calls it), things don't work properly.

algernon, 2019-03-15:
Hrm.... if I explicitly `memset(state_, 0, sizeof(state_))`, that does not help (as expected, it's zero'd by default due to being static - had to make sure though). So it's either wrong access, or screwed up logic somewhere.

I'll go and bisect, too.

gedankenexperimenter, 2019-03-15:
Yeah, I tried that, too. I'm trying another approach now. I'll let you know if I find anything.

algernon, 2019-03-15:
Hrm, bisect is yielding obviously wrong results. Another interesting find: the *first* OneShot requires two taps to activate. Every other within the timeout requires only one. So it appears the issue might be related to timeouts.

algernon, 2019-03-15:
The more I look, the more sure I am that this is due to either a wrong timeout check, or wrong handling of active state.

As an aside, `ActiveModColor` stopped working properly too, not sure whether due to OneShot changes or not.

gedankenexperimenter, 2019-03-15:
I was also guessing timeouts. But I haven't come up with anything helpful yet. Except that this:
```c++
void OneShot::inject(Key mapped_key, uint8_t key_state) {
  handleKeyswitchEvent(mapped_key, UNKNOWN_KEYSWITCH_LOCATION, key_state | INJECTED);
}
```
…totally fixes the problem for me, consistently. Without ever getting called. Commenting that function out entirely, it out compiles and runs, but with the erroneous first-tap-doesn't work problem.

gedankenexperimenter, 2019-03-15:
I still have no explanation why the above change seemed to help, but I think I've found the problem. In `afterEachCycle()`, if `hasTimedOut()` returns `true`, it calls `cancel`, which sets `should_cancel_`. `hasTimedOut()` returns `true` regardless of whether or not there are any active OneShot keys, if enough time has elapsed since `start_time_` was set, so it's `true` in almost every cycle. Changing it like this seems to fix the problem reliably:
```c++
  static bool hasTimedOut() {
    bool active{false};
    for (byte i{0}; i < ONESHOT_KEY_COUNT; ++i) {
      if (state_[i].active)
        active = true;
    }
    if (active)
      return Kaleidoscope.millisAtCycleStart() - start_time_ >= time_out;
    else
      return false;
  }
```
I'll open a PR when I get a chance, but I've got a bunch of debugging cleanup to do before I can, so feel free to take my idea and just fix it (if I'm right and that does fix it).

PR #605 (2019-03-16): Fix OneShot premature timeout
OneShot keys were failing to activate on the first press because of how the timeout was implemented. Most of the time, at the end of a cycle, `should_cancel_` was being set to `true` because the current time was being compared to the last time a OneShot key was activated, regardless of whether or not a OneShot key was active. As a result, OneShot keys that were pressed for the first time in a while would fail to register as OneShots.

This change prevents checking the timeout unless there is an active OneShot key.

Fixes #603.
commit 445d50a9 (2019-03-16): Fix OneShot premature timeout

OneShot keys were failing to activate on the first press because of how the timeout was
implemented. Most of the time, at the end of a cycle, `should_cancel_` was being set to
`true` because the current time was being compared to the last time a OneShot key was
activated, regardless of whether or not a OneShot key was active. As a result, OneShot
keys that were pressed for the first time in a while would fail to register as OneShots.

This change prevents checking the timeout unless there is an active OneShot key.

Fixes #603.
commit 9c973793 (2019-03-19): Fix OneShot premature timeout

OneShot keys were failing to activate on the first press because of how the timeout was
implemented. Most of the time, at the end of a cycle, `should_cancel_` was being set to
`true` because the current time was being compared to the last time a OneShot key was
activated, regardless of whether or not a OneShot key was active. As a result, OneShot
keys that were pressed for the first time in a while would fail to register as OneShots.

This change prevents checking the timeout unless there is an active OneShot key.

Fixes #603.
closed 2019-03-16 by algernon