keyboardio/Kaleidoscope #604: Keymasks, a general purpose feature for Kaleidoscope?

noseglasses, 2019-03-15:
This is not an actual issue, nothing is broken. It is rather a feature offer to the firmware core.

Recently I developed some code to conveniently generate, read and write keymask bitfields. They are defined as follows.
```cpp
KEYMASK_STACKED_PROGMEM(
   my_keymask, 
   0, 0, 0, 0, 0, 0, 0,
   0, 1, 1, 1, 1, 1, 0,
   0, 1, 1, 1, 1, 1,
   0, 0, 0, 0, 1, 1, 1,
            0, 0, 0, 0,
                     0,
                     
   0, 0, 0, 0, 0, 0, 0, 
   0, 1, 1, 1, 1, 1, 0, 
      1, 1, 1, 1, 1, 0, 
   1, 1, 1, 0, 0, 0, 0,
   0, 0, 0, 0,
   0
);
```
The entire data structure is set up at compile time and stored as a byte array in PROGMEM.

Currently keymaps can be read from PROGMEM like so
```cpp
bool key_is_masked = isBitSetPROGMEM(my_keymask, row*COLS + col);
```
Every keymask occupies a minimum amount of storage in PROGMEM, for the Model01 this means 8 bytes for 64 keys. There's also a RAM version available.

I think this might be a useful feature for all sorts of other plugins. Under the hood the implementation relies on the `KEYMAP_STACKED_LIST` macro proposed in https://github.com/keyboardio/Kaleidoscope/pull/600. 

Keymasks are going to be part of the future [Kaleidoscope-XKeymaps](https://github.com/CapeLeidokos/Kaleidoscope-XKeymaps) plugin. But the code could easily be transferred to the firmware core to become available to other plugins, too.

gedankenexperimenter, 2019-03-15:
I came up with a functionally similar generic for Kaleidoglyph (which I called Bitfield, because it was useful for other things, too), and found several uses for it. One nice one was that I included an iterator for range-based `for` loops that would just return a value for each set bit in the bitfield. I think this is a very good and useful idea.

gedankenexperimenter, 2019-03-15:
If anyone is interested, here's my `Bitfield` template class:
https://github.com/gedankenlab/Kaleidoglyph-Core/blob/master/src/kaleidoglyph/Bitfield.h

noseglasses, 2019-03-28:
Bitfields have been submitted as https://github.com/keyboardio/Kaleidoscope/pull/600

Closing this.

closed 2019-03-28 by noseglasses