keyboardio/Kaleidoscope #606: Pressing the shift key should speed up t

ChristianKleineidam, 2019-03-15:


gedankenexperimenter, 2019-03-16:
I like the idea in general, but I would prefer to use a special-purpose mouse key for the purpose, rather than `shift`.

ChristianKleineidam, 2019-03-16:
Why not shift? It's a standard modifier key that can be pressed and release with the thumb without taking extra keyboard-space.

gedankenexperimenter, 2019-03-16:
Because it also sends `shift` to the host, and that can effect the how the overall input is interpreted by programs running on the host. I'm thinking of things like Inkscape, Sketchup, and various games, where one might want to make fine adjustments to mouse pointer position while holding `shift`.

obra, 2019-03-18:
I think that adding new mouse keys to speed and slow the mouse speed makes
sense. That'd give the user flexibility to map them onto their keyboard
where they want them.  (I feel like there are cases where I've needed to
shift-drag things in the past. I'd love not to break that functionality.)
ᐧ

ChristianKleineidam, 2019-03-18:
Okay, in that case it would likely make sense to have different optional keys with different speedup.