keyboardio/Kaleidoscope #613: Dynamic livetime for LEDModes

noseglasses, 2019-03-18:
This is not an issue but rather a proposal.

As LEDModes are used exclusively, why not let them have dynamic lifetime? This would safe a lot of RAM and would not be too complicated given the fact that all information about LEDModes (type, etc.) is available at compile time in the sketch. 

The only major change this change would require is to not have LEDModes being plugins but rather let plugins handle lifetime of led modes through abstract factories.

```cpp
// Use the abstract factory design pattern

class LEDModeFactory_ {
   public:
      virtual ~LEDModeFactory_() {}
      virtual LEDMode *generate(void *raw_buffer) const = 0;
};

template<typename LEDMode__>
class LEDModeFactory : public LEDModeFactory_
{
   public:
      virtual LEDMode *generate(void *raw_buffer) const override { 
         return new (raw_buffer) LEDMode__{}; 
      }
};

// The base class
//
class Plugin {
   ...
   public:

      static constexpr bool hasLEDMode = false; // By default plugins do not have LEDModes
};

// Derived plugin classes
//
class APlugin : public Plugin {
   ...
   public:

      // All the listed class content below could easily be wrapped up in a macro, say
      // SERVICE_LED_MODE(ALEDMode)

      // By providing a boolean flag, we simplify template separation of 
      // LEDMode and non-LEDMode plugins
      //
      static constexpr bool hasLEDMode = true;

      // This constexpr method enables access to pointers to led mode factories
      // of global plugin instances and let them be stored in PROGMEM
      //
      constexpr const LEDModeFactory_ *getLEDModeFactory() { return led_mode_factory_; }

   private: 
      LEDModeFactory<ALEDMode> led_mode_factory_;
};
```

I could think of an interface
```cpp
void activateLEDMode(uint8_t mode_id);
LEDMode *getCurrentLEDMode();
```

A possible implementation could be to determine the maximum storage requirement by checking all available LEDMode types at compile time. Then a common buffer would be statically allocated and placement-new could be used to generate new LEDModes. As those are polymorphic classes they would have a virtual destructor anyway.

Roughly the code would look as follows
```cpp
constexpr int max_led_mode_size = ...; // Some template and macro magic based on
                                       // the list of supplied plugins
uint8_t led_mode_buffer[max_led_mode_size];
LEDMode *cur_led_mode = nullptr;

// A container class for LEDModeFactory_ pointers
const PROGMEM LEDModeFactoryArray(... generated by macro magic based on the list of supplied plugins ...);

void activateLEDMode(uint8_t mode_id) {
   // If there is currently a LEDMode alive, we call its virtual destuctor and let
   // it do the cleanup ...
   if(cur_led_mode) {
      cur_led_mode->~LEDMode();
   }
   
   // ... and generate a new led mode
   cur_led_mode = getLEDModeFactory(mode_id)->generate(led_mode_buffer);
}
```

gedankenexperimenter, 2019-03-19:
I like the idea of having LED modes (or, more generally, any one-at-a-time objects) having a dynamic lifetime, but how would this work with something like `LEDSolidColor`, where the class has parameters in its constructor, used to set internal variables? Would those all need to be replaced with template classes, and all the constructor parameters changed to template parameters?

noseglasses, 2019-03-19:
At first glance it would be tempting to introduce a LEDSolidColor<uint8_t, uint8_t, uint8_t> template - but with no gain.
This is because every LEDSolidColor instantiation would require its own (polymorphic) factory class and its own matching LEDMode. A lot of PROGMEM for code and vtables would be wasted to move three bytes of RAM for each solid color to PROGMEM.

Instead I would implement it as follows and accept that tree bytes will remain permanently stored in  RAM (as it is the case right now).
```cpp
// In plugin.h

// Convenience macro for LEDMode plugin implementors
//
#define DEFINES_LED_MODE \
   static constexpr bool hasLEDMode = true; \
   \
   constexpr const LEDModeFactory_ *getLEDModeFactory() { return this; } \
   \
   virtual LEDMode *generate(void *raw_buffer) const override {  \
      return new (raw_buffer) LEDMode__{this};  \
   } \
   \
   friend class LEDMode__; // Allow access to private color values

// In LEDEffect-SolidColor.h

namespace kaleidoscope {
namespace plugin {

// This class' instance lives permanently
//
class LEDSolidColor : public Plugin, public LEDModeFactory_ 
                                     // Multiple inheritance with interfaces
                                     // is ok
{
   public:

      LEDSolidColor(uint8_t r, uint8_t g, uint8_t b)
         : r_(r), g_(g), b_(b)
      {}

      PROVIDES_LED_MODE

   private:

      // This class' instance has dynamic lifetime
      //
      class LEDMode__ : public LEDMode
      {
         protected:

            LEDMode__(const LEDSolidColor *parent) 
               : parent_(parent) {}

            void onActivate(void) final;
            void refreshAt(byte row, byte col) final;
               // Can access color values as parent_->r_, ...

         private:

            const LEDSolidColor *parent_;
      };

      uint8_t r_, g_, b_;
};

}
}
```

Nesting `LEDMode__` is not strictly required but enapsulates nicely.

noseglasses, 2019-03-19:
RAM footprint for LEDEffects (member data + vptr).

LEDChaseEffect: 9 bytes
LEDBreathEffect: 4 bytes
LEDRainbowEffect: 11 bytes
LEDRainbowWaveEffect: 11 bytes

For the stock firmware having dynamic LEDMode lifetime would save at least 24 bytes of RAM but could be even more whenever multiple LED effects would be used together, that all require a lot of RAM to store their member data.

I neglected LEDSolidColors in above statistics due to the restrictions discussed above.

noseglasses, 2019-03-19:
In the future, it might be possible that we want more than one type of exclusive-access service provided by plugins. Now its LEDModes, but it could be SoundModes, SensorModes, ... The more of them we might have, the more important it will be to save RAM - at least on the Atmega.

The approach discussed here could be generalized to any type of services and give them dynamic lifetime.

gedankenexperimenter, 2019-03-19:
Regarding `LEDSolidColor`, _et al_: I can't follow your code at all. How would one create two different solid color modes and activate/deactivate them?

noseglasses, 2019-03-19:
With respect to the sketch, nothing would change. The current stock firmware sketch would need no modifications at all. All plugins will still be registered using `KALEIDOSCOPE_INIT_PLUGINS`.

> How would one create two different solid color modes and activate/deactivate them?

Exactly the same way as it is done right now.

Let me try to explain roughly how it works:

With a similar technique like the one used for https://github.com/keyboardio/Kaleidoscope/pull/612 a PROGMEM array of `LEDModeFactory_*` would be generated at compile time. A manager class like `LEDControl` would call `void activateLEDMode(uint8_t mode_id)` to switch between LED modes. `activateLEDMode(...)` under the hood calls the `virtual LEDMode *LEDModeFactory_::generate(void *raw_buffer)` method of the led mode factories to create polymorphic LED modes in a shared raw buffer (placement new).

The separation into at least two classes, the plugin and the actual led mode is necessary so that we can have plugin instances (that possibly do not have data members) with permanent lifetime - the once registered with `KALEIDOSCOPE_INIT_PLUGINS` and led mode classes whose lifetime is handled by `activateLEDMode(...)`.

The method `constexpr const LEDModeFactory_ *getLEDModeFactory()` is there so that a plugin can be a LED mode factory (by inheriting form an interface) or have a led mode factory (composition) as a member. This is similar to what just happens to the hardware classes gaining bootloader members. It hides an implementation detail at no cost (`constexpr`) and provides flexibility for future changes.

The proposed changes would be possible without breaking 3rd party plugins by testing existing plugins when generating the global pointer arrays. An old-style plugin (non bundle-plugins) that is derived from `LEDMode` would be treated as proposed in https://github.com/keyboardio/Kaleidoscope/pull/612 and a deprecation message would be emitted. New style plugins would share a common buffer and be handled like explained above. The `activateLEDMode(...)` method would handle both old and new style LED-plugins.

Hope that helps.

noseglasses, 2019-03-26:
A pull request that implements the proposed changes has been submitted as https://github.com/keyboardio/Kaleidoscope/pull/617.

Closing this issue. Redirect discussions to the corresponding PR.

closed 2019-03-26 by noseglasses