keyboardio/Kaleidoscope #616: Allow (LED) plugins to choose their own refresh rate

algernon, 2019-03-22:
As suggested by @gedankenexperimenter in [#593](https://github.com/keyboardio/Kaleidoscope/pull/593#issuecomment-473530698), LED plugins should have a way to set their own refresh rates.

This issue serves as a place to discuss how this should be implemented, and how it should behave.

algernon, 2019-03-22:
The [basic idea](https://github.com/keyboardio/Kaleidoscope/pull/593#issuecomment-473551172) is to add a new function or two, which plugins can call from `onActivate()` to set their own refresh rates. This would let `LEDControl` reset the refresh rate to the default when switching modes, yet, still allow plugins to override that.

Obviously, this only available, and only makes sense for LED modes. Plugins that light up leds outside of the LEDMode framework will have to use whatever refresh rate is set otherwise, they can't, and should not be able to set their own. We can make the refresh rate changing method protected, and add `LEDMode` as a friend class, maybe? Saving that, we can rely on policy, rather than strict rules.

Now, one thing I'm unsure about is how should we treat the default? Is it a minimum, a maximum, or just a value that can be overridden in either direction? I could make a case for all of these.

algernon, 2019-03-22:
Paging @gedankenexperimenter, @noseglasses and @obra for opinions (no rush, this is nice to have only).

gedankenexperimenter, 2019-03-22:
I don't have enough expertise with friend classes and inheritance to have an opinion on the best way to restrict access to the frame-rate setting function to LED modes.

I do think there should be a maximum frame rate that can be specified by the hardware module, which LEDControl won't exceed, even if requested. And I'm inclined to also have a minimum frame rate, because other plugins expect the LEDs to be responsive, even if they're not LED modes themselves. The default frame rate could either be hardware-dependent or not (perhaps the latter as a backup, in case the hardware doesn't specify it), and should be bounded by the other two, but independent of them, and reasonable for most LED modes.

On thought I just had is that the desired frame rate could be specified as a constant in the LED mode, rather than using a function in LEDControl. I'm not completely certain that would work, though, unless all LED modes were required to define this variable.

algernon, 2019-03-22:
Having to define a constant would be a breaking change: we can't easily have a default in `LEDMode` that individual modes can override. We could adjust in the constructor, but then we're pretty much at the same amount of code as with an explicit call to a LEDControl function. And in this case, the explicit call sounds like the better approach to me.

Liking the bounds idea too, and providing a `syncDelay` default at the hardware plugin level is also something we want, I believe. Thanks!

noseglasses, 2019-03-22:
Using a constant in LEDMode derived classes to define a desired refresh rate is not possible as LEDControl accesses them through a polymorphic pointer. It can't see the LEDModes' actual types.

I would go with @gedankenexperimenters suggestion to let the LED modes influence the refresh rate via a dedicated method of LEDControl.

noseglasses, 2019-03-22:
Non constexpr constants in base classes are hardly ever a good idea. If necessary, the same could usually be achieved using access-methods that return a value. Those can be reimplemented in derived classes and thus do not unnecessarily use static RAM. All this, of course, only with non-polymorphic classes.

algernon, 2019-03-22:
> Using a constant in LEDMode derived classes to define a desired refresh rate is not possible as LEDControl accesses them through a polymorphic pointer. It can't see the LEDModes' actual types.

Right... my original thinking was that it could access it like how stuff accesses `KeyboardHardware.foo` instead of `foo`. But this setup doesn't work for this case, so my entire thinking was wrong. Sorry for the noise!

So yup, dedicated method, as originally planned.

gedankenexperimenter, 2019-03-22:
> Using a constant in LEDMode derived classes to define a desired refresh rate is not possible as LEDControl accesses them through a polymorphic pointer.

That's what I expected; I just wasn't sure. And even if the correct class could be determined, it would be equivalent to requiring another vtable entry for an accessor method, which would be contrary to the other goals we want to achieve with LED modes.