keyboardio/Kaleidoscope #630: Document how to write macros

algernon, 2019-04-23:
Writing macros is another one of those things many of our users want to do, and where our [existing documentation](https://github.com/keyboardio/Kaleidoscope/blob/master/doc/plugin/Macros.md) is more suited for the slightly more experienced reader. We'd like to have a step-by-step howto, which might not explain everything, but lets someone new to Arduino and programming in general, get started.

TG1999, 2019-05-09:
I want to work on this project in Google Season of Docs. Any help to start will be great.

algernon, 2019-05-28:
Hi @TG1999! Sorry for the incredibly late reply.

The best way to get started with bot this and #235, is to first set up a development environment (see the "Getting Started" block on our [wiki](https://github.com/keyboardio/Kaleidoscope/wiki)), to make sure that any examples can be tested to at least compile. Next, is to look at some of the existing macros in our [examples](https://github.com/keyboardio/Kaleidoscope/search?l=C%2B%2B&q=macroAction). These examples (the ones in the `examples/` directory, that is) show some of the things one can do with macros. For a good step-by-step document, I think the best would be to come up with a few examples that show various aspects of macros. Some of them should be super simple, using the `MACRODOWN` helper (see the existing docs, linked to in the root issue), some should be more complicated like the [AppSwitcher](https://github.com/keyboardio/Kaleidoscope/blob/ac434360c755f6d19a7234f903f62b24771d5b5a/examples/Features/AppSwitcher/AppSwitcher.ino) example.

As for installing third party plugins, I'll comment on #235.

TG1999, 2019-06-03:
Thanks @algernon fo the help.

TG1999, 2019-06-28:
Hi @algernon I have submitted my proposal with display name TG99.