keyboardio/Kaleidoscope #634: Mouse movement left & up trumps right & down

gedankenexperimenter, 2019-04-26:
When using mouse movement keys, if I hold down both `Key_mouseL` and `Key_mouseR` simultaneously, the mouse cursor moves to the left, rather than holding in place, as I would expect. This happens because of [these lines in MouseKeys.cpp](https://github.com/keyboardio/Kaleidoscope/blob/5acb97940b7be559122f39d0d2bac80b53149cb4/src/kaleidoscope/plugin/MouseKeys.cpp#L94-102). If the left-movement bit is set, the right-movement bit never gets checked.

obra, 2019-04-27:
That is as designed. I'd considered the case you describe and decided that
it wasn't rational user behavior, so having simpler code that did something
not entirely reasonable with a not entirely reasonable input seemed like a
fine compromise. Do you see a way to make the code simpler and smaller and
to account for the case you describe?
ᐧ

algernon, 2019-04-27:
We can try something like this:

```diff
--- a/src/kaleidoscope/plugin/MouseKeys.cpp
+++ b/src/kaleidoscope/plugin/MouseKeys.cpp
@@ -91,15 +91,19 @@ EventHandlerResult MouseKeys_::beforeReportingState() {
     accelEndTime = millis() + accelDelay;
   }

+  int y = 0;
   if (mouseMoveIntent & KEY_MOUSE_UP)
-    moveY = -speed;
-  else if (mouseMoveIntent & KEY_MOUSE_DOWN)
-    moveY = speed;
+    y += -speed;
+  if (mouseMoveIntent & KEY_MOUSE_DOWN)
+    y += speed;
+  moveY = y;

+  int x = 0;
   if (mouseMoveIntent & KEY_MOUSE_LEFT)
-    moveX = -speed;
-  else if (mouseMoveIntent & KEY_MOUSE_RIGHT)
-    moveX = speed;
+    x += -speed;
+  if (mouseMoveIntent & KEY_MOUSE_RIGHT)
+    x += speed;
+  moveX = x;

   MouseWrapper.move(moveX, moveY);
```

This increases progmem use by about 28 bytes, and makes the cursor stop when opposite directions are held. There's a bit of movement on press & release due to the delay with which one's pressing/releasing the other key, but the cursor is in a standstill for the most part.

algernon, 2019-04-27:
To further elaborate, about pressing both not being rational user behaviour: while gaming with mouse keys, I found that I often want to move the cursor to one direction for a long time, but then stop for a little, and either continue the same direction, or switch to the opposite.

With the current behaviour, I have to release the original direction, then either press and hold the same one, or the opposite direction. With the code change above, I can continue holding, press and hold the other direction, and release the one I don't want. I find this marginally more convenient. (Based on about 15 minutes of gameplay, I can report back after more experimentation if need be.)

gedankenexperimenter, 2019-04-27:
Without the temporary variables, there's no difference in PROGMEM:
```
modified   src/kaleidoscope/plugin/MouseKeys.cpp
@@ -92,14 +92,14 @@ EventHandlerResult MouseKeys_::beforeReportingState() {
   }
 
   if (mouseMoveIntent & KEY_MOUSE_UP)
-    moveY = -speed;
-  else if (mouseMoveIntent & KEY_MOUSE_DOWN)
-    moveY = speed;
+    moveY -= speed;
+  if (mouseMoveIntent & KEY_MOUSE_DOWN)
+    moveY += speed;
 
   if (mouseMoveIntent & KEY_MOUSE_LEFT)
-    moveX = -speed;
-  else if (mouseMoveIntent & KEY_MOUSE_RIGHT)
-    moveX = speed;
+    moveX -= speed;
+  if (mouseMoveIntent & KEY_MOUSE_RIGHT)
+    moveX += speed;
 
   MouseWrapper.move(moveX, moveY);
```
Also, there is a difference between releasing the movement keys entirely and re-pressing them _vs_ stopping the movement by holding opposite movement keys – when re-starting movement, if I hold both `Key_mouseL` & `Key_mouseR`, then release one, the mouse starts moving at full speed immediately, because `MouseWrapper.accelStep` hasn't been reset to zero.

commit 6805b349 (2019-07-07): Make opposing mouse movement keys cancel each other

Instead of having `Key_mouseL` & `Key_mouseUp` override `Key_mouseR` &
`Key_mouseDn`, respectively, stop mouse movement on a given axis when both keys
are held simultaneously. Doing so will not reset acceleration, so when one of
them is released, the mouse will resume movement at full speed immediately.

Closes #634.
closed 2019-07-07 by obra