keyboardio/Kaleidoscope #639: TapDance broken after updating

jamesnvc, 2019-05-02:
I'm currently on Kaleidoscope at fbe335e; after pulling to most recent, TapDance keys seem to not be working in a very confusing way. 

I have two tapdance keys, both using `tapDanceActionKeys`. One goes between PcApplication & NoKey and the other does Home & End. After updating, the PcApplication key *sometimes* sends menu, sometimes sends End and the Home/End key does some crazy combination of keys.

My full layout [here](https://bitbucket.org/jamesnvc/keyboardiolayout)

Potentially related to #423, as I am using OneShot

blkbsstt, 2019-05-04:
I'm also seeing a problematic interaction between TapDance and OneShot.

I tried to put together some minimal sketches that can demonstrate.

This sketch works (tap dance keys on palms behave as expected (cycling through '(', '[', '{' or ')', ']', '}')): 
https://gist.github.com/blkbsstt/d9142ae69664947a3223e9ee996a04ae

This sketch, adding in OneShot and a single OSM key, doesn't work:
https://gist.github.com/blkbsstt/fe734cac5a4dbcf3887cd8ed2a975f12

TD(0) prints '}' (very occasionally I could get a '(', but never really repeatably). TD(1) is now LeftControl, according to `xev`.

blkbsstt, 2019-05-09:
I'm not seeing this behavior if I pull a more recent revision of the Kaleidoscope submodule (5acb9794).

algernon, 2019-06-21:
@jamesnvc can you try with the latest Kaleidoscope, and see if the problem is reproducible? (I can't repro with TD + OneShot combination)

If it still persists, can you change the order of `TapDance` and `OneShot` in `KALEIDOSCOPE_INIT_PLUGINS`, and see if that helps?

jamesnvc, 2019-06-22:
Seems to be working now; thanks @algernon!

closed 2019-06-22 by jamesnvc