keyboardio/Kaleidoscope #644: Ideas for improvements to mouse movement keys

gedankenexperimenter, 2019-05-10:
I have played around with mouse pointer movement keys recently, and found them somewhat lacking. I have a few ideas for ways to make them more useful, so I'll enumerate what I see as their shortcomings here to start a discussion.

### Responsiveness

Mouse movement acceleration takes a long time to get going. If I want to move the cursor from one side of the screen to the other, it's painful to wait while it crawls slowly at first without accelerating. See #632. I think it would be good to have a shorter acceleration period before it reaches top speed. As it is now, it's always quicker to take my hands off the keyboard and reach for the mouse.

### Precision

Most of the time, when moving the mouse cursor, I want it to go some considerable distance, but to a small target location. This means moving it fast, then slowing down for precise positioning at the end, which is difficult to do with the current system — I need to get it moving, release the movement keys when I get near the target, and start again, often with significant overshoot.

To address both of these issues, I think might work to have some keys that control mouse cursor _speed_. Hold down `Key_MouseFast`, and the top speed is higher, for quickly moving the cursor a long distance, or `Key_MouseSlow` to limit it for precise positioning. A practical way to slow down the cursor without resetting to zero would be very nice. See also #606.

The one really nice thing the mouse movement keys provide is the ability to limit cursor movement to only horizontal or only vertical movement. With better speed control, maybe they can become mor generically useful, as well.

ibash, 2021-06-03:
Also take a look at kinetic acceleration in QMK (which was originally based on UHK): https://docs.qmk.fm/#/feature_mouse_keys?id=kinetic-mode

QMK/UHK support having multiple speeds, and I've heard kinetic feels really good (though I haven't used it). A port might be in order.

gedankenexperimenter, 2021-06-04:
I'll have to actually try out QMK to see how mouse keys work there; thanks.