keyboardio/Kaleidoscope #647: Macros: repeat like a normal key

wxcafe, 2019-05-17:
So I've defined a few macros, for example 

```
case MACRO_CLAP:
	if (keyToggledOn(keyState))
		Unicode.type(0x1F44F);
	break;
```
which I use to type 👏 with a keyboard shortcut. Anwyay, My problem is that when I keep pressing A for example it starts typing one 'a' then after a short stop, the keyboard sends more keypress events and more 'a's appear. On macros though, this doesn't happen: if I keep pressing the key, it'll only input once. This is because I use `if (keyToggledOn(keyState))`, of course, but I can't really replace `Unicode.type()` with a `MACRO()`, and even if I did I can only roughly emulate the correct behavior by going `return MACRO(I(25), W(500), T(W), T(H), T(A), T(T), T(E), T(V), T(E), T(R));,

I would like a way to have the keyboard shortcut trigger the macro, and then (like any other key), wait for however many milliseconds and repeat the macro, and on and on (with a different time for the first repeat). Is there a way to do that?

gedankenexperimenter, 2019-05-18:
There is a way to do it, but it's much more complex than letting the host repeat a held key. I would do it with a separate function with a couple of static variables — one `uint16_t` timestamp, and a `bool` for the initial delay. It would also have two `constexpr` timeout values — one for the initial delay, the other for the repeat interval.  This function would take the `keyState` value as its parameter, and would set state variables appropriately when toggled on, and check for timeouts when held.

gedankenexperimenter, 2019-05-18:
I haven't tested this, but it might work:
```c++
void repeatMacro(byte key_state) {
  constexpr uint16_t repeat_delay{500};
  constexpr uint8_t repeat_interval{20};
  static bool waiting_for_hold;
  static uint16_t start_time;

  if (keyToggledOn(key_state)) {
    waiting_for_hold = true;
    start_time = Kaleidoscope.millisAtCycleStart();
    Unicode.type(0x1F44F);
    return;
  }

  uin16_t current_time = Kaleidoscope.millisAtCycleStart();
  uint16_t elapsed_time = current_time - start_time;
  if (waiting_for_hold) {
    if (elapsed_time > repeat_delay) {
      waiting_for_hold = false;
    } else {
      return;
    }
  }
  if (elapsed_time > repeat_interval) {
    Unicode.type(0x1F44F);
    start_time += repeat_interval;
  }
}
```

wxcafe, 2019-05-20:
Oh thanks a lot for the code! It doesn't seem to work correctly (I'm not a C dev but I guess it's because of the `uint16_t current_time = Kaleidoscope.millisAtCycleStart();` being the same time as `start_time = Kaleidoscope.millisAtCycleStart();`? I don't know Kaleidoscope well enough to debug that tho), but I was more wondering whether there was a built-in way to do this... and there doesn't seem to be :/

gedankenexperimenter, 2019-05-20:
It wouldn't be much more code to turn this into a full-blown plugin, with a lookup table that translates `Key` values to Unicode values. I don't really have the time and motivation required to do it myself, though, since its utility is quite limited.

wxcafe, 2019-05-20:
Honestly I'm also interested in repeating other macros, not just unicode. This was an example (that I chose because it was simple), but I use other, more traditional macros that I want to repeat the same way!

gedankenexperimenter, 2019-05-20:
Repeating macros would be even simpler, using a function pointer.