keyboardio/Kaleidoscope #665: An alternative approach to structuring device hardware and modules

noseglasses, 2019-07-15:
## Preface

Considering https://github.com/keyboardio/Kaleidoscope/pull/622, https://github.com/keyboardio/Kaleidoscope/pull/61 and what would be needed to have a general virtual hardware that simulates all possible physical hardwares, I propose to go a slightly different way.

## Introduction

This is an alternative approach to modularizing Kaleidoscope's hardware classes and to gain an implementation that is well suited also for general virtual hardware.

Instead of [CRTP](https://en.wikipedia.org/wiki/Curiously_recurring_template_pattern), the proposed approach will somewhat but not entirely rely on [_policy based design_](https://en.wikipedia.org/wiki/Modern_C%2B%2B_Design#Policy-based_design). The basic idea is to compose and derive classes on the level of types instead of the implementation level. This makes things more flexible, the implementation clearer and can help us to separate properties that a virtual hardware needs to know from implementation details of the actual keyboards.

It was me (@noseglasses) who suggested the CRTP approach used by @algernon in one of the aforementioned PRs. After seeing all the complications that emerged with the definition of modules and hardware (traits classes undefined types, etc.), I came to the conclusion that CRTP, while a very useful approach in other places, here is not the optimal strategy. Apologies for potentially having led in the wrong direction.

My ideas presented here are based upon and strongly influenced by @algernon's work on https://github.com/keyboardio/Kaleidoscope/pull/611 and https://github.com/keyboardio/Kaleidoscope/pull/622. Both PRs will be a good potential sources for copy-pasting code snippets for the newly proposed approach. I assume that it would require only little work to combine both PRs to obtain the implementation suggested here.

## Benefits of the new approach

The troublesome traits classes that the CRTP-approach needs, will disappear.
The same holds for the awkward casting that the CRTP-parent template needs to
access the derived class' members and this-pointer (`This()`, `device()`).

Problems with unknown types (like e.g. the `KeyAddr` type) entirely disappear.

Due to inheritence, macros like `
KALEIDOSCOPE_HARDWARE_INVENTORY` will probably not be necessary.

The design is much more obvious and readable.

## Terminology

In the following I will use the term _device_ (or _system control device_) instead of _keyboard_, to emphasize that the keyboard (and the keyboard functionality) is only one aspect of hardware and features of a modern programmable keyboard (which is also a pointer device and many other things). 

When it comes to implementation and class names, I will stick with the term _hardware_ for the overall thing (the keyboard with all its functionality) for legacy reasons. But I would like to add that I find the term _device_ more appropriate and potentially there's an even better term. _Hardware_ is just a very general term.

But as the term `KeyboardHardware` was used in Kaleidoscope for specific classes for a long time, let's stick with it - I will just omit the `Keyboard`-part in some places.

## Modules

Currently, I see the following pieces (modules) that together define the _device_. Some of these may be entirely hardware, others a combination of hard- and software. Missing components (if i missed any) are easy to add if the design would be approved. Also, possible there will be more components in the future (see an example about how to add new modules at the end of this document).

Current modules are:

* keyboard (the part of the hardware that deals with the physical keys)
* bootloader
* storage (e.g. EEPROM)
* MCU

## Hardware description

A _device_ is composed of a set of C++ classes, each of which is the interface to of one of the components mentioned above. The description of a device (for compatibility reasons here called _hardware_) may e.g. look as follows

```cpp
struct AHardwareDescription
{
   typedef SomeKeyboard Keyboard;
   typedef FooBootloader Bootloader;
   typedef BarStorage Storage;
   typedef TheMCU MCU;
};
```

## Component Descriptions

The types `Keyboard`, `Bootloader`, `Storage` and `MCU` are certainly implementation
dependent and, thus, will need to be ignored by the virtual hardware. Nevertheless, the virtual hardware needs to know some aspects about the physical hardware that it is supposed to simulate. Because of that I suggest to add description classes for each of the components. Those classes provide descriptive information about the modules but no implementation details.

For the keyboard module/feature this could e.g. be a class like

```cpp
struct SomeKeyboardDescription
{
[... 174 lines cut ...]
       void engage() {
         thrusters_.start();
       }      
       
    private:
       ...
       // Newly added
       Thrusters thrusters_;
};

// SomeParentHardwareDescription is directly or indirectly derived
// from DefaultHardwareDescription.
//
struct EnterpriseHardwareDescription : public SomeParentHardwareDescription
}
   typedef WarpThrusters Thrusters;
   typedef WarpThrustersDescription ThrustersDescription;
};

typedef Hardware<EnterpriseHardwareDescription> EnterpriseHardware;
```

Any existing device that will not add thrusters will not be able to fly to unknown galaxies. But it will happily keep on doing its job as a keyboard. Due to the stub thrusters class in `DefaultHardwareDescription` any keyboard will satisfy the firmware's interface requirements, though. 

Existing non-thruster devices can add thrusters later on to be able to engage.

obra, 2019-07-15:
After a couple read-throughs, I find this design really appealing.

Theoretically, we'd also be able to use this pattern to define which HID Driver an implementation wants to use.

Unsurprisingly, I have some comments on naming ;)

> In the following I will use the term device (or system control device) instead of keyboard, to emphasize that the keyboard (and the keyboard functionality) is only one aspect of hardware and features of a modern programmable keyboard (which is also a pointer device and many other things).

"Device" makes a lot of sense. And makes a lot more sense than "KeyboardHardware" - I think I'd be ok transitioning to "Device" with a tiny compat wrapper..

> keyboard (the part of the hardware that deals with the physical keys)

I -think- this is actually the "KeyMatrix" or possibly "KeyScanner"

> Currently, I see the following pieces (modules) that together define the device. Some of these may be entirely hardware, others a combination of hard- and software. Missing components (if i missed [...]

All of those things appear to me to be "Drivers" or "Device Drivers"

We're also probably going to want to have "LED" Drivers. Though in the case of the Model 01, We scan keys AND update LEDs by calling through to the same backend device.

noseglasses, 2019-07-16:
I considered @obra's suggestions regarding naming and applied the following changes to the above document.

* the term _hardware_ in class names and most remaining parts of the text was replace by _device_.
* replaced the term _module_ by _driver_ and added a short explanatory sentence about the relation of modules (hardware) and drivers.
* renamed _keyboard_ (driver) with _key scanner_.

noseglasses, 2019-07-16:
@obra, thanks for your feedback. I am happy that you like the proposal. Your naming suggestions are great. 

Let's see what @algernon has to say about all this. I am sure that he has a much better overview and will be able to point out possible pitfalls of this design that I might have overlooked.

obra, 2019-07-16:
> As the legacy hardware classes provided key scanner related methods like `isKeyswitchPressed(...)`, it is important to let the new `Device` template feature the same methods. Because of that, we simply derive the `Device` template from the key scanner class. Thus, the device _is_ a key scanner. The device template also supplies accessors for all of its drivers.

This makes me a little bit uncomfortable. It feels like it would end up conceptually cleaner if all the different drivers worked the same way.  If I were (naively) doing this, I'd be writing accessors in the device class to proxy onward into the driver in question. (Current implementations -also- expose LED related methods.)

> struct Model01KeyScannerDescription
> {
>     static constexpr uint8_t matrix_rows = 4;
>     static constexpr uint8_t matrix_cols = 16;
> };
> 

That really is more of a key matrix description than a keyscanner description. And even then, it's the logical matrix, not the physical matrix. Hrm.

> // The device class, compatible with Kaleidoscope's legacy hardware
> // classes.
> //
> typedef Device<Model01HardwareDescription> Model01;
> 
> typedef Model01 KeyboardHardware; // legacy typedef used by the firmware
>

Since I don't see it anywhere in the example, Where would the 'Model01' be used? Or is it going to end up optimized out to:

typedef Device<Model01HardwareDescription>  KeyboardHardware

?

noseglasses, 2019-07-17:
> As the legacy hardware classes provided key scanner related methods like isKeyswitchPressed(...), it is important to let the new Device template feature the same methods. Because of that, we simply [...]

> This makes me a little bit uncomfortable. It feels like it would end up conceptually cleaner if all the different drivers worked the same way. If I were (naively) doing this, I'd be writing [...]

It would definitely be cleaner not to use inheritance that way. I just suggested it in the first place to keep the required changes to the remaining firmware at a minimum. With _accessors_ you mean methods like `isKeyswitchPressed(...)` that pass the respective calls on to equally named methods of the drivers, no? The latter would also work without changes to the rest of the firmware and would definitely be preferable. I will update the respective parts in the proposal.

> That really is more of a key matrix description than a keyscanner description.

My initial idea was to have description classes that are closely linked to the drivers. This also reflects naming. i just added `matrix_rows` and `matrix_columns` but there might be more information that describes the key scanner. Now as the entire firmware uses KeyAddr to iterate the key matrix, I am not entirely sure if it wouldn't be better to directly define the `KeyAddr` type instead of `matrix_rows`,... 

>  And even then, it's the logical matrix, not the physical matrix. Hrm.

Under which circumstances would the firmware want to know about the physical matrix (wiring)? From my point of view the physical layout only needs to be known to the key scanner and LED-related driver classes.

> Since I don't see it anywhere in the example, Where would the 'Model01' be used? Or is it going to end up optimized out to:

Right. I added `Model01` only as I was not 100% sure if the class is not used somewhere in a Model01-only plugin (like the test mode). I would also find it better to not expose the name of the keyboard to the firmware but use only the typedefed `KeyboardHardware` (deprecate?) or rather `KeyboardDevice` instead. Will update the proposal.

obra, 2019-07-26:
@algernon You mentioned that you had some thoughts on this. I'd love to be able for us t omove forward on it. Can you tell me a bit about what you were thinking?

obra, 2019-08-07:
I've talked to @algernon a bit about this. It sounds like he's generally pretty positive about it, but has some concerns around how to minimize boilerplate.

I'd really like to see something like this happen.

algernon, 2019-08-13:
I've spent a few days playing with this idea, and in general, I like it, a lot. It solves many of the problems of the current architecture, and makes the code much, much cleaner - at the cost of being far more verbose. I'm convinced that this is the way forward.

By verbose, I mean that someone implementing a new device, or driver, will have to write a lot of boilerplate. Part of the task can be made easier with macros, but it's still more verbose than I'd like. Mind you, it's not worse than the status quo, we require a fair amount of boilerplate now, too. But if we're going to redesign things, we may as well try to reduce boilerplate too. Since the amount of code to write isn't prohibitively more than now, even if we can't fix the verbosity, we should still go forward with this idea.

Another thing I'm a bit torn about is the compatibility code. That's a noticable amount of code to maintain. Granted, it will be optimized out by the compiler, so the extra code is only in the source. In my experiments, I wasn't able to fully localize the compat code, and had them scattered around a few files. Not a huge deal, just a minor annoyance. On one hand, we try to remain backward compatible, so compat code is necessary. We also try to present a simple API to the end-user, and sometimes they will want to use hardware-provided functions directly: the compat functions make that easier. On the other hand, not having them makes the classes simpler.

So, I have a proposal!

What if we came up with a way to extract out the description classes from other code? If we had a script that'd pull them out from a single class, and split it into description and implementation? We could then commit both the source and the extracted/generated files into git, to make things simpler. This would reduce clutter a bit, and would make it considerably easier to implement hardware classes. I do not yet know how the source in this case would look, mind you.

noseglasses, 2019-08-13:
@algernon, could you provide an example about where you expect that much boilerplate code. The proposed approach combines inheritance and composition, in order to keep the amount of boilerplate at a minimum, appart from other advantages.

Regarding your scripting idea, what type of scripting language do you imagine and how could it be integrated in the build system? What advantages would it provide compared to an approach where both parts of the code, description and implementation, would reside in the same file and could be activated through preprocessor macros?

algernon, 2019-08-13:
By boilerplate, I mostly mean the various classes and typedefs. It's not (significantly) more than what we have now, but I'm hoping we can improve upon that. Macros help a lot, too, indeed.

Regarding scripting: I was thinking perl, python or even shell/awk, depending on what we end up needing. The results would be committed to git, so we don't need to fully integrate it with the Arduino build system, only ours. Like, a `make hardware-plugins` target could do the trick, perhaps? The advantage of this over one part or the other being activated by preprocessor macros is that it feels like less of a hack. Having two things in the same file, and only one of them being active depending on a macro is something I find rather ugly.

noseglasses, 2019-08-14:
> The advantage of this over one part or the other being activated by preprocessor macros is that it feels like less of a hack. Having two things in the same file, and only one of them being active depending on a macro is something I find rather ugly.

I am not entirely convinced that what you propose is going to be less a hack. Mainly because I expect that we will have to add some sort of tags to the code to make it easy to parse by some scripting language. Those tags would at least be as ugly/verbose as `#ifdef` clauses.

If we decide to go that way, I would suggest to use some established markup language like  XML to do the tagging and make parsing and code generation as simple as possible.

Anyway, I do not feel too comfortable with such an approach and would rather stick with plain C++ and some `#ifdefs` and keep the build procedure simple. If regarded necessary we could add some comments that briefly explain why the preprocessor conditionals are there.

noseglasses, 2019-08-14:
Although the scripts are not intented to run during the normal build process, adding additional make targets which need to be invoked explicitly add additional complexity in terms of documentation and also might confuse developers of new hardware.

I find myself browsing the build files quite frequently to lookup the names of less common make targets (as I am typically too lazy to find out if and where they are documented).

noseglasses, 2019-08-14:
Moreover, we will either loose portability (shell/awk) or require developers to install additional software (perl/python). Unfortunately, Arduino seems not to ship with a java interpreter on all platforms. Otherwise that would have been a solution that works without additional software - and  FWIW - who likes Java anyway ;-)

algernon, 2019-08-14:
Experimenting a bit more, it turns out that the `#ifdef`-based method isn't all that bad as I first thought. I'm preparing a proposal PR using that, hopefully by tomorrow.

(While tempting, the proposal will not include #611 yet, that'll be handled there, after this one)

noseglasses, 2019-08-15:
Great news and thanks for working on this.

algernon, 2019-10-07:
I've made *some* progress on this, but got stuck a bit. To validate the design and its flexibility, I chose to port Atreus over to the proposed new API, along with ATMegaKeyboard. It's _almost_ there!

For now, I went with multiple inheritance and some small classes (instead of having embedded members and wrapper compat functions), because that was shorter to implement. We have an `ATMegaScanner` template class (this is 90% `ATMegaKeyboard`), a `NoLeds` driver, etc. The Atreus defines an `AtreusScannerDescription`, and aliases `ATMegaScanner<AtreusScannerDescription>` to `AtreusScanner`. So far, this is all nice and dandy, compiles just fine. But when linking, `AtreusScanner::keyState_` is undefined. I have no idea how to define these. I tried defining them in `Atreus.cpp`, but got all kinds of different errors. I pushed my current state to the [`hardware/new-api`](https://github.com/keyboardio/Kaleidoscope/tree/hardware/new-api) branch, if someone wants to give it a try.

I will continue exploring my possibilities too, but wanted to get a quick status update out regardless.

noseglasses, 2019-10-07:
While the code of template class instances is generated automatically by the compiler in every compilation unit where it is used, the same does not hold for the static member data.

The static members of the specific template instance of `ATMegaScanner` must be initialized somewhere, preferabley in `Atreus.cpp`. 

```cpp
template<> uint16_t kaleidoscope::driver::scanner::ATMegaScanner<AtreusScannerDescription>::previousKeyState_[] = {}
// Add similar initializations for any other static members of template class ATMegaScanner here
```

This should hopefully fix the linker errors.

algernon, 2019-10-07:
AHA! So I need the initializer there too, and it can't be in the header. That will complicate things a bit, but at least I can compile stuff now successfully.

Thanks!

algernon, 2019-10-07:
Well, things compile now, but the keyboard fails to enumerate. Progress nevertheless!

algernon, 2019-10-07:
...fixed!

Now to clean up the code, and port all other boards too. Thanks again @noseglasses for the quick help!

noseglasses, 2019-10-07:
You sure, you wanna stick with the multiple inheritance approach instead of proper composition?

algernon, 2019-10-07:
I will do proper composition, that's part of the cleanup.

On another note, this first attempt made the example Atreus firmware smaller: -316 PROGMEM and -2 RAM is the difference between master and the current `hardware/new-api` branch. This is suspiciously large, but the keyboard appears to function. Once done with the cleanup, if the difference remains, I will investigate what's causing it.

algernon, 2019-10-07:
Pushed the first batch of cleanups, it now uses composition everywhere (I think...), and compiles to pretty much the same thing. Keyboard still functions. I broke all other keyboards with this push though, because the `kaleidoscope::Hardware` base class was redone: it now provides the legacy wrappers, so that we don't have to implement them for every single board.

noseglasses, 2019-10-07:
The description classes will be especially important to define virtual hardware. Because of that I'd suggest to consistently define description classes for all components (scanner, LEDs, MCU, etc.) and let them be empty classes if necessary. By this means, we can add properties to the components' base description classes later on if necessary. 

Both device description classes as well as component description classes should consistently be defined through inheritance, thus inheriting the descriptions of components of the base classes. 

`ATMegaScannerDescription` e.g. should IMHO be derived from some `ScannerDescription` base class and only re-define `debounce`.

In the same way I would suggest to derive any new device description class either from a common base device description class that is the root class for all descriptions (which will likely contain empty typedefs to exemplify what components are required/supported) or a description class of a concrete hardware. Any typedefs in derived description classes will shadow the typedefs of the base class and will thus be used instead.

A keyboard that is e.g. similar to the Atreus but has LEDs could then be derived from Atreus by deriving a `LEDAtreusDescription` that only redefines the `LEDs` and `LEDsDescription` typedefs.

algernon, 2019-10-07:
A'ight! We now have `driver::scanner::NoOpScanner` (and `NoOpScannerDescription`), `driver::led::NoLeds`, `driver::mcu::NoMCU`. These are used by `kaleidoscope::HardwareDescription`, which is the base class for `AtreusDescription`. `Atreus` itself is a derivative of `Hardware<AtreusDescription>`.

The base `NoOpScannerDescription` only has `matrix_rows` and `matrix_cols`, because the PINs are not a universal thing, there are scanners that don't have those (eg, the other half of the ErgoDox, wherein an expander is used instead). The pins are added by `AtmegaScannerDescription`, and filled in by `AtreusScannerDescription`.

The `Atreus.h` file is still messier than I want it to be, but we're getting there.

algernon, 2019-11-12:
With #695 being merged, this is resolved. \o/

closed 2019-11-12 by algernon