keyboardio/Kaleidoscope #670: Active modifier LED highlight doesn't work all for the first ~minute after boot

wearpants, 2019-07-31:
Using OneShot mode, the active modifier LED plugin doesn't highlight any keys for the first minute or so after boot, even though the modifiers work properly.

Might be the first few dozen keypresses rather than time-based.

gedankenexperimenter, 2019-07-31:
My guess is that it's being overridden by the BootGreeting effect.

algernon, 2019-07-31:
I asked @wearpants on Discord, and he confirmed: if one changes layers, the highlight will work after that. The problem is caching. `ActiveModColor` will scan the keymap on setup and at layer change. For some reason, the scan on setup doesn't yield the results we want.

algernon, 2019-08-16:
I _think_ I have a fix for this, a simple one too. Unfortunately, while trying to verify if it really works, I discovered that ActiveModColor doesn't seem to restore the original color. Not sure whether that's an ActiveModColor or a Colormap issue yet, though.

PR #674 (2019-08-16): ActiveModColor: Cache the list of modifiers at setup time too
In order for the plugin to work without having to switch layers once, it needs to scan the keymap for modifiers at setup time too. We do this by calling `onLayerChange()`, which already does that.

Fixes #670.
commit 9dfbe78e (2019-08-16): ActiveModColor: Cache the list of modifiers at setup time too

In order for the plugin to work without having to switch layers once, it needs
to scan the keymap for modifiers at setup time too. We do this by calling
`onLayerChange()`, which already does that.

Fixes #670.
closed 2019-08-16 by obra