keyboardio/Kaleidoscope #673: OneShot keys timeouts preventing other OneShot keys from working

Nevon5, 2019-08-16:
A short while after the first one shot key is double tapped neither tapping or double tapping other OneShot keys seems to have any effect, even though LED-ActiveModColor lits the key. So for example if I double tap the “function” key and then the Ctrl key and then wait a couple of seconds and then double tap the Shift key, all three keys are lit up but using the arrow keys behaves like the Shift key is not pressed.

https://community.keyboard.io/t/combining-multiple-oneshot-keys-not-working-as-expected/3120

gedankenexperimenter, 2021-05-26:
This is probably fixed by #1024. I'll see if I can reproduce it and write a testcase for it when I get time.

gedankenexperimenter, 2021-05-26:
I've got a testcase that reproduces this problem, and demonstrates that #1024 fixes it.

closed 2021-05-28 by algernon