keyboardio/Kaleidoscope #680: Macro that worked fine in April of 2018 now repeats like mad

fitz, 2019-09-02:
Got a new Model 01 this week, flashed my own layout onto it and there's a weird behavior with the macro for my star and pipe key -- it now prints out a gazillion pipes or stars
pressing it once unshifted and once shifted results in:
***************************************************|||||||||||||||||||||||||||||||||||||||||||||||||||||||

When it used to (as of April 2018) print

*|

The function is called starPipeMacro and you can find the entire sketch here: 

https://github.com/fitz/Keyboardio-Maltron-Layout/blob/master/Model01-Firmware.ino

gedankenexperimenter, 2019-09-02:
@algernon — I feel like this must be related to the scan order bug in Kaleidoscope-Macros we fixed about a year ago, but I'm currently at a loss as to why this is happening. I haven't had time to try to reproduce and analyze it yet, but I'll give it a try later this week if you haven't figured it out by then. My current guess is that it's caused by an interaction between plugins, but that guess is very low-confidence.

Maybe I'll even see if it's now possible to port Kaleidoglyph-Unshifter to Kaleidoscope, since that plugin handles the general case that this particular macro is meant for.

gedankenexperimenter, 2019-09-02:
Once I took a few minutes to look at this on a desktop instead of my phone, the answer became clear. @fitz – your `starPipeMacro()` function is based on an older version of `anyKeyMacro()`, and that's the source of the problem. `kaleidoscope::hid::pressKey()` now takes a second argument, which is a boolean that should be `true` if the key state is toggled on, and `false` if it is being held. Here's a version that works:

```c++
static void starPipeMacro(uint8_t keyState) {
  static Key lastKey;
  bool toggledOn = false;
  bool isShifted = kaleidoscope::hid::wasModifierKeyActive(Key_LeftShift)
      || kaleidoscope::hid::wasModifierKeyActive(Key_RightShift);

  if (keyToggledOn(keyState)) {
    toggledOn = true;
    if (isShifted) {
      lastKey.keyCode = Key_Backslash.keyCode;
    } else {
      lastKey.keyCode = Key_8.keyCode;
      lastKey.flags = lastKey.flags | SHIFT_HELD;
    }
  }

  if (keyIsPressed(keyState))
    kaleidoscope::hid::pressKey(lastKey, toggledOn);
}
```

Note that you don't need to make `lastKey` a `static` variable in this function. It was only `static` for `anyKeyMacro()` so that the value would persist from one cycle to the next.

fitz, 2019-09-02:
@gedankenexperimenter Perfect! I can *| all I want now. Thank you so much!

(Closing)

closed 2019-09-02 by fitz