keyboardio/Kaleidoscope #682: Erroneous key repetition with Qukeys

neh, 2019-09-09:
I'm having a problem with Qukeys where the last key _before_ I press a qukey is repeated anywhere from 2-8 times (roughly, here's an uncorrected example just frommmm writingg this issue). The specific situation is that I'm trying out a small 36 key layout, and I have a number/symbol layer that I access via a qukey on the space key. I've tried setting the overlap threshold to many values from 0 to 100, and it doesn't seem to make a big difference. Is there any way to mitigate this? Anything I can test? Something I just don't understand about qukeys?

gedankenexperimenter, 2019-09-10:
Can you share your sketch? I'll take a look as soon as I can, but there might be something specific about your sketch that matters.

neh, 2019-09-10:
I knew I'd forget to include something useful... here it is: https://github.com/neh/Model01-Firmware/blob/master/Model01-Firmware.ino

gedankenexperimenter, 2019-09-10:
Thanks. I'll give it a try tonight, and hopefully find a solution by sometime tomorrow.

gedankenexperimenter, 2019-09-10:
Well, I've tested it out and so far I can't reproduce the problem. The fact the you get up to eight repetitions of a key suggests that the Qukeys queue is involved, since there are up to eight entries therein. However, that might just be coincidence, because that queue stores press and release events separately. I'll move on to code inspection, but I have a couple of questions that might help:

When do the extra characters appear? Do they show up when you _press_ the qukey, or when you _release_ it? Do they ever appear when you _hold_ the qukey? Do they only appear on the layer-shift qukey, or also on ones that are modifiers when held?

Last, if I can't reproduce it, and I come up empty on code inspection, I can try sending you a debug version that logs to the serial port, if you're willing.

neh, 2019-09-10:
I'm not sure whether 8 is an actual limit, I haven't been keeping a close count; it's usually a fairly small number of repetitions though.

The repetitions are happening when I release the qukey. I just noticed that it also happens when the qukey does a layer shift (I was typing `e/` and got `eeee/`, and / is on the symbol layer accessed by the space key). They have appeared, occasionally, on a held qukey, and those times I can get more than 8 repetitions. I've used ctrl/esc and tab/gui qukeys for a long time and never had any issues with them, but I did have similar issues when testing home-row modifier qukeys that I assumed at the time were due to my sloppier typing when I speed up, and I just gave up on them intending to dig into it again later. I just managed to reproduce those a few times to confirm that, so both layer-shift and modifier qukeys.

I'm happy to help debugging any way I can... I'd love for this to work smoothly. Thanks for your efforts, I really appreciate it.

gedankenexperimenter, 2019-09-10:
> They have appeared, occasionally, on a held qukey, and those times I can get more than 8 repetitions.

That's significant. If you can definitely confirm this, it rules out a whole area of potential problems.

One more thing, which may be tricky to be sure about — Does it only happen when you roll over from some other key to the qukey? i.e. Does it happen sometimes if the qukey isn't pressed until after the previous key is released?

neh, 2019-09-10:
I'll try to confirm the held qukey situation, but it's a tough one. It's only happened twice, and I just remember that it was kind of unusual when it happened, not typical typing.

I'm pretty certain that it only occurs when rolling over from another key (the key that then repeats), as this only really happens when I'm typing faster and I can never trigger it when I'm deliberately being more precise and/or slowing down.

neh, 2019-09-10:
I've been able to reproduce the held qukey situation pretty reliably when typing IP addresses. If I roll from the `.` (which is just a normal key) to the space qukey to get to the numbers, I'll end up with things like `...4`. If I precisely press and release the `.` first, then no repetition occurs. Then, while typing this response, I noticed that I can get it to happen quite simply just by pressing a normal key, rolling to the space qukey, and holding the qukey down. Happens every time.

gedankenexperimenter, 2019-09-11:
Just to be sure, you are releasing the `.` key before it's usual repeat, correct?

neh, 2019-09-11:
I am, it's a very quick `.` and then hold the space qukey and I get a few extra `.`s.

gedankenexperimenter, 2019-09-11:
I hate to say it, but I'm completely stumped. Try as I might, I have been unable to reproduce this bug, and I'm out of ideas from the source code.

I can suggest a couple of things still to try. First, power cycle the keyboard and see if it happens immediately when it's turned on (before using any other features, I mean). Maybe there's some precondition that triggers it.

Second, try a brand new, clean build, from a new checkout from GitHub, guaranteeing that the only code being used (other than your own sketch) is from the current Kaleidoscope master branch. I'm grasping at straws, but maybe there's some cruft in your build tree — I've seen this happen before.

If none of that helps, I'll put together a debug version that logs to the serial port to try to get more internal state information. I'm still thinking about what I want to capture with that (this is the first time I've been completely unable to reproduce a Kaleidoscope bug on my own keyboard; I usually don't have to worry about a long lag time between trials, so I want to get lots of information the first time).

neh, 2019-09-11:
I just tried after power cycling and can get it to happen immediately, on the first keypresses. I'll try a clean build soon and see how that goes.

gedankenexperimenter, 2019-09-11:
> I am, it's a very quick `.` and then hold the space qukey and I get a few extra `.`s.

Do those extra characters appear simultaneously, on release of <kbd>space</kbd>, or do they repeat one at a time while you hold the qukey?

neh, 2019-09-11:
Neither, actually... they appear all at once after maybe half a second of holding the qukey, before releasing it. So the initial `.` appears, and about half a second later (the amount of time feels consistent) a few more appear, then nothing else happens, whether I continue holding or release the qukey.

noseglasses, 2019-09-13:
@gedankenexperimenter, i currently can't spend too much time on this but if you provide me with detailed instructions what to test, I'd be happy to help.

neh, 2019-10-28:
I've spent more time with this over the last few weeks, and can't eliminate the problem. I think it's forced me to type a bit more precisely, which is fine, but the problem still occurs. As a point of comparison, I just finished building a new keyboard running qmk, and I replicated my layout there and do not have any issue with key repetition.

gedankenexperimenter, 2019-10-31:
I still haven't reproduced this bug a single time (using the sketch you supplied), nor heard of anyone else seeing this problem with key repeats on rollover. Comparisons to QMK's Mod-tap keys are unfortunately not relevant; it uses a completely different algorithm.

My best guess right now is still just a shot in the dark — that there's something wrong with your build environment, and your compiled firmware is not the same as mine when I build your sketch.

gedankenexperimenter, 2020-10-07:
I caught a typo I made in the original version of my fix for this one (#684). Maybe with that PR, this bug is fixed?

PR #684 (2020-11-11): Prevent rollover to a qukey from causing unintended repeats
When rollover occurs from a non-modifier key to a qukey, if we delay the release event of that key until after the qukey's state is resolved, and if the hold timeout is set to a fairly large value (on the order of 500ms), unintended repeats would occur for a key that was actually only tapped. To prevent this, if there's only one event in the queue (the press of the qukey), and we see a release of a non-modifier key that's not the qukey, it's okay to allow that release event to skip the queue and simply proceed as if it had been releas [... cut]
closed 2020-11-11 by obra