keyboardio/Kaleidoscope #686: Cannot flash/find serial port from within WSL

benze, 2019-09-13:
I'm trying to build and flash from within WSL.  I have installed the arduino tools and am able to compile/make the firmware successfully.  However, when it comes to flashing, I get an error message that it cannot detect the keyboard automatically.

```
Building ./Model01-Firmware 0.0.0-g
v1.22-67-g4299-dirty into /tmp/kaleidoscope-eric/sketch/82753643153034630-Model01-Firmware.ino/output...
- Size: firmware/Model01-Firmware/Model01-Firmware-0.0.0-g
v1.22-67-g4299-dirty.elf
  - Program:   25110 bytes (87.6% Full)
  - Data:       1317 bytes (51.4% Full)

To update your keyboard's firmware, hold down the 'Prog' key on your keyboard,
and then press 'Enter'.

When the 'Prog' key glows red, you can release it.

ls: cannot access /dev/serial/by-id/: No such file or directory
Looking for USB device with vid=1209 and pid=2301
ERROR: I couldn't find a USB device matching the keyboard's USB Vendor and Device IDs

I couldn't autodetect the keyboard's serial port.

If you see this message and your keyboard is connected to your computer,
it may mean that our serial port detection logic is buggy or incomplete.
In that case, please report this issue at:
         https://github.com/keyboardio/Kaleidoscope
make: *** [flash] Error 1

```

I'm running WSL on Windows 10 build 1809.  I can see the serial port at /dev/ttyS4.  But my WSL runtime of Ubuntu does not have a /dev/serial

```
eric@WLPTM196:/mnt/c/dev/Projects/thirdparty/keyboardio/Model01-Firmware$ ls /dev/serial
ls: cannot access /dev/serial: No such file or directory
eric@WLPTM196:/mnt/c/dev/Projects/thirdparty/keyboardio/Model01-Firmware$ stty -F /dev/ttyS4
speed 57600 baud; line = 0;
eof = <undef>; start = <undef>; stop = <undef>;
-brkint -imaxbel
-echo
```

Dan I specify the location/device number to use for flashing instead of relying on autodetection?

obra, 2019-09-13:
Actual flashing on windows may need to be done by running avrdude by hand
or by running windows-side avrdude.
ᐧ

benze, 2019-09-13:
I hacked around a little with the kaleidoscope-builder and manually set the DEVICE_PORT and DEVICE_BOOTLOADER_PORT to /dev/ttyS4, however when I try to flash, I get the following error:

```
avrdude: ser_open(): can't open device "/dev/ttyS4": Input/output error
```

I'm quite perplexed at this point.  I know for sure that my device is /dev/ttyS4 (if I disconnect the keyboard, then I cannot read from /dev/ttyS4), but not sure how to get avrdude to run.

What do I need to do to execute avrdude from windows?  I've tried running the equivalent command-line in windows, but don't know how to program the keyboard.  avrdude.exe just seems to be waiting indefinitely for some action, but not sure what.

Thanks,

Eric

obra, 2019-09-13:
The first thing to check on the Windows side is whether everything works
correctly from the Arduino IDE.  Instructions for that are on the
Kaleidoscope wiki.
ᐧ

benze, 2019-09-14:
The Arduino IDE worked fine using the sources from within the IDE only (ie
using the board manager and the example firmware).  Granted that was a
couple of weeks ago, so maybe I should rerun it.  But post that, I'd really
like to get this thing running from CMD line.

Are the steps to program from CMD line the same?  Run avrdude and hold down
the prog button?  Or does something else have to be executed first?  The
make file seems to trigger a reset or something first before running
avrdude.

Thanks

Eric

benze, 2019-09-14:
This is quickly becoming an exercise in frustration.  I have since spun up a Linux VM (VirtualBox) to compile an program the device.  The /dev/serial/by-id/ directory exists and the keyboard is found/listed there:

```
eric@XMEANT ~/dev/Projects/thirdparty/keyboardio/Model01-Firmware $ ls /dev/serial/by-id/
usb-Keyboardio_Model_01_Ckbio01E-if00
```
However, when I run make flash, the keyboard reset which happens before the flash disconnects it and it doesn't reconnect in time.  So the next step of auto-discovering the DEVICE_PORT_BOOTLOADER fails.  

So I have disabled the reset by setting in kaleidoscope-builder.conf:
```
NO_RESET=1
```

But now, when I run make flash, it is unable to find the correct device PID:
```
Looking for USB device with vid=1209 and pid=2300

Examining usb-Keyboardio_Model_01_Ckbio01E-if00

  ID_MODEL_ID 2301 != 2300
ERROR: I couldn't find a USB device matching the keyboard's USB Vendor and Device IDs
```

I can only assume then that means the keyboard has to be put into a bootloader mode by holding down the Prog button during bootup.  But the keyboard is not detected by Windows and mapped to the VM when it is booted with the Prog key down.

Is there now way to get this to work?

obra, 2019-09-14:
Hiya,

I'm sorry this is being so frustrating for you.

You are correct that the keyboard enters bootloader mode for a limited time
after holding down "Prog" as it's resetting.

On the windows side, I believe that all you're missing is the "force device
reset and avrdude flash" steps. To capture the "right" windows-side
commands, you can use the Arduino IDE: Open up its settings and turn on
"verbose mode" for flashing. That should show the commands you need to run
to flash a firmware (while holding down prog.)

I dearly wish this all worked better on the Windows side, but none of the
regular developers are Windows users.

On the VM side:
Windows is pretty pokey at handing off devices to VMs. My -recollection- is
that VirtualBox lets you hardcode devices that should always be handed
immediately to the VM.
ᐧ

benze, 2019-09-14:
> My -recollection- is that VirtualBox lets you hardcode devices that should always be handed immediately to the VM.

Indeed there is a setting to force it to hand off immediately to the VM, but that relies on Windows detecting it time, but that doesn't seem to be happening.  There is an option  (filter) that I can set the Vendor ID and Device ID to be handed off and have set them to 1209 and 2300 but I guess it takes longer to detect when booting up.  I've tried to force the bootloader mode by holding down the prog button as I plug in the keyboard but that too doesn't get detected and remapped over to the VM.  I've even gone as far as forced a serial port mapping in the VM to a COM port but that doesn't seem to be working either.

This is my first experience with an Arduino so I'm not sure which things are Keyboardio specific or not.  Such as the bootloader mode; I don't know if that is something. Designed into the firmware, or if it is something standard on an Arduino.  So I'm not sure how to keep poking at this to get a setup up and running properly.

obra, 2019-09-14:
@benze - A couple questions:

1, which I should have asked from the beginning: Is your end goal "program
your keyboard with custom firmware", "change your keyboard layout", or
"have a commandline development environment for your keyboard firmware"?
Depending on what you're really trying to do, there may be a better
solution.

2. Did you see my note about getting the reset and flash steps out of the
Arduino IDE?
ᐧ

benze, 2019-09-14:
@obra I guess it's a bit of both; right now, I'd like to build a custom firmware with some custom functionalities (ex: QuKeys, OneShot, SpaceCadet, etc).

Once I get that up and running, I would eventually like to see if I can play around with the firmware itself, but to be honest, with the amount of effort it is taking just to get this to build and flash, I don't know how easy/difficult it will be to tear into the firmware itself and modify functionality without breaking the board.

>2. Did you see my note about getting the reset and flash steps out of the Arduino IDE?

I tried to flash it from the IDE.  The only message I got about the reset was:
```
Forcing reset using 1200bps open/close on port COM4
```

So I still don't know which command-line the IDE triggers to reset the board.  I scoured around the arduino forums and found someone who had pasted the following as a reset command:
```
C:\arduino\hardware/tools/avr/bin/avrdude -C hardware/tools/avr/etc/avrdude.conf -patmega32u4 -cstk500v1 -P\\.\COM4 -b19200 -U hfuse:r:high.txt:s -U lfuse:r:low.txt:s
```

But without understanding the full hardware design of the keyboard, nor a thorough understanding of the bootloader/programming process, I'm fearful just blindly testing a command in case I brick the arduino by overwriting some of the lfuse/hfuse bits that mustn't be changed.

benze, 2019-09-15:
Finally success with a VirtualBox VM.  I had to play around with settings to get this to work though.  Final recipe was to add a _Settings->USB Filter_ with the following values: 
- VendorId: 1209
- DeviceId: 2300

I also had to modify `kaleidoscope-builder` from `sleep 3` to `sleep 5` after the reset command as 3 seconds was not enough for Windows to discover the device, assign it to the VM, and have Ubuntu auto-detect it and update its device map.  Similarly 7 seconds is too long and caused the keyboard to drop out of bootloader mode.  I don't know if the punt of time it waits in bootloader mode is configurable.

With this configuration, I can detect and Flash the device successfully.  However I am very concerned about the timing and fearful that a different windows config/etc could cause an additional delay and cause the setup to fail.  I would have liked to see the keyboard remain in bootloader mode much longer and not only have a 1-2 second window for the communication to launch.

I also noticed that the device name and serial number change when in bootloader mode, which was why my usb filter was originally failing to detect it.  Is this normal behaviour/expected for a serial number to change between modes like that?

obra, 2019-09-15:
It's not a serial number, so much as a "device id" -- It's not an uncommon
pattern for a device's bootloader to have a different id than its user
firmware. That make it easier for host-side tools to detect which mode the
device is in.

The amount of time the keyboard stays in bootloader mode is a trade-off
between "easy to flash from from slow tools" and "doesn't get in the user's
way when they don't mean to be in the bootloader"  -- In retrospect, I
agree that a longer timeout is warranted for the future. Because of the
nature of the bootloader, this isn't something we can change on shipped
devices without using an external flashing tool. (The bootloader is
supposed to be impossible to mess with over USB.)

One thing we -could- do would be to change kaleidoscope-builder to start
probing immediately after the stty command that reboots the keyboard, and
to check multiple times a second for up to, say, 10 seconds. I'd absolutely
take a patch or a seperate github issue for that.
ᐧ

closed 2020-04-27 by obra