keyboardio/Kaleidoscope #693: Flushing firmware breaks keyboard layout on 1.9.6

AleksanderPopov, 2019-10-04:
*Description:* After flushing keyboard layout does not match config layout
*Pre-conditions:* 
1. Use this config https://github.com/AleksanderPopov/.dotfiles/blob/master/firmware/firmware.ino
2. Setup Arduino with 1.9.6 keyboardio board using this instruction: https://github.com/keyboardio/Kaleidoscope/wiki/Add-keyboard-support-to-Arduino
[image]
*Steps:*
1. Flush firmware

*Expected result:* keyboard layout matches the config
*Actual result:* keyboard layout looks like 
[image]
So layout are completely messed up, have no idea why it's happening...
*Note:* Screenshot with actual keyboard layout was taken using Chrysalis, which can understand correctly which layout is in keyboard now ( for some reason )

AleksanderPopov, 2019-10-14:
any updates on that? it's still an issue.

obra, 2019-10-18:
It sounds like you were coming from an older firmware with a different
EEPROM layout. In the Chrysalis settings, you should have the ability to
reset the EEPROM. That -should- do it.
ᐧ

AleksanderPopov, 2019-10-21:
That seems to do the trick. Thank you @obra

closed 2019-10-21 by AleksanderPopov