keyboardio/Kaleidoscope #700: Implement device for the Empros Systems Operation Keyboard (the Tardis Keyboard)

rljacobson, 2019-10-21:
Assign to @rljacobson. This issue tracks the device implementation for the Empros Systems Operation Keyboard (the Tardis Keyboard).

# Background

The [Empros Systems Operation Keyboard, a.k.a the Tardis Keyboard](https://github.com/rljacobson/smurd), is actually a self-contained computer designed to control utility power systems for the National Grid Power Company in the UK, but since the computer part of the device cannot connect to a monitor or desktop computer, a custom controller is required to use the keyboard with a modern computer. The controller hardware has already been assembled.

# Foreground

As discussed in keyboardio/Kaleidoscope-Bundle-Keyboardio#22, I will begin implementing this device based on @algernon's branch for #695. When I have the implementation mostly complete, I will check back in with @algernon for more assistance. 

# Tasks

- [ ] Create a device class with hardware-specific implementation details.
- [ ] Use or adapt `Kaleidoscope-HIDAdaptor-KeyboardioHID.h`
- [ ] Adapt AVR-dependent components for use with ARM devices
    - [ ]  `Hardware::attachToHost` 
    - [ ] `Hardware::detachFromHost`
    - [ ] EEPROM stuff
    - [ ] `FirmwareDump`
    - [ ] `HostPowerManagement`
    - [ ] `USBQuirks`
    - [ ] Build system?
- [ ] Create indicator LED plugin / plugin for monochrome LEDs
- [ ] Investigate requirements for [Chrysalis](https://github.com/keyboardio/Chrysalis) integration, and update this list with the requirements as tasks

rljacobson, 2019-11-08:
@algernon Take a look at the code in [this repo](https://github.com/rljacobson/tardis) and let me know what you think. That repo is just a temporary home until I understand the best way to build the project. I have left some ToDo's in the code, and in some of them I have tagged you with a question about how to implement something or other. 

Since I am hung up on the build system—more like I haven't even attempted anything—I wonder if I should try to build the project with Platformio, which is a cross platform build system for embedded development. If you guys like how it works, maybe I could port the existing build system to Platformio as well. 

As I think about this, a successful move to Platformio would open the door to factoring out the Arduino dependencies themselves, making Kaleidoscope framework independent as well. Framework independence might not be a goal this project is interested in pursuing at the moment, but it's an interesting *idea* at least.

obra, 2019-11-12:
Hi Robert,

I'll let @algernon talk about the code itself, but I figured I should chime
in on platformio.  It's something we've investigated in the past and found
to be not quite the right fit. They're really focused on a single IDE and
everything running in their ecosystem. In terms of becoming more portable,
there's been some work on a cmake build system, which I suspect is more
likely to be our path forward.
ᐧ

rljacobson, 2020-01-02:
@obra Yes, I discovered a lot of previous discussion after I posted. You guys have already done the analysis on this question for sure.

> They're really focused on a single IDE and everything running in their ecosystem.

I'm not familiar with the history of the project, but they definitely have good support for a variety of IDEs: https://platformio.org/install/integration. (And of course the command line.) Everything running in their ecosystem is how they tame the cross-platform beast, so that's a deal-breaker.

But again, it looks like you guys have looked at this from every angle already.

obra, 2020-01-02:
After a bit of recent feedback, it sounds like we may want to revisit the
"absolutely not" feeling on PlatformIO.  It sounds like a lot has changed
on their side, and a lot has changed on our side, too.
ᐧ