keyboardio/Kaleidoscope #707: MagicCombo should use `KeyAddr`

algernon, 2019-11-04:
The `MagicCombo` plugin currently uses key indexes, which are 8-bit integers. This was made this way before `KeyAddr` existed. Now that it does, it makes more sense to use those, because then we don't have to play with key indexes, and can remove a whole lot of obsolete code (including the `keyIndex` macro/constexpr).

noseglasses, 2019-11-04:
We have the special invalid `KeyAddr` value. Could that serve to express MagicCombo's special case that was formerly expressed by the value `0`?

algernon, 2019-11-04:
Yes. Mind you, no user-code I know of uses the special 0 value. It doesn't make any sense to use it either. I'm pretty sure the special handling of it is a left-over from when we _did_ use a sentinel, before switching over to a separate, explicit length.