keyboardio/Kaleidoscope #715: Fix up virtual device support

algernon, 2019-11-07:
A fair number of things, especially in the device plugins, need to be aware of virtual devices, and be guarded with `KALEIDOSCOPE_VIRTUAL_DEVICE` or similar.

obra, 2019-11-07:
As we implement this in boards.txt and platforms.txt, I feel like we should
do it as a CPU menu per device, rather than fully duplicating the device
list.
ᐧ

noseglasses, 2019-11-09:
Now as the new hardware API is ready (kudos to @algernon for his patience and persistence when working on this huge set of changes) I am optimistic to start again working on the virtual builds soon. I want to use some spare time here and there to plan and discuss the implication of the design with respect to the new API.

Up to now we had a separate `Kaleidoscope-Hardware-Virtual` plugin which is something that I do not like too much. I feel like we could merge the virtual hardware to the core repo to become a proper device alongside all the other 'physical' devices. What's your opinion about that idea?

> As we implement this in boards.txt and platforms.txt, I feel like we should
do it as a CPU menu per device, rather than fully duplicating the device
list.

@obra, could you please explain what you mean by 'CPU menu per device'?

in my most recent version of a virtual build, I introduced a `virtual` Arduino architecture that comes with its own `boards.txt` and `platform.txt`. The individual platform file is mainly used to sneak in the `-DKALEIDOSCOPE_VIRTUAL_BUILD=1` to the compiler command line. The boards file is highly redundant but was necessary to point the build to the `virtual` core.

If we could move the virtual core stuff to a device directory in the core Kaleidoscope repo, the `virtual` arduino core would possible become obsolete. Then, we just would need to find a way to conditionally add the `-DKALEIDOSCOPE_VIRTUAL_BUILD=1`. Is that what you suggested, @obra?

obra, 2019-11-12:
> @obra, could you please explain what you mean by 'CPU menu per device'?

Have a look at how the various versions of the Atreus are implemented in boards.txt. Arduino has a facility for adding sub-menus to devices for things like "different controller" or various different features. I think this would end up being a less disruptive way to implement virtual devices than doubling the length of the device list in the Arduino IDE's dropdown. When I tried out the old virtual device stuff that way, it was distracting and confusing. (It's not a great bit of UI from Arduino and it was hard to tell which version of the core I had selected.)

I -think- we still want the virtual core, since it really is another Arduino platform, just targeting a soft implementation on an x86 host. Eventually, I think that's a really valuable thing we will be able to contribute to the Arduino ecosystem. I'd love to be able to give folks an easy way to write tests for their sketches.

But I don't think we need a separate boards.txt or platform.txt for it. (Well, we -may- need the platform.txt to get to the right set of compilers, but I'm not sure.)

noseglasses, 2019-11-19:
> Have a look at how the various versions of the Atreus are implemented in boards.txt. Arduino has a facility for adding sub-menus to devices for things like "different controller" or various [...]

Great. That will do the job and avoid the duplicate boards.txt. Will have a try.

closed 2025-06-01 by obra