keyboardio/Kaleidoscope #717: EEPROM Settings and firmware flashing with changed plugins are not friends

obra, 2019-11-07:
We  really need to rethink how we do EEPROM storage ordering across firmware flashes. The simplest might just be to always dump eeprom over focus before flash and then restore eeprom afterward.

algernon, 2019-11-07:
Dumping and restoring in and of itself is not enough, because we may need to rearrange it too. That's where #344 comes in handy: it allows us to dump the layout! That way we can compare the previous and the current, and rearrange accordingly.

obra, 2019-11-07:
Right. Sorry. I meant a...semantic dump, rather than a binary dump.
ᐧ

algernon, 2022-06-08:
Our CLI tools don't do this, but Chrysalis does semantic save & restore now. Do we want to teach our CLI tools to do the same, too?

obra, 2022-06-09:
it’d be nice.