keyboardio/Kaleidoscope #724: key_K suddenly stopped working

wisdominant, 2019-11-19:
'key_K' suddenly stopped working.
I tried to flash on the build of original ino file, every other key seems to be working fine, except the 'k' key.
I am running OSX 10.14.6
This was the output I get when I flash
Is this the hardware glitch?? 
Please, any help will be greatly appreciated...
```
BOARD_HARDWARE_PATH="/Users/user/Documents/Arduino/hardware" /Users/user/Documents/Arduino/hardware/keyboardio/avr/libraries/Kaleidoscope/bi
n//kaleidoscope-builder flash
Building ./Model01-Firmware 0.0.0 into /var/folders/wb/wh089nb90_q8358nsjy8cgn80000gn/T//kaleidoscope-user/sketch/54246735-Model01-Firmware.
ino/output...
- Size: firmware/Model01-Firmware/Model01-Firmware-0.0.0.elf
  - Program:   25712 bytes (89.7% Full)
  - Data:       1310 bytes (51.2% Full)

To update your keyboard's firmware, hold down the 'Prog' key on your keyboard,
and then press 'Enter'.

When the 'Prog' key glows red, you can release it.

2019-11-19 15:16:33.735 system_profiler[19580:117252] SPUSBDevice: IOCreatePlugInInterfaceForService failed 0xe00002be
2019-11-19 15:16:37.016 system_profiler[19594:117611] SPUSBDevice: IOCreatePlugInInterfaceForService failed 0xe00002be
Connecting to programmer: .
Found programmer: Id = "CATERIN"; type = S
    Software Version = 1.0; No Hardware Version given.
Programmer supports auto addr increment.
Programmer supports buffered memory access with buffersize=128 bytes.

Programmer supports the following devices:
    Device code: 0x44
```

obra, 2019-11-19:
that sounds like a hardware failure. Please email us at help@Keyboard.io

hackmalife, 2020-02-03:
Thanks to @obra, I got my keyboard back and k_key works.
I will report the following troubleshooting in a [separate issue](https://github.com/keyboardio/Kaleidoscope/issues/809)

closed 2020-02-03 by hackmalife