keyboardio/Kaleidoscope #726: Device API: What belongs where (props/implementation)

noseglasses, 2019-11-21:
This is not an actual issue but rather a question that IMHO could do with some discussion and clarification.

The new device API introduced `...Props` classes that describe device and drivers. AFAIK, it is currently not exactly defined what parts of information belong in `...Props` classes and what in the corresponding implementation classes.

An example that demonstrates why I bring up this topic is the method 
```cpp
uint8_t getLedIndex(uint8_t key_offset)
``` 
that is currently part of the LED driver implementation. During my work on the virtual device it turned out that the simulator needs to have access to this method as it has to map a key offset to the matching LED index. This is e.g. useful when during future simulation based integration testing the actual color value of a LED is meant to be compared with a set value. But the physical devices driver classes are not build (excluded by `#ifndef KALEIDOSCOPE_VIRTUAL_BUILD` clauses). That's why the virtual driver classes cannot ask their physical siblings but have to ask their `...Props` classes instead.

My solution is to move `getLedIndex` to the LED drivers properties class.

I understand the Props classes to contain anything that is necessary to describe the implementation. Thus, they may contain any type of stateless, hardware agnostic information in terms of constants and (not necessarily `constexpr`) static methods. This perspective considers mappings through functions as descriptive information, too.

In other words, only implementation details that relate to actual hardware and that cannot be build for the virtual architecture, as well as anything that is stateful **must** be part of the implementation classes.

@obra and @algernon, what's your opinion about this? 

In any case, we should document the answer to this question somewhere in the developer relevant parts of the docs (unless its not already there and I just missed it).

algernon, 2019-11-21:
I think things like `getLEDIndex()` should indeed be moved to `Props`, because they're just lookups, really. Ideally, the led index table would be a (private) member of the `Props` struct too. In a sense, mapping is just a way to access a property, the LED index corresponding to a key. As such, Props is a good place for it.

I do feel we may need better wording, but in principle, I agree with you.

(You didn't miss it, we do not currently have decent docs for the new API. That's something I plan to fix in the near future, but didn't get there yet.)

obra, 2019-11-21:
Hrm. Part of the point of naming it "Props" was that it was my
understanding that it was "just" a data structure and wouldn't have code in
it.
ᐧ

noseglasses, 2019-11-21:
> Hrm. Part of the point of naming it "Props" was that it was my
understanding that it was "just" a data structure and wouldn't have code in
it.

The way I envisioned the design when I came up with it was to provide a way to define a clear separation between a __architecture independent,  stateless__ description and a possibly hardware-dependent implementation of device and drivers. That should facilitate the implementation of a generic virtual device. Now, after I have implemented that, I can definitely confirm that that plan worked out perfectly. It is really **very** easy to support all keyboards.

A second design goal of the new API was to make the definition of new devices as simple as possible by allowing them to be composed from mostly existing parts. As far as I can see, for the majority of the currently supported keyboards that also worked out nicely. Their device description is very compact.

I do not see any problem with description classes having run time functions (as long as they are architecture independent and stateless). That's - FWIW - the reason why I called those classes *description* in my initial design. I had no intention to restrict their content to mere constant data (properties). 

@obra, do you have any specific concerns?

algernon, 2019-11-21:
If `getLedIndex()` is constexpr, then it's essentially a fancier way of saying `Props::key_led_index[key_offset]`. It's compile-time, and Props is still data-only (at runtime), with a few compile-time accessors to make a few lookups look nicer. I think that's reasonable.

I wouldn't want run-time methods in there, either.

noseglasses, 2019-11-21:
>  it's essentially a fancier way of saying Props::key_led_index[key_offset]

No, that's a misconception. Mind you, `constexpr` is a mere information that a function *can* possibly be evaluated at compile time. 

Marking a function `constexpr` and not providing an implementation causes a compiler error as soon as that function is evaluated in a `constexpr` context. Only because the latter does not happen yet with `getLedIndex()`, the compiler lets you get away with it. Marking that function `constexpr` is, thus, pointless. The same holds for `key_led_index`. Marking it both `constexpr` and `PROGMEM` is contradictory - `const PROGMEM` is correct. It works, though, as `constexpr` implies `const` and the compiler is free to generate a symbol for it as soon as it is going to be used in a non-constexpr context.

But as the mapping from key offsets to LED indices is needed at runtime, there's no way to do this differently. 

>  I wouldn't want run-time methods in there, either.

Why? Just your guts or do you have technical concerns?

noseglasses, 2019-11-21:
> Marking a function constexpr and not providing an implementation causes a compiler error as soon as that function is evaluated in a constexpr context. 

This might better read "Marking a function constexpr and not providing an implementation _in the same compilation unit_ causes a compiler error as soon as that function is evaluated in a constexpr context.".

noseglasses, 2019-11-21:
FWIW, because `constexpr` works that way, it is easily possibly to be confronted with _undefined symbol_ warnings for `static constexpr` or global `constexpr` variables. That happens when the address or a reference of such functions is taken.

obra, 2019-11-21:
> The way I envisioned the design when I came up with it was to provide a way to define a clear separation between a architecture independent, stateless description and a possibly hardware-dependent [...]

> A second design goal of the new API was to make the definition of new devices as simple as possible by allowing them to be composed from mostly existing parts. As far as I can see, for the majority [...]

My priorities were in the opposite order, but I'm happy we could accommodate both. 

However, it sounds like we've intercommunicated a bit about the intent of the Props classes.

While I wasn't in love with the initial "Description" name, it seemed pretty clear to me, as a  "Description" is explicitly data and strongly implied to me that there was no code involved.

As we worked through names, the whole reason I ended up going with 'Props' was because these were supposed to be really unambiguously data, not code.

I'd been concerned that these things might contain code and got reassured that they didn't in https://github.com/keyboardio/Kaleidoscope/pull/695#issuecomment-544841461

There was a little bit of follow up in these comments:

https://github.com/keyboardio/Kaleidoscope/pull/695#issuecomment-546509116
https://github.com/keyboardio/Kaleidoscope/pull/695#issuecomment-546513753

To me, the natural place for accessors for that data is in the paired implementation class (or implementation baseclass) It was my expectation that if the simulator would need to implement an LED driver, say, there'd be some kind of SimulatedLEDDriver implementation class that you'd use. 

If these things need code in them, then the design doesn't work like it was described to me :/

noseglasses, 2019-11-22:
@obra, a distinction between data and code does not make sense to me here. You still did not provide a technical reason for why you are so focused on it. 

It would definitely help my understanding if you could provide examples for where you expect something to go wrong once we would allow "code" in Props classes.

I am not a computer scientist so my theoretical background is limited. 

In theory one can replace any stateless function with a constant data lookup table and vice versa. So if a lookup table is a property/description, a stateless function is one as well.

I see other aspects that are of greater importance being considered.

1. stateful <--> stateless

That's important for the descriptive (property) aspect of the Props classes. Although it is not entirely possible to prevent statefullness with C++ functions as the language allows static variables in functions, I would expect it to be enough to put big fat warnings in the Base driver classes function description that demand statelessness. 

Another possibility for statefulness in Props classes would be instanciation. This can easily be prevented by
making the Base drivers' constructors private. Then the compiler will disallow any derived classes' instantiation.

2. architecture agnostic <--> arch. dependent

Having portable Props-methods is very important to allow the virtual devices to mimic any physical devices entirely
based on their stateless description (or rather the description of their interfaces).

This requires to document the Base driver classes' methods in a way that unmistakenly makes
clear that all derived classes may only reimplement them in a way that they build on **any** platform.

3. compile time constant <--> runtime data

That is something I wrapped my head around for a while. I find that we
should define and document in the Base driver classes which functions we enforce to be 
constexpr (as we really need them at compile time).

The enforcement can be implemented by adding static_asserts to the Base device 
class that call those methods. Only if a method is constexpr, it can be
used in a static_assert that is evaluated at compile time.

Other methods, like the LED index mapping method I would suggest not to tag constexpr
in the Base drivers. Derived drivers are then free to make their possible re-implementation constexpr which gives
the compiler the freedom to choose between compile time and runtime evaluation
at the methods' point of use. Mind you, we are in a non-polymorphic context so the
derived class members will always shadow the base classes' and `constexpr` is not part of the function signature in C++.

> My priorities were in the opposite order, but I'm happy we could accommodate both.

For the first of your priorities - modular devices - we would not have needed any Props classes at all. The separation in drivers and their implementation classes would have been enough to provide the required modularity. 

Thus, the Props classes that we are talking about entirely serve the second priority - virtual builds.

> To me, the natural place for accessors for that data is in the paired implementation class (or implementation baseclass)

Agreed. An accessor is a method that possibly returns stateful data. That's not what `getLedIndex()` is about. But talking about it, it would be wise to rename it to something without the accessor name prefix `get`. Or do you see any need for a stateful LED mapping? If so, things are entirely different.

> It was my expectation that if the simulator would need to implement an LED driver, say, there'd be some kind of SimulatedLEDDriver implementation class that you'd use.

Exactly. That's the way it works. The class name is slightly different in my implementation but there's a simulated LED driver class that mimics the hardware's LED driver based on its description - including the `key_offset` to `index` mapping. The same do the key scanner and the EEPROM.

noseglasses, 2019-11-22:
@obra, would you feel better if we would enforce statelessness of Props classes' functions by generally enforcing them to be `constexpr` (which are by definition stateless)?

I found a way to define arrays like `key_led_index` and let them context dependent be evaluated at compile time or at runtime. The nice thing about this approach is that event though the compiler is free to runtime instantiated `constexpr` functions, they are still 100% stateless which is guaranteed by the standard.

Here is an example that demonstrates some of these aspects (enforced statelessness and prevented props-class instantiation):
```cpp
#include <stdint.h>
#include <avr/pgmspace.h>

//*********************************************************
// Header content
//*********************************************************

// This example demonstrates how an array can be defined in a header and be used for
// compiletime and runtime evaluation depending on the context.

struct LEDDriverProps {
    static constexpr int8_t ledIndex(uint8_t key_offset) { return 0; }
    private:
        LEDDriverProps() {}  // Prevent instantiation of this class and derived classes
};

struct Model01LEDDriverProps : public LEDDriverProps {

    // Declaring this PROGMEM array weak is important to avoid linker errors. The
    // array will be instanciated in every compilation unit where the header is included
    // but finally only one instance is going to end up in the firmware binary (PROGMEM).
    //
    __attribute__((weak))
    static constexpr int8_t key_led_map[] PROGMEM = {
        3, 4, 11, 12, 19, 20, 26, 27,     36, 37, 43, 44, 51, 52, 59, 60,
        2, 5, 10, 13, 18, 21, 25, 28,     35, 38, 42, 45, 50, 53, 58, 61,
        1, 6, 9, 14, 17, 22, 24, 29,     34, 39, 41, 46, 49, 54, 57, 62,
        0, 7, 8, 15, 16, 23, 31, 30,     33, 32, 40, 47, 48, 55, 56, 63,
    };

    static constexpr int8_t ledIndex(uint8_t key_offset) { return key_led_map[key_offset]; }
};

class Model01LEDDriver {
    public:
        int8_t getLedIndex(uint8_t key_offset) { 
            return pgm_read_byte(&Model01LEDDriverProps::key_led_map[key_offset]); 
        }
};

struct Model01Props {
    typedef Model01LEDDriverProps LEDDriverProps;
    typedef Model01LEDDriver LEDDriver;
    //...
};

// A bad LED driver whose implementer did not adhere to the constexpr requirements
// of ledIndex(...).
//
struct BadLEDDriverProps : public LEDDriverProps {
    static int8_t ledIndex(uint8_t key_offset) { return 0; } // OOPS, non constexpr
};
struct BadLEDDriver {};

struct BadDeviceProps {
    typedef BadLEDDriverProps LEDDriverProps;
    typedef BadLEDDriver LEDDriver;
};

template<typename _DeviceProps>
struct DeviceBase {
   static_assert((_DeviceProps::LEDDriverProps::ledIndex(0), 1), 
      "If your code fails to compile here, your LED device props class has implemented a "
      "non-constexpr method ledIndex(...)");
};

typedef DeviceBase<Model01Props> Model01;
typedef DeviceBase<BadDeviceProps> BadDevice;

// Prove that compile time access to ledIndex and thus key_led_map is possible.
//
static_assert(Model01LEDDriverProps::ledIndex(2) == 11, "constexpr access to key_led_map fails");

//*********************************************
// cpp-file content
//*********************************************

// But still...every constexpr function has the potential to become a runtime function
// when evaluated in a non-constexpr context. if that functions was declared as constexpr,
// the standard enforces its statelessness which makes it 100% proof against side effects.

int8_t enforcedRuntimeEvaluation(uint8_t key_offset)
{
    return Model01LEDDriverProps::ledIndex(key_offset); // remember, ledIndex(...) is declared constexpr
}

// Here, we let the compiler decide whether it will evaluate ledIndex(...) at runtime or at
// compile time.
//
int8_t possibleRuntimeEvaluation()
{
    // avr-gcc 5.4.0 opts for runtime evaluation even with -03
    return Model01LEDDriverProps::ledIndex(1); // Can do compile time evaluation
}

int8_t possibleRuntimeEvaluation2()
{
    // If we push the compiler it a little, it has to go for compile time evaluation of ledIndex(...)
    constexpr auto tmp = Model01LEDDriverProps::ledIndex(1); // Must do compile time evaluation
    return tmp;
}

int main(int argc, char**argv) {

    Model01LEDDriver led_driver; // LED drivers can be instantiated 
    auto runtime_led_id = led_driver.getLedIndex(4);

    // Model01LEDDriverProps led_driver_props; // Driver Props classes not because we make
                                               // their costructors private.

    Model01 model_01; // The model 01 is a good device...
    // BadDevice bad_device; // Will fail to compile as BadLEDDriverProps implemented a non-constexpr
                             // function ledIndex(...) which will be caught by the static_assertion in
                             // template class DeviceBase

    return runtime_led_id;
}
```
You can play with this example [here](https://www.godbolt.org/z/XaZhmG).

noseglasses, 2019-11-22:
FWIW, @algernon, could we have a 
```cpp
int8_t getLedIndex(KeyAddr key_addr);
```
instead of
```cpp
int8_t getLedIndex(uint8_t key_offset);
```
?

algernon, 2019-11-22:
If you figure out how to get `KeyAddr` there without the compiler being very upset, yes.

obra, 2019-11-23:
>
> It would definitely help my understanding if you could provide examples
> for where you expect something to go wrong once we would allow "code" in
> Props classes.
> I am not a computer scientist so my theoretical background is limited.

To me, this is much more about "human factors" than about the technical
side.  It's not about deep C++ stuff, so much as when a regular developer
tries to build things with them.

If the Props classes are more than straight datastructures, we've given
them the wrong name.

If Props classes were to have code, It's not at all clear to me when you'd
want a given bit of code in the 'Props' class vs the paired implementation
class. And that means that it's going to be something that's harder for
developers to work with. That had been part of what I'd been getting at
when you put together the first implementation and I tried to ask for a
complete spec of what would go in a device's "Description"

I don't think it's reasonable to demand that developers who are trying to
implement Kaleidoscope for a device break apart all of their code into two
classes based on whether that function is best implemented using
hardware-specific APIs, which is, I -think- where the implementation you're
proposing draws the line.

@algernon and I were talking this through yesterday. My gut feeling is that
any sort of accessor or helper you'd want for the data in a Props class
ought to be in the implementation's base class, which can be used by either
the real implementation or a simulator implementation.  He's said he hopes
to get this written up in the next day or two.

I'm absolutely certain we can get to something that feels reasonable to
everybody. It may just take some work and more talking :)

ᐧ

noseglasses, 2019-11-23:
> My gut feeling is that
any sort of accessor or helper you'd want for the data in a Props class
ought to be in the implementation's base class, which can be used by either
the real implementation or a simulator implementation.  

@obra & @lgernon, I am not sure if I understand correctly how that's meant to work.
It would probably easiest if you could demonstrate that in terms of a code example that deals with `getLedIndex`. That's currently the only function that causes trouble with the virtual device. But should prepare for it to become necessary to add more functions like that in the future that also must be available to the virtual device in order to mimic correct behavior.

algernon, 2019-11-23:
Here's roughly how I imagine it would work:

### `driver/led/Base.h`

```c++
struct BaseProps {
  static constexpr uint8_t led_count = 0;
  static constexpr uint8_t key_led_map[] = {};
};

template <typename _Props>
class Base {
 // ...
 uint8_t getLedIndex(uint8_t key_offset) {
    return pgm_read_byte(_Props::key_led_map[key_offset]);
  }
 // ...
};
```

### `device/vendor/Board.h`
```c++
struct BoardLEDProps: public led::BaseProps {
  static constexpr uint8_t led_count = 2;
  static constexpr uint8_t key_led_map[] PROGMEM = { 0 };
};
```

### `device/vendor/Board.cpp`

```c++
constexpr uint8_t key_led_map[] PROGMEM;
```

# Further comments

I'm not yet sure how the virtual hardware would be set up, but this _hopefully_ would allow them to reuse the same function and data. Mind you, this does not - yet - account for the case where we have _less_ LEDs than keys, would need to figure something out for that (eg, by using `KeyAddr`, but that's complicated). Thankfully, none of our supported boards are like that at the moment, so figuring that out is not crucial right now.

Right now, the base class has a dummy `getLedIndex()` implementation, and derived LEDDrivers implement their own. With the above setup, we wouldn't need a dummy + implementations, just a common one, and derived classes wouldn't need their own.

noseglasses, 2019-11-23:
```cpp
static constexpr uint8_t key_led_map[] = {};
```
You can't have a zero sized constexpr array, unfortunately.

The rest of your code could work and looks like a decent solution for the time being.

Thanks, @algernon, for the code example. I will update my virtual hardware implementation PR accordingly.

> Mind you, this does not - yet - account for the case where we have less LEDs than keys, would need to figure something out for that (eg, by using KeyAddr, but that's complicated).

The entries in `key_led_map` could be set to a sentinel value for those keys without LEDs. For the case of less LEDs than keys, it is known how many keys are available by looking at the key scanner props or the `KeyAddr` type.

noseglasses, 2019-11-23:
There's one more thing that needs to be considered. What about those devices that wire their LEDs in the same order as keys? 
They will have to define a trivial { 0,1,2,...} array. Not necessary with a constexpr function.

algernon, 2019-11-23:
Those devices can either have a trivial array, or override the `getLedIndex()` method in the derived class. That'd make the virtual hardware a bit harder to do, but worst case, they could have the trivial array for virtual-only, and the `getLedIndex()` override otherwise. On the other hand, that's more trouble than its worth.

In any case, I'd worry about that once we see a board that wires LEDs the same order as the keys. :)

noseglasses, 2019-11-23:
> If the Props classes are more than straight datastructures, we've given
them the wrong name.

@obra, that's only true for your definition of *properties*. You still did not provide any explanation of why you fear even stateless constexpr functions so much and how they differ from constexpr arrays. Gut feelings are not convincing to me.

> If Props classes were to have code, It's not at all clear to me when you'd
want a given bit of code in the 'Props' class vs the paired implementation
class. And that means that it's going to be something that's harder for
developers to work with.

What belongs where, i.e. what inventory can be re-implemented, is defined in the Props and implementation base classes the same way as with other interfaces in Kaleidoscope. There's no room for speculation.

> I don't think it's reasonable to demand that developers who are trying to
implement Kaleidoscope for a device break apart all of their code into two
classes based on whether that function is best implemented using
hardware-specific APIs, which is, I -think- where the implementation you're
proposing draws the line.

i agree that it's not perfectly convenient to do so. But the same developers, on the other hand, have a firmware available whose development is supported by a simulated test bench that will protect them from possibly subtly broken commits and that enables simulation based testing also for their hardware, without any extra implementation work to do.

But still, the new API already demands breaking apart the implementation for good reasons. It is hopefully soon well documented and it is defined what to put where. 

So this breaking apart thing is a general question and has nothing to do with the `getLedIndex(...)` problem. This appears to be more about whether it's a good API or not. i like it a lot the way @algernon implemented it. But anyway, currently it's the only one that meets all its design targets. Ideas for a better one are certainly welcome...

noseglasses, 2019-11-23:
> Those devices can either have a trivial array

Sounds like a waste of PROGMEM.

> or override the getLedIndex() method in the derived class. That'd make the virtual hardware a bit harder to do

*harder to do* is a nice euphemism for *impossible*.

noseglasses, 2019-11-23:
The more I think about it the more complicated a pure data based Props solution gets. We should have another constant `max_key_led_id` that helps `getLedIndex(...)` with those keyboards that have no key LEDs at all.

```cpp
int8_t getLedIndex(uint8_t key_offset) {
   return (key_offset > max_key_offset) ? no_key_led : key_led_ids[key_offset];
}
```

noseglasses, 2019-11-23:
:rage2:

Ok, I give up, I surrender, I capitulate ... not to you guys, but to the f..king Harvard architecture that doesn't speak proper C++. Wait let's better blame it on avr-gcc not on the architecture, but anyway.

Look at this simplified version of the code that I naively proposed yesterday:

```cpp
struct Model01LEDDriverProps : public LEDDriverProps {

   __attribute__((weak))
   static constexpr int8_t key_led_map[] PROGMEM = { 11, 12, .... };

   static constexpr int8_t ledIndex(uint8_t key_offset) { 
      return key_led_map[key_offset]; 
   }
};

class Model01LEDDriver {
   public:
      int8_t getLedIndex(uint8_t key_offset) { 
         return pgm_read_byte(&Model01LEDDriverProps::key_led_map[key_offset]); 
      }
};
```

Here's the explanation why this is fundamentally broken:

As a constexpr method like `ledIndex(...)` can be instantiated in a non-constexpr context and thus become a normal runtime function, it will become
```cpp
int8_t ledIndex(uint8_t key_offset) { 
   return key_led_map[key_offset]; // Attempts to read from SRAM at address
                                   // key_led_map + key_offset
}
```
instead of
```cpp
int8_t ledIndex(uint8_t key_offset) { 
   return pgm_read_byte(&key_led_map[key_offset]); 
}
```

That's a nice example for C++ code that works nicely on Von-Neumann and fails on Harvard.

`pgm_read_byte` is a macro that injects assembler code. And that's not supported in C++11's constexpr functions.

Now, frustrated as I am, I withdraw my idea of stateless methods in Props. Let's go with the data only solution!

noseglasses, 2019-11-23:
What about a property `led_ids_equal_key_offsets` or similar? That would enable

```cpp
int8_t getLedIndex(uint8_t key_offset) {
   if(led_ids_equal_key_offsets) {
      return key_offset;
   }
   // Will be optimized out for !!led_ids_equal_key_offsets
   //
   return (key_offset > max_key_offset) ? no_key_led : pgm_read_byte(&key_led_ids[key_offset]);
}
```

algernon, 2019-11-23:
> What about a property `led_ids_equal_key_offsets` or similar?

That works for me. Not sold on the naming (will post a few other ideas as soon as kids are asleep), but the idea is fine with me.

obra, 2019-11-23:
> 
> 
> What about a property led_ids_equal_key_offsets or similar?
Let’s add it the first time we see real hardware that needs it?

noseglasses, 2019-11-23:
Would one of the hardware special agents please be so kind and enlighten me about why it is so unlikely that someone builds a keyboard that wires LEDs like keys?

obra, 2019-11-23:
> 
> 
> Would one of the hardware special agents please be so kind and enlighten me about why it is so unlikely that someone builds a keyboard that wires LEDs like keys?

The simplest answer is that very often the number of rows and columns you have for an LED matrix driver chip don’t match the rows and columns you have for the key matrix. Also, often, there are other leds, maybe for status or underglow. 

And sometimes, changing the ordering makes routing the circuit board easier. 

I’ve never seen a case where we didn’t need a mapping table to get the positions aligned. That certainly doesn’t mean it can’t be done. But adding the functionality to support it without ever seeing it in the field feels like premature optimization.

noseglasses, 2019-11-23:
> But adding the functionality to support it without ever seeing it in the field feels like premature optimization.

Agreed. Thanks!

noseglasses, 2019-11-23:
This discussion led to https://github.com/keyboardio/Kaleidoscope/pull/729. Closing.

closed 2019-11-23 by noseglasses