keyboardio/Kaleidoscope #730: EEPROM storage for TapDance

algernon, 2019-11-25:
We want Chrysalis to support TapDance, at least in a simplistic way. For that, we need to be able to store tapdance sequences in EEPROM. With the recent introduction of `DynamicMacros`, we can use a similar technique for TapDance too: we'll store a list of keys in EEPROM, terminated by `Key_NoKey`. Focus would allow getting or setting the whole lot at once, parsing is up to the other side (and will be considerably easier than DynamicMacros, because this is just a list of 2-byte Keys, with a sentinel at the end).

On the firmware side, this could be implemented as a separate plugin from TapDance, to make the feature optional. We could name it `EEPROM-TapDance` for example. All it would do is provide an `onSetup()` for initial settings & cache load, `onFocusEvent()` to allow query & update (and update would refresh the cache too, just like `DynamicMacros`). In addition, we'd provide a `.performAction()` method, which could be called from `tapDanceAction()`. It would return a bool, signalling whether the action was found in - and performed from - EEPROM, or not.

This would allow us to write a `tapDanceAction()` like this:

```c++
void tapDanceAction(uint8_t tap_dance_index, KeyAddr key_addr, uint8_t tap_count,
                    kaleidoscope::plugin::TapDance::ActionType tap_dance_action) {
  if (EEPROMTapDance.performAction(tap_dance_index, key_addr, tap_count, 
                                   tap_dance_action))
    return;

  switch (tap_dance_index) {
  case 0:
    return tapDanceActionKeys(tap_count, tap_dance_action, Key_Tab, Key_Escape);
  }
}
```

We also need a way to tell the `EEPROMTapDance` plugin how many built-in plugins we have. We need this so we can have both built-in and eeprom-stored dances. We can't just add an offset to the `performAction()`'s first argument, because the plugin uses that as an index into eeprom too. So, something like `EEPROMTapDance.setBuiltinDances(N)` would be needed. The plugin would then use this to offset the tap dance index when looking up from Storage. User sketches would need to call this from `setup()`. Since we already advocate for using an enum, we could recommend adding a `LAST_BUILTIN_DANCE` member there, and use that as `N`.

noseglasses, 2019-11-25:
> We also need a way to tell the EEPROMTapDance plugin how many built-in plugins we have. 

You mean build-in plugins of any type, i.e. essentially the number of plugins passed to `KALEIDOSCOPE_INIT_PLUGINS`? Or do you refer to the number of registered tap-dance plugins?

There are certainly means to determine both at compile time (during invocation of `KALEIDOSCOPE_INIT_PLUGINS`).

Are there possibly other plugins that might need to query such compile-time known statistical information about the registered plugins? We could make this available via an additional hook and thus free the user from having to pass such information (like `EEPROMTapDance.setBuiltinDances(N)`) explicitly from within the sketch. Having the user pass such information can be problematic if plugins are added or removed later, after the `N` was hard-coded with likely a literal constant.

Here's an idea how we could add a service that somewhat automatizes this: https://github.com/keyboardio/Kaleidoscope/issues/732

algernon, 2019-11-25:
> Or do you refer to the number of registered tap-dance plugins?

I wanted to write "built-in dances", but thinko'd there, so nothing to do with the number of plugins.

noseglasses, 2019-11-25:
Could `EEPROM::onSetup()` possibly scan the keymap and search for the tap dance key with the highest ID?

algernon, 2019-11-25:
Oooh, yes, yes, it could. Great idea, thank you!

algernon, 2019-11-26:
I think I found an issue with scanning the keymap in `EEPROMTapDance::onSetup()`: as far as IDs go, there's zero difference between a built-in and an eeprom-stored dance. If we can the keymap, we'll see both. If we only scan the built-in keymap, that's misleading, because built-in keymaps should be able to refer to eeprom-stored dances too.

We could solve this issue by having a separate range for eeprom-stored dances, but... that'd complicate other parts of the code far too much.

So I think the safest is to require an explicit setup step.

commit b415a859 (2019-12-06): New plugin: DynamicTapDance

This implements an extension to the `TapDance` plugin, allowing us to store
`tapDanceActionKeys()`-esque lists in Storage. The core idea here is very
similar to that of `DynamicMacros`: we dump/restore the full list via Focus,
build a cached index on setup and any updates, and play back the selected key
when need be.

Unlike `DynamicMacros`,this plugin is built on top of `TapDance` and cannot
function without it.

Fixes #730.
PR #734 (2019-12-06): New plugin: DynamicTapDance
This implements an extension to the `TapDance` plugin, allowing us to store `tapDanceActionKeys()`-esque lists in Storage. The core idea here is very similar to that of `DynamicMacros`: we dump/restore the full list via Focus, build a cached index on setup and any updates, and play back the selected key when need be.

Unlike `DynamicMacros`,this plugin is built on top of `TapDance` and cannot function without it.

Fixes #730.
closed 2019-12-06 by obra