keyboardio/Kaleidoscope #736: Speeding up Kaleidoscope builds by conditionally preventing #includes

noseglasses, 2019-11-28:
C++ is definitely not the fastest programming language when it comes to compile times, especially when heavy use of templates is made. 

In Kaleidoscope, templates are mainly used to automagically setup infrastructure in the sketch, e.g. for the plugin API, transient LED modes and  - possibly soon to come - sketch exploration. 

Therefore, we have to accept that the sketch builds longer than other compilation units.

Unfortunately, slow builds also affect other compilation units, also those that do not require any magic at all. This is because Kaleidoscope has this central header `Kaleidoscope.h`. All infrastructure that is needed for the magic stuff is pulled in by this header, a **lot** of stuff.
That stuff in consequence is also pulled in by every compilation unit that includes `Kaleidoscope.h`. 

Compilers don't know what they need/use later on in a compilation unit. They have to do syntactic checks regardless. Thus, a huge lot of code is compiled over that is actually used by the sketch alone.

Here's an idea how to solve this problem:

If we had a macro, say `KALEIDOSCOPE_SKETCH`, that we define at the compiler command line, only for the sketch, we could conditionally exclude stuff from being pulled in by other compilation units, by wrapping the `#include` directives in `#ifdef` guards.

Although we are talking about a firmware that still compiles comparably fast, we should take care not to let compile times grow too much. I feel they begin to hurt me during development. Thus, they can be expected to hurt users too. 

Our CI testing with Travis could also need some speed up.

I didn't work on ways to define a macro like `KALEIDOSCOPE_SKETCH`. Ideas are welcome.
Once we can teach our build system how to do that, I will go through the code and try to speed things up.

algernon, 2019-11-29:
I've been pondering about this idea for the past few hours, and I have a counter-proposal. Our build times are very long, indeed, a huge part of that is having to build a _lot_ of examples, 49, to be exact. What if we didn't build all of them? What if we only built a subset? We'd cut down on `make smoke-examples` time considerably, without having to augment `Kaleidoscope.h` with a ton of ifdefs.

Which subset? Well, what if we built a few **big** examples, one for each device? These would pull in _all_ of the plugins, and use as many features as possible. Obviously, we'd need to ignore firmware sizes in this case, because the results will be far too big for most of our boards. But this would dramatically cut down on build times.

We'd still keep the current examples, just wouldn't run them in Travis. Having One Big Example Sketch also makes development easier, because if one's working on things that affect multiple plugins, then building one example is faster and easier than building one for each affected plugin.

obra, 2019-11-29:
Is our caching stuff still working right?
ᐧ

algernon, 2019-11-29:
It is, that's why it takes _only_ 10 minutes. :)

noseglasses, 2019-11-29:
> without having to augment Kaleidoscope.h with a ton of ifdefs.

The problematic headers are mostly in kaleidoscope_internal. What if we would add the ifdefs there without touching Kaleidoscope.h? 
Inclusion is a lightning fast process compared to parsing and compilation.

But what you proposed makes sense, too. We sould do both. But I gess adding ifdefs only to four or five proplematic files can already do **a lot**, judging from experience with similar cases.

algernon, 2019-11-29:
A few files, especially if they're inside `kaleidoscope_internal` - I'm fine with that.

noseglasses, 2019-11-29:
If we find a way to pass a macro only to the sketch, I will be able to provide timing figures within short time, that will show if my idea pays off or not.

algernon, 2019-11-29:
Not sure if it works, but we might be able to add a `recipe.ino.o.pattern` recipe to `platform.txt`, but not sure if Arduino supports that... :|

Alternatively, we could use a wrapper around the compiler, which checks if we're compiling `*.ino.cpp`, and inject the flag then. But that's very ugly.

noseglasses, 2019-12-04:
Closing this discussion in favor of PR https://github.com/keyboardio/Kaleidoscope/pull/737 

I still regard rethinking the test bench as suggested by @algernon, and possibly separating tests from examples a good idea.

closed 2019-12-04 by noseglasses