keyboardio/Kaleidoscope #750: We should have an include-what-you-use makefile target for maintainers

obra, 2019-12-05:
See https://github.com/keyboardio/Kaleidoscope/pull/737 for more details.

@noseglasses may have input.

noseglasses, 2019-12-06:
I played with this a bit. Although, iwyu is a drop in compiler replacement, it always exits non zero, to signal build systems that no object file was actually build. That's what prevents us from using it just as e.g. 
```
LOCAL_CFLAGS="-x c++ -I$(clang -print-resource-dir)/include" CC=/usr/bin/iwyu CXX=/usr/bin/iwyu ARCH=virtual VERBOSE=1 make
```

The arduino build system has multiple stages and terminates the build pretty soon when errors occur.

That's why it seems impossible to integrated it with the build system.

What I did it when I ran iwyu on the code base was to exact the compiler command line from verbose make output for an arbitrary Kaleidoscope source file. Then I wrote a little shell script that looped over all source files and called iwyu with the extracted compiler command line args. Mainly the include directories must be set correctly, other flags are probably ignored as iwyu does not do a true compile.

Due to build settings changing over time, the iwyu command line in such a skript must be maintained manually. That's why I did not submit it to become part of Kaleidoscope's maintenance tools.

obra, 2019-12-10:
Getting that script into maintainer-tools, even if it's not perfect and not
integrated is still totally better than not. If it's there, there's a
chance that somebody other than you will end up doing some of the work :)
ᐧ

noseglasses, 2019-12-10:
i submitted my IWYU-bash-script in a separate commit of https://github.com/keyboardio/Kaleidoscope/pull/762.

Closing this.

closed 2019-12-10 by noseglasses