keyboardio/Kaleidoscope #752: Some deprecation dates in UPGRADING.md expired

noseglasses, 2019-12-06:
Those are the topics where a code cleanup appears to be due.

* Finer OneShot stickability control
* EEPROMKeymap mode
* keymaps array and KEYMAPS and KEYMAPS_STACKED macros

algernon, 2020-01-06:
OneShot has already been changed, the old APIs were dropped _before_ the deprecation date was added to UPGRADING.md. Oops. (Removed in January, date added in February...)

I'll open a PR to axe the rest.

algernon, 2020-01-06:
There's also a bunch of other expired things, including one I'm not sure about: do we really want to use `kaleidoscope::Runtime` instead of `Kaleidoscope`? (See [here](https://github.com/keyboardio/Kaleidoscope/blob/ba47ccf4e5377e7533a48e966bc8718adb8faa41/docs/UPGRADING.md#classglobal-instance-kaleidoscope_kaleidoscope-renamed-to-kaleidoscoperuntime_kaleidoscoperuntime)). I seem to recall talking about it, but it escapes me where we did that.

(There's also still plenty of code out there using `Kaleidoscope`, and we do not currently emit a warning, which would make it unfair to remove it now, even though the deprecation expired.)

obra, 2020-01-07:
@algernon, I think the end date on Kaleidoscope vs Kaleidoscope.Runtime got misentered as 2019 when it should have been 2020.

I -thought- that the plan was to keep the old Kaleidoscope alias around, but to prefer Kaleidoscope.Runtime inside the core.

PR #787 (2020-01-07): Remove deprecated interfaces and update UPGRADING.md
This removes a number of deprecated interfaces:

- `EEPROMKeymap.setup(n, mode)`, which has been a no-op.
- The old 2d `keymaps` array.

It also updates `docs/UPGRADING.md`, by adding things missing from the ToC, and adding removal dates (2020-03-15) for the following deprecated items:

- The old row/col-based indexing.
- The old hardware API and related symbols.

Fixes #752, and also fixes #751.
closed 2020-01-07 by obra