keyboardio/Kaleidoscope #754: Raise: HID report does not appear to work

algernon, 2019-12-06:
On the Raise, while everything else seems to function correctly, keyboard HID reports aren't being sent. The following code in `loop()` does not produce any HID report:

```c++
  kaleidoscope::hid::pressKey(Key_A);
  kaleidoscope::hid::sendKeyboardReport();
  delay(100);
  kaleidoscope::hid::releaseKey(Key_A);
  kaleidoscope::hid::sendKeyboardReport();
  delay(100);
```

The problem may or may not be in Kaleidoscope, KeyboardioHID would be the obvious source of the problem, but reporting it here in case it isn't KeyboardioHID.

algernon, 2019-12-06:
The reason I'm skeptical about `KeyboardioHID` being the problem, is that when I switch to Dygma's fork of it, which works with their older Kaleidoscope fork, I still don't get any reports.

algernon, 2019-12-06:
And its not just keyboard reports... mouse events aren't received by the host either.

algernon, 2019-12-06:
Oh. This might end up being a HID Descriptor issue! Looks like the first thing in the descriptor is ConsumerControl. That might confuse things.

algernon, 2019-12-06:
Hm, no, that's not it. Descriptor appears to be the same when using the factory firmware too. Darn. I had high hopes there.

noseglasses, 2019-12-06:
It seems the whole firmware is broken. With a fresh clone of the bundle and the M01 stock firmware my keyboard shows no reaction. Maybe thats the virtual HID issue. @algernon, hadn't you fixed that (ifndef guards)?

algernon, 2019-12-06:
I didn't add the ifdef guard yet, because the virtual hid doesn't get compiled for me. (Haven't checked the Model01 yet, mind you).

Meanwhile, I found the Raise issue, [KeyboardioHID/HID.cpp](https://github.com/keyboardio/KeyboardioHID/blob/master/src/HID.cpp#L105): we need to send that whole thing as one packet on SAMD:

```c++
    uint8_t p[64];
    p[0] = id;
    memcpy(&p[1], data, len);
    return USBDevice.send(pluggedEndpoint, p, len+1);
```

noseglasses, 2019-12-06:
Ok, seems that's two different matters. Will submit a PR with a fix for the M01 after lunch.

noseglasses, 2019-12-06:
> because the virtual hid doesn't get compiled for me

But why? The virtual hid is now part of the core repo. That means it is compiled unconditionally. That's why we need the `#ifdef KALEIDOSCOPE_VIRTUAL_BUILD` guards in the first place. 

Experiments show that any `.cpp` file that is added to any place below `src` in the core repo will be compiled.

algernon, 2019-12-06:
No clue, to be honest. I didn't have time and energy to investigate yet. It might get compiled, but it is not interfering with the Raise right now. It might interfere with others, I haven't checked yet.

closed 2019-12-07 by obra