keyboardio/Kaleidoscope #758: Sketch explorer does not support zero-length keymaps

algernon, 2019-12-08:
In [my own sketch](https://git.madhouse-project.org/algernon/Model01-Sketch/src/branch/master/src/keymap.h#L80), I do not have a keymap defined, because it's all in EEPROM (to save PROGMEM space). The sketch explorer doesn't like this much:

```
.../Kaleidoscope/src/kaleidoscope_internal/sketch_exploration/keymap_exploration.h:212:53: error: no matching function for call to 'kaleidoscope::sketch_exploration::StaticKeymap::keymapAdaptor(const Key [0][64])'
```

...and a couple of other, similar errors happen, all stemming from the same root cause:

algernon, 2019-12-08:
I can work this around by removing the empty `KEYMAPS()` from my sketch, and adding this:

```c++
namespace kaleidoscope {
namespace sketch_exploration {
void pluginsExploreSketch() {}
}
}
```

But it would be nicer if I didn't have to do that.

noseglasses, 2019-12-08:
Sorry for this. I will see what I can do to fix this.

noseglasses, 2019-12-08:
We should have an example that matches/tests your configuration (once the bug is fixed).

algernon, 2019-12-08:
It's okay, my case is a very, very edge case. There are a few other related things which could be improved in this case (`keyFromKeymap()` for example throws an out-of-bounds warning in my case, too).

algernon, 2019-12-09:
Fixed by #759.

closed 2019-12-09 by algernon