keyboardio/Kaleidoscope #766: Symlinking virtual/boards.txt to avr/boards.txt does not work on Windows

noseglasses, 2019-12-11:
On Windows/MinGW git replaces symlinks with files that just contain the relative path to the file the symlink points to. This is due to the bad support for symlinks that on many Windows versions require administrator access for being created. I heard rumors that this has been fixed recently but git seems to stay on the safe side (or is my git version probably outdated?).

This is problematic as it prevents us from using the same `boards.txt` file for both the virtual and the physical hardware.

noseglasses, 2019-12-11:
Here's some information about the topic: https://github.com/git-for-windows/git/wiki/Symbolic-Links

alfalfasprossen, 2020-05-16:
Configuring git to create symlinks on windows actually works with relatively new git versions, as long as you run it with admin privileges. 

In an existing repository you can enable it with `git config --local core.symlinks true` and then do a hard reset to remove the files that were created instead of the symlinks before.

I think the best solution in this specific case would be to add a windows-specific cloning instruction to the README that uses `-c core.symlinks=true`, as the site noseglasses linked to explains, which enables symlinks in the new repository only.

obra, 2020-05-18:
Thanks for the magic incantation, Johannes! I've updated the READMEs for
both repositories with the relevant notes.

commit 97d4f4e9 (2020-05-18): Add a note about symlinks on Windows, thanks to Johannes Becker.

Fixes #766
closed 2020-05-18 by obra