keyboardio/Kaleidoscope #773: Warnings with arduino 1.8.10

algernon, 2019-12-13:
When compiling current master with Arduino 1.8.10, which uses gcc 7 instead of gcc 5.4, there are a number of warnings for pretty much every example:

```
BOARD_HARDWARE_PATH="/home/algernon/Arduino/hardware" /home/algernon/Arduino/hardware/keyboardio/avr/libraries/Kaleidoscope/bin//kaleidoscope-builder build-all
Building ./Model01 0.0.0 into /tmp/kaleidoscope-algernon/sketch/3807972-Model01.ino/output...
In file included from /home/algernon/src/keyboardio/Kaleidoscope-Bundle-Keyboardio/avr/libraries/Kaleidoscope/src/kaleidoscope/driver/keyscanner/Base.h:22:0,
                 from /home/algernon/src/keyboardio/Kaleidoscope-Bundle-Keyboardio/avr/libraries/Kaleidoscope/src/kaleidoscope/driver/keyscanner/None.h:22,
                 from /home/algernon/src/keyboardio/Kaleidoscope-Bundle-Keyboardio/avr/libraries/Kaleidoscope/src/kaleidoscope/device/Base.h:28,
                 from /home/algernon/src/keyboardio/Kaleidoscope-Bundle-Keyboardio/avr/libraries/Kaleidoscope/src/kaleidoscope/device/ATmega32U4Keyboard.h:23,
                 from /home/algernon/src/keyboardio/Kaleidoscope-Bundle-Keyboardio/avr/libraries/Kaleidoscope/src/kaleidoscope/device/keyboardio/Model01.h:32,
                 from /home/algernon/src/keyboardio/Kaleidoscope-Bundle-Keyboardio/avr/libraries/Kaleidoscope/src/Kaleidoscope-Hardware-Model01.h:20,
                 from /home/algernon/src/keyboardio/Kaleidoscope-Bundle-Keyboardio/avr/libraries/Kaleidoscope/src/kaleidoscope/device/device.h:22,
                 from /home/algernon/src/keyboardio/Kaleidoscope-Bundle-Keyboardio/avr/libraries/Kaleidoscope/src/kaleidoscope_internal/device.h:19,
                 from /home/algernon/src/keyboardio/Kaleidoscope-Bundle-Keyboardio/avr/libraries/Kaleidoscope/src/kaleidoscope/Runtime.h:19,
                 from /home/algernon/src/keyboardio/Kaleidoscope-Bundle-Keyboardio/avr/libraries/Kaleidoscope/src/kaleidoscope/layers.cpp:17:
/home/algernon/src/keyboardio/Kaleidoscope-Bundle-Keyboardio/avr/libraries/Kaleidoscope/src/kaleidoscope/key_defs.h: In static member function 'static kaleidoscope::Key kaleidoscope::Layer_::getKeyFromPROGMEM(uint8_t, KeyAddr)':
/home/algernon/src/keyboardio/Kaleidoscope-Bundle-Keyboardio/avr/libraries/Kaleidoscope/src/kaleidoscope/key_defs.h:61:20: warning: array subscript is above array bounds [-Warray-bounds]
     return keyCode.value_;
                    ^~~~~~
/home/algernon/src/keyboardio/Kaleidoscope-Bundle-Keyboardio/avr/libraries/Kaleidoscope/src/kaleidoscope/key_defs.h:54:18: warning: array subscript is above array bounds [-Warray-bounds]
     return flags.value_;
                  ^~~~~~
- Size: firmware/Model01/Model01-0.0.0.elf
  - Program:   17774 bytes (62.0% Full)
  - Data:       1149 bytes (44.9% Full)
```

This only seems to happen with 1.8.10, I don't see the warnings with 1.8.9 (but I don't quite trust my build environment at this point).

noseglasses, 2019-12-13:
That's probably not related to what happens in the `Key` class but to the calling scope. In `Key` there's no array access.

Maybe it's because of the forward declaration https://github.com/keyboardio/Kaleidoscope/blob/ed667bc97c721e568f073fcff1aa74ecf83f100d/src/kaleidoscope/keymaps.h#L23

noseglasses, 2019-12-13:
In `layers.cpp` only this forward declaration is visible. This could be solved by moving the key-from-progmem implementation to the new sketch footer where the array dimensions are known.

Anyway, it seems that the compiler warns unjustified.

obra, 2020-02-14:
Fixed by d5c6bf28dae8e6e3c86a7c73c892b13ce84d48dd

closed 2020-02-14 by obra