keyboardio/Kaleidoscope #788: Clarify some deprecations

algernon, 2020-01-07:
We currently have an entry in `UPGRADING.md` about `Kaleidoscope` and `kaleidoscope::Runtime`, with an end-date in 2019. That date is probably wrong, and should've been 2020. But more important is clarifying what is deprecated and where. See [here](https://github.com/keyboardio/Kaleidoscope/issues/752#issuecomment-571429125) for some comments, but the gist of it is this question: Do we want to remove the global `Kaleidoscope` symbol, or keep it, but prefer using `kaleidoscope::Runtime` within the core firmware?

obra, 2020-01-07:
I believe it to be the latter, but would love someone to pull out the
discussion from the pull request that added Runtime
ᐧ

noseglasses, 2020-01-07:
> but would love someone to pull out the
discussion from the pull request that added Runtime

https://github.com/keyboardio/Kaleidoscope/pull/737#issuecomment-562672115

algernon, 2020-01-07:
Lovely, thank you! I'll update UPGRADING.md a bit later.

PR #789 (2020-01-07): docs/UPGRADING.md: Clarify the kaleidoscope::Runtime parts
Fixes #788.
commit e75676de (2020-01-07): docs/UPGRADING.md: Clarify the kaleidoscope::Runtime parts

Fixes #788.
closed 2020-01-07 by algernon