keyboardio/Kaleidoscope #794: missing rebootBootloader method

technomancy, 2020-01-09:
The old Atreus sketch (not Atreus2) fails to compile as it calls `Kaleidoscope.rebootBootloader();` in order to reset the device, but this apparently was removed:

```
/home/phil/src/Kaleidoscope/examples/Devices/Technomancy/Atreus/Atreus.ino: In function 'const macro_t* macroAction(uint8_t, uint8_t)':
/home/phil/src/Kaleidoscope/examples/Devices/Technomancy/Atreus/Atreus.ino:86:18: error: 'class kaleidoscope::Kaleidoscope_' has no member named 'rebootBootloader'
     Kaleidoscope.rebootBootloader();
```

Or maybe the way it's supposed to be invoked got changed?

obra, 2020-01-09:
We've got a different way to get into the bootloader that's not so sketch
dependent. The host tickles the keyboard with a 9600 bps serial connection.
That starts the bootloader, whether or not the sketch knows about it. Then,
for production units, we check to see if the escape key is held. If not, no
reprogramming. (This prevents malicious flashing)
ᐧ

algernon, 2020-01-09:
Are you sure you're compiling against a recent Kaleidoscope? Because `Kaleidoscope.rebootBootloader()` is the _new_ interface, not the old one (the old one was `KeyboardHardware.resetDevice()`).

If you have a Kaleidoscope version in - say - `~/Arduino/hardware`, or one installed via the IDE's board manager, that will usually take precedence over the one in the current tree, and thus the compile will fail.

technomancy, 2020-01-09:
Thanks; I forgot there were two separate repositories; for some reason I thought that would be handled for me using submodules.

When I update that repo I get a different error, and now I can't build `Atreus` or `Atreus2`:

```
$ make
BOARD_HARDWARE_PATH="/home/phil/src/Arduino/hardware" /home/phil/src/Arduino/hardware/keyboardio/avr/libraries/Kaleidoscope/bin//kaleidoscope-builder build-all
Building ./Atreus 0.0.0 into /tmp/kaleidoscope-phil/sketch/2228452-Atreus.ino/output...
cat: /home/phil/src/Arduino/hardware/keyboardio/avr/libraries/Kaleidoscope/src/kaleidoscope_internal/sketch_preprocessing/sketch_header.h: No such file or directory
cat: /home/phil/src/Arduino/hardware/keyboardio/avr/libraries/Kaleidoscope/src/kaleidoscope_internal/sketch_preprocessing/sketch_footer.h: No such file or directory
exit status 1
/home/phil/src/Arduino/hardware/keyboardio/avr/build-tools/makefiles//rules.mk:80: recipe for target 'build-all' failed
make: *** [build-all] Error 1
```

closed 2020-01-09 by technomancy