keyboardio/Kaleidoscope #795: Qukeys: SpaceCadet Emulation

mokkun, 2020-01-09:
I'm trying to use Qukeys as a replacement for Space Cadet as described in the [Qukeys documentation](https://github.com/keyboardio/Kaleidoscope/blob/ba47ccf4e5377e7533a48e966bc8718adb8faa41/docs/plugins/Qukeys.md#spacecadet-emulation).

I've used the [example provided in the Kaleidoscope repository](https://github.com/keyboardio/Kaleidoscope/blob/master/examples/Keystrokes/Qukeys/Qukeys.ino) as a sandbox. The example defines some mappings in the following format:

```ccp
kaleidoscope::plugin::Qukey(0, KeyAddr(2, 1), Key_LeftGui),      // A/cmd
kaleidoscope::plugin::Qukey(0, KeyAddr(2, 2), Key_LeftAlt),      // S/alt
...
```

These work just fine. What I'm trying to achieve is to print `A` when `cmd` is pressed (reverse of the first mapping).

So I went ahead and changed the first mapping to:

```cpp
kaleidoscope::plugin::Qukey(0, KeyAddr(4, 2), Key_A),      // cmd/A
```

Assuming `Qukeys` is enabled (toggle key was not pressed) and that the current layout is `0` (default), `KeyAddr(4, 2)` should point to `Key_LeftGui`.
But when I tap the key, it does not print `A`, it functions as a normal `Gui` key (`CMD` on Mac). Long pressing it has the same effect.

The docs [state the following](https://github.com/keyboardio/Kaleidoscope/blob/ba47ccf4e5377e7533a48e966bc8718adb8faa41/docs/plugins/Qukeys.md#spacecadet-emulation): 

> It is possible to define a Qukey on a key with a primary value that is a modifier. In this case, the qukey is treated specially, and the primary value is used when the key is held, rather than the alternate value.

So I'm not sure what I'm doing wrong and how to achieve the desired behavior.
I've tried several different ways and could not find any documentation on how to make a `cmd/A` mapping.

gedankenexperimenter, 2020-01-09:
I think you've got the wrong coordinates. `KeyAddr(4, 2)` seems like it should be the coordinates for that key, but there is no row `4`; instead, the thumb keys are _columns_ 7 & 8. I can never remember the order, nor the location of the diagram showing the correct key coordinates.

gedankenexperimenter, 2020-01-09:
I should update the Model01 example sketch with a few SpaceCadet-emulation keys to make this easier.

mokkun, 2020-01-10:
Thanks! You were right; I got the coordinates all wrong. 🤦‍♂ 
Here is the correct way of doing it:

```ccp
kaleidoscope::plugin::Qukey(0, KeyAddr(3, 7), Key_LeftParen), // LSHIFT / (
kaleidoscope::plugin::Qukey(0, KeyAddr(3, 8), Key_RightParen), // RSHIFT / )
```

It would be great to have an example of that. For a while, I thought it was not possible, but after reading the source code, I found that there was support for it and got confused.

Now rereading the docs, I notice that we do have a reference to a file that contains the key mappings:

> For the Keyboardio Model 01, key coordinates refer to [this header file](https://github.com/keyboardio/Kaleidoscope-Hardware-Model01/blob/f469015346535cb864a340bf8eb317d268943248/src/Kaleidoscope-Hardware-Model01.h#L267-L279).

I remember clicking on it, but thought that it was the same as the normal mapping we have in the [example file](https://github.com/keyboardio/Kaleidoscope/blob/master/examples/Keystrokes/Qukeys/Qukeys.ino). Silly of me, of course, didn't read it. If I had, I would have known that `r3c7` is a reference to `keyIndex(3, 7);` which is the exact key coordinate I was looking for. 🤦‍♂ 

That said, I also found some [documentation in the `Virtual.cpp`](https://github.com/keyboardio/Kaleidoscope/blob/e425d8ac5a65fbc86179d738991f90cbe879952a/src/kaleidoscope/device/virtual/Virtual.cpp#L85) file that was clearer to me at first sight.
Maybe we could have a link to that or document the key mapping in a similar fashion to make it easier for newbies like me to understand it. 😄 

On a side note, since we are talking documentation, I found another small issue in the docs for [The Wildcard Layer](https://github.com/keyboardio/Kaleidoscope/blob/master/docs/plugins/Qukeys.md#the-wildcard-layer):

> There is a special value (`Qukeys::layer_wildcard`) that can be used in place of the layer number in the definition of a `Qukey`. 

I've tried to use `Qukeys::layer_wildcard`, but I got compile errors. Changing it to `Qukeys.layer_wildcard` works. I'm not super familiar with C++, but since `layer_wildcard` [is a `constexpr`](https://github.com/keyboardio/Kaleidoscope/blob/e425d8ac5a65fbc86179d738991f90cbe879952a/src/kaleidoscope/plugin/Qukeys.h#L148), maybe it is not possible to access it with `::`?

> I should update the Model01 example sketch with a few SpaceCadet-emulation keys to make this easier.

I can make a PR during the weekend with these updates to the docs if you don't mind. 👍 

Thanks for the help and the great work with Qukeys, it works like a charm. 🎉

closed 2020-01-10 by mokkun