keyboardio/Kaleidoscope #798: Caps lock not in sync wut

JonathanTroyer, 2020-01-14:


obra, 2020-01-14:
Is the indicator always 100% wrong? Or is it inconsistent? 

Can you tell us a bit more about what you're seeing?

algernon, 2020-01-14:
I've heard about similar problems, with the other Lock keys too, but wasn't able to consistently reproduce it yet. What I gathered, is that things work initially, until the lock key is toggled off after being toggled on. At that point, Kaleidoscope will think the lock LED is on, even though it isn't.

On the other hand, this might be a case of MacOS treating separate keyboards separately, so turning capslock on on the Model01 will not affect the built-in keyboard on a laptop. Thus, no indicator lights, either.

JonathanTroyer, 2020-01-14:
Here's a state machine that I believe covers it all. Default refers to capslock off and LED off, cl is shorthand for capslock, m1 shorthand for Model01, bi shorthand for built-in.

- default -cl m1-> no LED, cl on for m1 and bi
  - -cl m1-> default
  - -cl bi-> LED on, cl on for m1 and bi
    - -cl m1-> LED on, cl off for m1 and on for bi
    - -cl m1-> parent state
    - -cl bi-> LED off, cl on for m1 and off for bi
      - -cl m1-> default
    - -cl bi-> parent state