keyboardio/Kaleidoscope #803: MagicCombo masking original keypress

dk-88, 2020-01-19:
Currently the MagicCombo plugin does not mask the key presses used in a MagicCombo. According to Algernon this is a design choice, but it would be nice to have the option to mask the original key presse.

Because the oringal key presses are not masked s using a magic combo also types the letters used in the combo. For example mapping home to the J and K buttons results in pressing the keys JK and Home while the desirable result would just be pressing the Home key.

Here a link to the original discussion on the [Forum](https://community.keyboard.io/t/mapping-multiple-normal-keys-to-one-chord-key/1584/8).

algernon, 2022-09-27:
It's been a while I last had an attempt at fixing this, and we have some cool toys since, like the event queue. That might be something that lets us avoid sending the combo participants until we know we want to, and replay them in order in that case.

gedankenexperimenter, 2022-10-04:
Deciding on the exact parameters for this will be tricky.  Yes, MagicCombo could become a plugin that queues events if they might match a combo in order to suppress them, but if it's used in this way, it could cause a lot of unintended output when users roll over from one key to another.  For example, if someone defines a MagicCombo that turns <kbd>A</kbd> + <kbd>S</kbd> into `home`, it would be very easy to trigger the combo unintentionally when trying to send two separate taps.  We don't notice these problems now because people don't use MagicCombo with pairs of keys on the base layer—because it doesn't mask the normal output of those keys.

Conditionally masking the output of keys can be done, but it opens a huge can of worms, including timeouts, overlap percentages, and other tricks that we'll need to use to guess user intent from ambiguous input.  The plugin would rapidly take on a similar complexity to Qukeys, and still couldn't guarantee that it won't produce unintended output.

gedankenexperimenter, 2022-10-05:
I think the original request could be satisfied by the implementation of a new plugin for "adaptive keys" (#1266).  That one is likely to be restricted to bigrams only (because variable-length sequences are going to be problematic to store in EEPROM), but that may prove sufficient, and wouldn't end up making MagicCombo unwieldy for older devices with small MCUs.

obra, 2022-10-05:
Indeed, I think "Adaptive Keys" probably wants to be its own plugin if
someone wants to implement it.

gedankenexperimenter, 2022-10-05:
The "Adaptive Keys" plugin design I've got in mind would help with this even if someone wants to execute more complex/arbitrary MagicCombo code by providing a way to suppress the keys involved in the MagicCombo, without requiring MagicCombo to suppress those keys itself.