keyboardio/Kaleidoscope #804: Modifier-like layer keys

algernon, 2020-01-22:
Suggested in Dygmalab/Bazecor#34, the idea is to have a `LockLayer` variant that by itself, will do nothing, will not lock to any other layer. But pressed together with a number key, it will switch to that layer. This is - I believe - most useful if it does a layer move, instead of layer activation, so that we can easily move to any layer.

I'd call this key `Key_LayerSelect`. When tapped alone, it's a no-op. When held alone, no-op. When held, any number key (0->9) will lock to the appropriate layer (and the number key will be masked until release). Holding a number key and _then_ pressing `Key_LayerSelect` should not switch layers though. Holding any other key should not alter the above described behaviour. That is, `LayerSelect + A + 3` should result in moving to layer three, and `A` being input.

noseglasses, 2020-01-22:
I like that idea.

When I read your description it occurred to me that people might want to use other keys than (0-9) to select specific layers. Keys that in another context might have a different meaning.

What do you think about a mapping from key codes to layer ids? Or a mapping from KeyAddr to layer ids? Both make sense to me. 0-9 isn't enough anyway for any layers above 9.

algernon, 2020-01-22:
Hm. I can make the method that decides which layer to switch to (if any) overrideable, and then the limits are endless.

noseglasses, 2020-01-22:
Yeah, something like
```cpp
...weak...
uint8_t layerSelect(Key key, KeyAddr addr) {...}
```

jtroo, 2020-02-13:
Could this be replicated by existing functionality using an intermediate layer? The hold button does ShiftToLayer() and the intermediate layer has buttons mapped to LockLayer. Not sure if letting go of ShiftToLayer will go back to the original layer though.

gedankenexperimenter, 2020-10-04:
Since #805 was closed, I think this should probably also be closed.

obra, 2020-10-05:
I'm closing this because it's possible to do with a custom layer, though if someone has a "real' use case that needs this, I'm happy to see it reopened.

closed 2020-10-05 by obra