keyboardio/Kaleidoscope #809: the device is flashed, sketch does not apply

hackmalife, 2020-02-03:
A while ago, I posted an issue about problem that k<sub>key</sub> is not working, so I sent it back to Jesse, he fixed the problem, and returned it back to me. But, now I cannot flash the device. When I `make flash`, the red light glows through keys and seemed to be successfully flashed but the sketch is not applied to the device. (The log remains exactly same as [one](https://github.com/keyboardio/Kaleidoscope/issues/724#issue-524802407) I originally reported) I tried to do it with Arduino IDE (with version 1.8.10), it does not report any error/warning when I try to flash but turns out that no keymapping change is applied to the device. Finally, I tried a single swap of keys using Chrysalis, and it did work. But I still prefer to do it on command line rather than GUI.

By the way, I was in phase of migrating to Arch Linux anyways. So, if it works in new OS, it was fine for me. 
When I `make flash` in Linux, the log seems to be success (red glowing keys and stuff), but again, no keymapping change happened. The log is like this:

    BOARD_HARDWARE_PATH="/home/jisoo/Arduino/hardware" /home/jisoo/Arduino/hardware/keyboardio/avr/libraries/Kaleidoscope/bin//kaleidoscope-builder flash
    Building ./Model01-Firmware 0.0.0-gv1.22-75-g03be into /tmp/kaleidoscope-jisoo/sketch/2229060-Model01-Firmware.ino/output...
    - Size: firmware/Model01-Firmware/Model01-Firmware-0.0.0-gv1.22-75-g03be.elf
      - Program:   25794 bytes (90.0% Full)
      - Data:       1381 bytes (53.9% Full)
    
    To update your keyboard's firmware, hold down the 'Prog' key on your keyboard,
    and then press 'Enter'.
    
    When the 'Prog' key glows red, you can release it.
    
    
    Connecting to programmer: .
    Found programmer: Id = "CATERIN"; type = S
        Software Version = 1.0; No Hardware Version given.
    Programmer supports auto addr increment.
    Programmer supports buffered memory access with buffersize=128 bytes.
    
    Programmer supports the following devices:
        Device code: 0x44

The sketch I use in this step is Vanilla one, so that I am trying to restore it back to default just to see if it works.

When I verify the sketch from Arduino IDE, it fails compiling producing the following error messages:

    Arduino: 1.8.9 (Linux), Board: "Keyboardio Model 01"
    The current selected board needs the core 'arduino:arduino' that is not installed.
    The current selected board needs the core 'arduino:arduino' that is not installed.
    WARNING: library Kaleidoscope-HIDAdaptor-KeyboardioHID claims to run on (avr) architecture(s) and may be incompatible with your current board which runs on (virtual) architecture(s).
    fork/exec /ar: no such file or directory
    Error compiling for board Keyboardio Model 01.
    
    This report would have more information with
    "Show verbose output during compilation"
    option enabled in File -> Preferences.

The last option I can think of is to factory reset device, but I am not sure how to do it. Any help is greatly appreciated, Thanks.

obra, 2020-02-04:
> A while ago, I posted an issue about problem that kkey is not working, so

It occurs to me that you may be running into issues with having both
firmware-defined layouts and Chrysalis-defined layouts in EEPROM. We put in
a lot of work to try to make sure that a simple firmware update wouldn't
destroy your custom layouts.

One way to check this is to try removing all but one LED mode from your
custom sketch and flashing. If I'm right, you'll have the "wrong" key
layouts and the correct (limited) LED modes.
ᐧ

hackmalife, 2020-02-04:
I am afraid I am not crystal clear about your suggestion. Removing all but one LED mode from your custom sketech means clearing all layers except Layer 0 (default) in Chrysalis app?  Or fix .ino file (sketch) so that it has only one layer and `make flash`?

hackmalife, 2020-02-05:
@obra I ripped out entire layer in Chrysalis, now I can't flash at all since my Prog key is completely gone! I should have been more cautious.... How can I restore Prog key back??

noseglasses, 2020-02-05:
You can try holding the PROG key while plugging in the keyboard. That will always work even with a completely messed up sketch.

algernon, 2020-02-05:
Also, the `Prog` key is tied to the location, not what is mapped to it. So even if you set it to something else, it will still work as a prog key (it will also do whatever you mapped it to, mind you). The easiest is to set it to "Blank", as that'll do nothing. But since the functionality of the key is tied to its location, it will still work as a `Prog` key.

Also, what @obra meant is that this issue usually happens when you have a keymap stored in the keymap's EEPROM, which will override the sketch by default. The easiest thing to do is reset the EEPROM to factory settings, from within Chrysalis (Keyboard Preferences -> Advanced). That will clear the EEPROM keymap, and then the in-sketch one will take effect.

hackmalife, 2020-02-05:
I finally get this my device flashed by command line. Thank you
I found the resetting to factory in Chrysalis, and enabled 'show hardcoded layers' and disabled 'use custom layers only' options. Then, the hidden layer which I've been flashing showed up in the layer 1 in layout settings. I guess it is a side effect that I am not able to fix these layers in Chrysalis, it is displayed with the lock icons and prevent any customization. Anyways, thanks for the help!

closed 2020-02-05 by hackmalife

algernon, 2020-02-05:
FYI, once you cleared the EEPROM, the next time you view the layout in Chrysalis, it will automatically copy the layout from the sketch to EEPROM, and if you re-enable "Use custom layers only" (and disable "show hardcoded layers"), you'll be able to edit your new layout.

The in-sketch layout is not editable with Chrysalis, because it is in a read-only part of the keyboard's brain, which can only be written to during flashing. Chrysalis can only edit things in EEPROM - and that's why it copies the in-sketch layout to EEPROM, if EEPROM is empty.

fredcallaway, 2021-03-18:
For documentation purposes, I ran into a similar problem with a used (second) Model 01, where flashing the keyboard with my pre-existing layout had no effect. I fixed the problem by simply resetting to factory settings in Chrysalis. Presumably, the previous owner had set some EEPROM settings, which had to be cleared for the hardware settings to take effect.