keyboardio/Kaleidoscope #810: Flashing suddenly stops working

hackmalife, 2020-02-08:
When I try to flash, this happend, I have no clue what causes it:
`
BOARD_HARDWARE_PATH="/home/jisoo/Arduino/hardware" /home/jisoo/Arduino/hardware/keyboardio/avr/libraries/Kaleidoscope/bin//kaleidoscope-builder flash
Building ./Model01-Firmware 0.0.0-gv1.22-75-g03be-dirty into /tmp/kaleidoscope-jisoo/sketch/2622420-Model01-Firmware.ino/output...
- Size: firmware/Model01-Firmware/Model01-Firmware-0.0.0-gv1.22-75-g03be-dirty.elf
  - Program:   28456 bytes (99.2% Full)
  - Data:       1445 bytes (56.4% Full)

To update your keyboard's firmware, hold down the 'Prog' key on your keyboard,
and then press 'Enter'.

When the 'Prog' key glows red, you can release it.

ls: cannot access '/dev/serial/by-id/': No such file or directory
Looking for USB device with vid=1209 and pid=2300
ERROR: I couldn't find a USB device matching the keyboard's USB Vendor and Device IDs
Unable to detect a keyboard in bootloader mode. You may need to hold the 'Prog' key or hit a reset button
`

obra, 2020-02-08:
Hi Jisoo,

It sounds like your system may not have some of devfs.  Just to make sure
the computer is seeing the keyboard, can you 'lsusb' immediately after the
failure?
ᐧ

hackmalife, 2020-02-09:
I am not sure what has happened, but rebooting the system did the trick..
sorry for the noise.
Thanks.

closed 2020-02-09 by hackmalife

obra, 2020-02-09:
No worries. It’s always better for me when the bug goes away.