keyboardio/Kaleidoscope #811: Idea: option to make Qukeys LT() and MT() similar to QMK version

jtroo, 2020-02-13:
Recently got myself a used Model01 and I'm loving it! I tried out Kaleidoscope for a while and mapped my Ergodox keymapping mostly to it, but due to one issue I've since changed it to QMK.

Maybe the answer is just "keep using QMK" but I wanted to give Kaleidoscope a try.

With that said, my issue is a design of Qukeys which conflicts with my use case. My configuration has my home row keys on the left hand using LT() and MT(). In QMK, these functions will not activate the layer / modifier until `TAPPING_TERM` has passed. However, in Kaleidoscope the layer / modifier will apply even if timeout has not passed as long as the second keypress is released before the first. Here's an example:

Setup:
The physical A key is mapped to `MT(Meh, A)`

Physical key sequence:
1. A down
2. J down
3. J up
4. A up

For Kaleidoscope when this sequence is put in, the key combination sent is always `MEH(A)`, irrespective of how quickly the sequence is done.

In QMK, this sequence will send `AJ` if `A up` occurs before `TAPPING_TERM` has passed since `A down`, and `MEH(A)` otherwise. This is the behaviour I'm looking for. My typing patterns are such that this type of key sequence occurs frequently.

Please feel free to close this issue if you don't think it provides value.

gedankenexperimenter, 2020-03-01:
[Sorry for taking so long to reply; I haven't been doing any programming for a while, and haven't been keeping up with issues on GitHub.]
This is the first time I've heard from someone who wants this behaviour. It's a bit surprising to me that, while typing, you hold home row keys while pressing and releasing other keys, and don't want the modifiers, but if that's the way you type, perhaps there are others who share that habit. It can be done, of course, and it probably won't make the compiled firmware much bigger than it already is. I'll need to review the code to determine if it will be a simple as I'm guessing (I don't recall if the Kaleidoscope version of Qukeys queues both press and release events; if so it should be fairly simple).

gedankenexperimenter, 2020-03-01:
After quickly reviewing the code, I'm pretty sure that two conditionals added on lines 208 & 231, with an appropriate boolean and a configuration method will do the trick.

jtroo, 2020-04-02:
Closing due to lack of commitment from myself to implementing/testing and to reduce number of open issues. I may come back to it at some point.

closed 2020-04-02 by jtroo