keyboardio/Kaleidoscope #812: Atreus2 firmware leaves fn key stuck

technomancy, 2020-02-13:
Ever since updating to a more recent build of Kaleidoscope (previously I was on a build from the summer that worked fine) I've observed that frequently the `fn` key is not releasing reliably. The current layer sometimes will not change back to the base when releasing, and it's locked in place, requiring you to tap the fn key in order to get back to the base layer. Occasionally tapping fn once does not work, and an additional tap is needed.

I could try doing a bisect to try to hunt down the cause, but the time range is pretty huge so if anyone knows a likely culprit off the top of their head that'd be helpful.

indirect, 2020-02-13:
I'm also experiencing this. I would love suggestions about how to improve, and I'm thinking about switching it to a oneshot just to work around the way it doesn't seem to toggle successfully. 😬

algernon, 2020-02-17:
Just to be clear: the problem is getting _back_ to the base layer? Does switching to any other layer work fine?

indirect, 2020-02-17:
In the default sketch, there are only two layers, qwerty and fn. I’ve experienced trouble both getting into and getting out of fn, but I haven’t added any other layers yet to see if it’s layer specific.

algernon, 2020-02-17:
Interesting. Then I may have reproduced something else!

algernon, 2020-02-18:
Hm, in the [default sketch](https://github.com/keyboardio/Kaleidoscope/blob/master/examples/Devices/Technomancy/Atreus2/Atreus2.ino#L50-L52) there are three layers, `QWERTY`, `FUN` (Fn) and `UPPER`. `Fun + Esc` should get you to `UPPER`, where `Esc` and `Fun` both should get us back to the base QWERTY layer.

But, I think I managed to reproduce _something_!

If I press and hold `Fun` and type a bit on layer 1, then release `Fun` and quickly tap it again, I get stuck at layer 1 until another `Fun` tap - even though it is not a oneshot key, so shouldn't have the double-tap functionality!

algernon, 2020-02-18:
The problem is likely not in `layers.cpp`, because the problem happens without the `moveToLayer` stuff, and we didn't have anything that changed behaviour in there before that since quite a while. Curious!

algernon, 2020-02-19:
Hmm.... adding some logging to `Layer::eventHandler()`, it seems that key releases do not always reach it at all. Which points to debouncing issues. But dropping the debounce from the default 3 to 0, effectively disabling it, we still get cases where the release doesn't reach `Layer::eventHandler()`. Perhaps we're handling the key before that, for some odd reason?

algernon, 2020-02-19:
Nope, it's not that we handle it elsewhere, `handleKeyswitchEvent()` doesn't see it either.

algernon, 2020-02-19:
It's not even an invalid key address, as placing logging at the very top of `handleKeyswitchEvent()` doesn't see the toggle-off event. So I guess we're looking at the ATmegaKeyScanner code now.

algernon, 2020-02-19:
The interesting thing is, that `pressedKeyswitchCount()` will report 0, but we do not call `handleKeyswitchEvent()` for some reason, while there is definitely a change in key state.

algernon, 2020-02-19:
With `pressedKeyswitchCount()` reporting 0, we know that reading the PINs is fine, so it's not like we're reading when the PIN isn't hot yet. 

But just to be sure, I tried `mcu.disableJTAG()` and `mcu.disableClockDivision()` - no effect. I tried disabling optimizations and adding more NOPs, still reproducible.

If it worked during the summer, looking at the logs, the most likely culprit might be the new device API. But I don't see how that'd result in this....

algernon, 2020-02-19:
_Increasing_ debouncing doesn't seem to have any effect either. This is starting to get very weird.

obra, 2020-02-19:
it may be time to bisect to find the cause

algernon, 2020-02-19:
Yeah, doing that. There's.... a lot to bisect.

algernon, 2020-02-24:
Turns out my initial fears that this might be related to the device APIs or a compiler change were unfounded! Bisecting revealed that the culprit is 5b409b1da5a287b05a4bb2e21381065316e10bd2.

algernon, 2020-02-24:
First attempt at a fix failed: I noticed that `debounce_matrix_` is _not_ `volatile`, while it is potentially modified during the scan, which happens during the interrupt. Simply marking it volatile sadly did not fix the issue.

Neither did decreasing the interrupt frequency.

closed 2020-02-24 by obra

technomancy, 2020-02-25:
Whew; what a relief! This is like night and day; nice work.

PR #815 (2020-02-24): driver::keyscanner::ATmega: Revert scanning during the interrupt
Instead of scanning during the interrupt, do so in the main context, and only use the interrupt for signaling that we need to scan. This resolves a problem where scanning took too long, and we ended up missing events.

Fixes #812.
commit dd686ee1 (2020-02-24): driver::keyscanner::ATmega: Revert scanning during the interrupt

Instead of scanning during the interrupt, do so in the main context, and only
use the interrupt for signaling that we need to scan. This resolves a problem
where scanning took too long, and we ended up missing events.

Fixes #812.
commit 059d4723 (2020-02-25): Guard some places against interrupts

When checking or updating the key state, we need to do so with interrupts
disabled for the _whole_ duration of our operation, otherwise we may end up in
an inconsistent state. This means that we disable interrupts for the whole of
`actOnMatrixScan` (for hardware that uses interrupts for key scanning, anyway),
and whenever we use `pressedKeyswitchCount`, `previousPressedKeyswitchCount`,
`isKeyswitchPressed`, or `wasKeyswitchPressed`. These latter ones can't be
guarded within the functions, because we need to guard not just a single call to
them, but all of  [... cut]