keyboardio/Kaleidoscope #813: Flashing trouble with Keyboardio Atreus

technomancy, 2020-02-23:
Running `make flash` in the `Atreus2` sketch dir you get this:

```
$ make flash
BOARD_HARDWARE_PATH="/home/phil/src/Arduino/hardware" /home/phil/src/Arduino/hardware/keyboardio/avr/libraries/Kaleidoscope/bin//kaleidoscope-builder flash
Building ./Atreus2 0.0.0 into /tmp/kaleidoscope-phil/sketch/1992966-Atreus2.ino/output...
- Size: firmware/Atreus2/Atreus2-0.0.0.elf
  - Program:   20958 bytes (73.1% Full)
  - Data:        937 bytes (36.6% Full)

To update your keyboard's firmware, hold down the 'Prog' key on your keyboard,
and then press 'Enter'.

When the 'Prog' key glows red, you can release it.

Looking for USB device with vid=1209 and pid=2303

Examining usb-Keyboardio_Atreus_Bootloader_atreus-if00

  ID_MODEL_ID 2302 != 2303
ERROR: I couldn't find a USB device matching the keyboard's USB Vendor and Device IDs

I couldn't autodetect the keyboard's serial port.

If you see this message and your keyboard is connected to your computer,
it may mean that our serial port detection logic is buggy or incomplete.
In that case, please report this issue at:
	 https://github.com/keyboardio/Kaleidoscope
/home/phil/src/Arduino/hardware/keyboardio/avr/build-tools/makefiles//rules.mk:80: recipe for target 'flash' failed
make: *** [flash] Error 1
```

This same message appears whether I use the recessed reset button on the bottom of the keyboard of whether I hold down the Esc key to trigger the bootloader. It appears to be looking for 2303 (the actual keyboard PID) but finding 2302 (the bootloader PID). I'm uncertain which should be correct here.

Fixing the first message which refers to the non-existent "Prog" key will require changes to the `kaleidoscope-builder` shell script. I expect the right thing to do would be to tell the user to hold down Esc instead of Prog, but do we have an existing way in that script to distinguish between the two boards?

obra, 2020-02-23:
The Atreus does indeed have a distinct pair of USB PIDs (Product Ids)  2302 for the bootloader and 2303 for the runtime.  (The model 01 uses 2300 and 2301.) 

With Kaleidoscope, if you're hitting the reset button, you're going to be in the bootloader already, so "make flash-from-bootloader", though we could clearly make kaleidoscope-builder smarter if we see that something's already in the bootloader.

However, I have no idea how you're ending up in the bootloader if you hold down Esc while the keyboard is already plugged in.  Are you, perhaps, holding in Esc while plugging in the keyboard?

The other half of your issue is that diagnostic message, which we -absolutely- need to figure out a way to show per device.  I believe the right answer is going to be to use boards.txt. @algernon - thoughts?

technomancy, 2020-02-24:
Thanks for the info. My bad; the Escape thing was based on something you said in a previous issue and it seemed like it worked yesterday but it may have been in the bootloader already from pressing the reset button earlier. I'm used to the bootloader only remaining active for a few seconds on my other keyboards, but it looks like this one stays much longer.

I have updated the udev rules and copied them into `/etc/udev/rules.d` (patch at #814) and I have confirmed that I have write permission to `/dev/ttyACM0`.

I realized I can reproduce the problem much more simply without the makefile or any shell scripts by using `avrdude` directly on the compiled hex file in `/tmp/...`:

```
$ avrdude -p atmega32u4 -c avr109 -U flash:w:Atreus2-0.0.0.hex -P /dev/ttyACM0
avrdude: ser_open(): can't open device "/dev/ttyACM0": Device or resource busy

avrdude done.  Thank you.
```

(Would it be helpful to open a separate issue for the Makefile message as it's not directly related to the failure to flash?)

obra, 2020-02-24:
I've got an in progress local branch to fix the message (and to generalize
some of how we pull out arduino preferences.)

I believe the issue you're running into with that avrdude line is that
before you run it, you need to:

# stty -f /dev/ttyACM0 1200 hupcl

 while holding down the Esc key to get the keyboard booted into bootloader
mode.

I typically run the whole thing as stty ...; sleep 2; avrdude....
ᐧ

technomancy, 2020-02-24:
Sorry--to be clear, I've confirmed (using `dmesg` and the fact that the keyboard is non-responsive to input) that the device is in bootloader mode already. When I run the `stty` command after using the reset button to enter the bootloader, I get `stty: /dev/ttyACM0: Device or resource busy`.

obra, 2020-02-24:
Oh. Um. Are you sure that /dev/ttyACM0 is the -correct- serial port for the
keyboard?

What does the following command give you while the keyboard is in
bootloader mode?

ls -la /dev/serial/by-id/

ᐧ

obra, 2020-02-24:
I've just pushed an update to kaleidoscope-builder that automatically pulls in board-specific flashing instructions and automatically deals with the board already being in bootloader mode. I'd love to know if that resolves this issue for you.

technomancy, 2020-02-24:
Thanks. I've confirmed the port is correct; here's the `dmesg` output when I hit the reset button:

```
[1451115.880122] usb 6-2: new full-speed USB device number 3 using uhci_hcd
[1451116.089581] usb 6-2: New USB device found, idVendor=1209, idProduct=2302
[1451116.089588] usb 6-2: New USB device strings: Mfr=2, Product=1, SerialNumber=3
[1451116.089593] usb 6-2: Product: Atreus Bootloader
[1451116.089597] usb 6-2: Manufacturer: Keyboardio
[1451116.089601] usb 6-2: SerialNumber: atreus
[1451116.095106] cdc_acm 6-2:1.0: ttyACM0: USB ACM device
```

Still getting basically the same results from the latest master:

```
~/src/Kaleidoscope/examples/Devices/Technomancy/Atreus2 $ make flash
BOARD_HARDWARE_PATH="/home/phil/src/Arduino/hardware" /home/phil/src/Arduino/hardware/keyboardio/avr/libraries/Kaleidoscope/bin//kaleidoscope-builder flash
Building ./Atreus2 0.0.0 into /tmp/kaleidoscope-phil/sketch/1992966-Atreus2.ino/output...
- Size: firmware/Atreus2/Atreus2-0.0.0.elf
  - Program:   20958 bytes (73.1% Full)
  - Data:        937 bytes (36.6% Full)

To update your keyboard's firmware, hold down the 'Prog' key on your keyboard,
and then press 'Enter'.

When the 'Prog' key glows red, you can release it.

Looking for USB device with vid=1209 and pid=2303

Examining usb-Keyboardio_Atreus_Bootloader_atreus-if00

  ID_MODEL_ID 2302 != 2303
ERROR: I couldn't find a USB device matching the keyboard's USB Vendor and Device IDs

I couldn't autodetect the keyboard's serial port.

If you see this message and your keyboard is connected to your computer,
it may mean that our serial port detection logic is buggy or incomplete.
In that case, please report this issue at:
	 https://github.com/keyboardio/Kaleidoscope
/home/phil/src/Arduino/hardware/keyboardio/avr/build-tools/makefiles//rules.mk:80: recipe for target 'flash' failed
make: *** [flash] Error 1
~/src/Kaleidoscope/examples/Devices/Technomancy/Atreus2 $ make flash-from-bootloader
BOARD_HARDWARE_PATH="/home/phil/src/Arduino/hardware" /home/phil/src/Arduino/hardware/keyboardio/avr/libraries/Kaleidoscope/bin//kaleidoscope-builder flash-from-bootloader
Building ./Atreus2 0.0.0 into /tmp/kaleidoscope-phil/sketch/1992966-Atreus2.ino/output...
- Size: firmware/Atreus2/Atreus2-0.0.0.elf
  - Program:   20958 bytes (73.1% Full)
  - Data:        937 bytes (36.6% Full)

To update your keyboard's firmware, hold down the 'Prog' key on your keyboard,
and then press 'Enter'.

When the 'Prog' key glows red, you can release it.

WARNING: your udev rules are currently configured to suggest
that your keyboard is suitable for use by ModemManager.  This
means that there is a risk of ModemManager interfering.  To avoid
this, copy

    /home/phil/src/Kaleidoscope/Kaleidoscope-Bundle-Keyboardio/avr/libraries/Kaleidoscope/bin/../etc/99-kaleidoscope.rules

to /etc/udev/rules.d
avrdude: ser_open(): can't open device "/dev/ttyACM0": Input/output error
Connecting to programmer: .
Found programmer: Id = "CATERIN"; type = S
    Software Version = 1.0; No Hardware Version given.
Programmer supports auto addr increment.
Programmer supports buffered memory access with buffersize=128 bytes.

Programmer supports the following devices:
    Device code: 0x44

avrdude: error: programmer did not respond to command: set addr
avrdude: error: programmer did not respond to command: set addr
avrdude: butterfly_recv(): programmer is not responding
avrdude: error: programmer did not respond to command: set addr
```

Not sure why the udev warning is still showing; I've updated the file in `/etc/udev/rules.d`.

obra, 2020-02-24:
The fact that it's mentioning 'Prog' suggests you're not current on
Kaleidoscope or the bundle.  Can you try pulling to current master?

Once you do that, can you try 'make flash VERBOSE=1'?
ᐧ

technomancy, 2020-02-24:
Dang; I keep forgetting there's other copies of Kaleidoscope tucked away places I don't expect. I had my main Kaleidoscope checkout updated but once I updated the various repos in `~/src/Arduino/hardware` I was able to get further. Do you have any recommendations for how to keep all this stuff in sync? Thanks.

The verbose log is long so I've pasted it at https://p.hagelb.org/kaleidoscope-flash-log.html

After the `make flash` completed, dmesg showed `usb 6-2: USB disconnect, device number 14` and repeated removal and reconnecting of the board had no effect. The only way I could get the keyboard to register at all is to use the reset button, which could successfully bring it to the bootloader for a time, but it simply disconnects after leaving the bootloader; it does not reconnect as a keyboard.

At this point, using `make flash-from-bootloader` with the reset button did do the trick, and the keyboard is back to being operational. Several successive attempts with `make flash` failed, however once I turned on `VERBOSE=1` they started succeeding, and now it seems to be able to succeed consistently.

It's working now for me, but I'm not sure what to make of the first flash which rendered the whole board unresponsive. Operator error?

indirect, 2020-02-24:
This is sort of a side-comment, but my personal approach is to symlink my checkout of Kaleidoscope into the place Arduino expects it to be--that has worked well for me in the past.

technomancy, 2020-02-25:
I think the problem was that Arduino needs the whole "kaleidoscope bundle" repo, where Kaleidoscope is just a submodule of that bundle. I'm trying it now with `~/src/Kaleidoscope` being a symlink to the submodule within the bundle in Arduino; we'll see if that's any better. It looks like the bundle changes pretty infrequently.

algernon, 2020-02-25:
> The other half of your issue is that diagnostic message, which we -absolutely- need to figure out a way to show per device. I believe the right answer is going to be to use boards.txt. @algernon - thoughts?

I think the best way forward is to retire `kaleidoscope-builder` in the long run. But your recent changes to make the message keyboard agnostic is a good solution until we can replace the builder (if we decide we really want to, that is).

> It's working now for me, but I'm not sure what to make of the first flash which rendered the whole board unresponsive. Operator error?

I can't reproduce the problem with current master.

obra, 2020-02-25:
> I think the problem was that Arduino needs the whole "kaleidoscope
bundle" repo, where Kaleidoscope is just a submodule of that bundle. I'm
trying it now with ~/src/Kaleidoscope being a symlink to the submodule
within the bundle in Arduino; we'll see if that's any better. It looks like
the bundle changes pretty infrequently.

The bundle changes more frequently when we're prepping new boards, but yeah.

> After the make flash completed, dmesg showed usb 6-2: USB disconnect,
device number 14 and repeated removal and reconnecting of the board had no
effect. The only way I could get the keyboard to register at all is to use
the reset button, which could successfully bring it to the bootloader for a
time, but it simply disconnects after leaving the bootloader; it does not
reconnect as a keyboard.

That sounds like the firmware didn't get flashed correctly, leaving the
keyboard "bricked"  - That could happen if something starts flashing and
then the keyboard is disconnected or the flasher dies.   It's not -good-,
but isn't catastrophic so long as the bootloader exists. If it happens
frequently, we've got a bug in the flashing infrastructure that should be
treated as critical.

> At this point, using make flash-from-bootloader with the reset button did
do the trick, and the keyboard is back to being operational. Several
successive attempts with make flash failed, however once I turned on
VERBOSE=1 they started succeeding, and now it seems to be able to succeed
consistently.
That's super weird, since VERBOSE=1 should only affect output, not what
actually runs.  If this becomes reproducible again, please do flag it.

ᐧ

technomancy, 2020-02-26:
Thanks. I think since I can't reproduce this anymore that we can close this out even if we don't know why it happened; I'll reopen if I can find more.

closed 2020-02-26 by technomancy