keyboardio/Kaleidoscope #816: Evaluate interrupt-time keyscanning again

algernon, 2020-02-24:
So with #815, we moved keyscanning out of the interrupt, because we were missing keystrokes. What I did not explore then, as an alternative fix, is to disable the interrupt during `actOnMatrixScan`. We should try that, because that's a better fix than reverting.

algernon, 2020-02-25:
Hmm... guarding the whole of `actOnMatrixScan()` with a `cli()`/`sei()` pair, and having scanning in the interrupt handler seems to not exhibit the issue found in #814.  That's a huge no-interrupts lock, though, I'd love to be a bit more granular. Guarding the read within the inner loop and the write in the outer, separately, is - as one would expect - not sufficient, because there's a fair chance that an interrupt handler would kick in between the two, and we'd store a different value than what we checked. In practice, that led to a state where if `Fn` got stuck, we couldn't unstuck it.

Guarding within the outer loop seems to result in the same code after optimization, but at least the source looks better.

I've yet to check how this affects debouncing and cycle speeds, and if all goes well, I'll submit a PR reverting #815, and applying the `cli/sei` fix instead.

algernon, 2020-02-25:
I'm also pondering about guarding other places that read or write the key state arrays (such as `pressedKeyswitchCount` and `isKeyswitchPressed`).

We also need to comment up the functions explaining why the guard is there, and also comment up those where we do use them (directly or indirectly), but which do not require a guard (eg, debouncing, which happens during the interrupt).

obra, 2020-02-25:
at a certain point, everything touching those arrays should move into accessors So it’s easy to put guards in the right place.
but if we do it too granular We still run the risk of acting on inconsistent reports

algernon, 2020-02-25:
> at a certain point, everything touching those arrays should move into accessors So it’s easy to put guards in the right place.

It's not that easy, I'm afraid. Just guarding the accessors is _too_ granular. Right now, the stuff that accesses these directly are all nicely contained in the device plugins. Nothing outside of that has direct access. They can only call `pressedKeyswitchCount`, `previousPressedKeyswitchCount`, `isKeyswitchPressed`, `wasKeyswitchPressed`, `readMatrix`, `scanMatrix`, `actOnMatrixScan`.

`actOnMatrixScan`, `pressedKeyswitchCount` and `previousPressedKeyswitchCount` can be guarded within the functions, because they do all the work themselves, they aren't run in a loop for a number of keys. The others... those would need different guards. Most of those are called during `onKeyswitchEvent`, which is called during `actOnMatrixScan`, so is already guarded.

This means that we only need to guard `isKeyswitchPressed`, `wasKeyswitchPressed`, `readMatrix` and `scanMatrix` when they're called outside of `onKeyswitchEvent`. `scanMatrix` is - in most cases - just a wrapper around `actOnMatrixScan` (or that + `readMatrix`), so we don't really need to guard it at all.

Which leaves us with `*KeyswitchPressed`, which is - at the moment - used by two plugins: `HardwareTestMode` and `MagicCombo`. Both seem straightforward to guard.

TL;DR: I do not think accessors would be very useful here, because we already have a very limited set of methods with direct access, and we can't do guarding within the accessors anyway.

algernon, 2020-02-25:
Hmm... guarding `*KeyswitchCount` within the functions is... maybe a bad idea. In `MagicCombo`, for example, we'd want to have interrupts disabled for _both_ counting pressed keyswitches, and checking individual state. If we guard in `pressedKeyswitchCount`, then there's a small window between that and the loop starting where interrupts are enabled, so we might end up with out-of-sync data.

Thus, we need to guard these on the caller side. Thankfully, there's only one plugin that uses `*KeyswitchCount()`: `MagicCombo`.

obra, 2020-02-25:
Continuing to think about this, the amount of guarding we need to do
continues to lead me to want to hang out with our current implementation
for a while, even though I was the one who first pushed hard to move the
scan into the interrupt. The change leads to a bunch of possible
implementation problems if folks don't -know- they need to guard accesses.
ᐧ

commit 059d4723 (2020-02-25): Guard some places against interrupts

When checking or updating the key state, we need to do so with interrupts
disabled for the _whole_ duration of our operation, otherwise we may end up in
an inconsistent state. This means that we disable interrupts for the whole of
`actOnMatrixScan` (for hardware that uses interrupts for key scanning, anyway),
and whenever we use `pressedKeyswitchCount`, `previousPressedKeyswitchCount`,
`isKeyswitchPressed`, or `wasKeyswitchPressed`. These latter ones can't be
guarded within the functions, because we need to guard not just a single call to
them, but all of  [... cut]