keyboardio/Kaleidoscope #820: Ergodox ez prog button

Kethku, 2020-03-04:
in the qmk firmware I think I saw that there was a way to bind a key which would set the keyboard into a mode which would accept a new firmware. On my model 01 this was hardwired to the prog key, but I'm wondering if there is an equivalent way to configure the ergodox

obra, 2020-03-04:
Kaleidoscope doesn't need the button - it uses the serial port to reboot to
the bootloader. On the Model 01, what we have is a safety lockout inside
the bootloader: If that key isn't held the bootloader won't let you flash
the keyboard.
ᐧ

Kethku, 2020-03-04:
Weird. Only way I've been able to upload a firmware is by pressing a paperclip button in the top right hand side of the keyboard. This lets the teensy uploader to actually send the hex file over.

obra, 2020-03-04:
Oh. I forgot that it's running the Teensy's bootloader.

@algernon may have more input here
ᐧ

algernon, 2020-03-04:
Hmm... I thought we had a macro in the ErgoDox EZ firmware shipped with Chrysalis that'd reset to bootloader mode, but we don't. I'll fix that soon!

Meanwhile, if you compile your own firmware, and add a macro that calls `Kaleidoscope.device().rebootBootloader()` on activation, and place that on a key somewhere (any layer will work), then you can activate that key to enter programmable mode, without having to use a paperclip.

Kethku, 2020-03-04:
Very cool! Thanks so much!

Kethku, 2020-03-05:
Quick question: I can't seem to get that line to compile. Is there something specific I need to import at the top of my ino file? It fails with "kaleidoscope::Kaleidoscope_ has no member named device".

algernon, 2020-03-05:
Oh, you might be running an older version Kaleidoscope then. Try `ErgoDox.resetDevice()` instead.

Kethku, 2020-07-07:
I've updated my sources and attempted the new line, but it seems like that doesn't trigger the teensy loader to install new firmware like the paperclip button does. Do you have any other ideas for how to emulate such a system? I know its possible with the qmk firmware, but I would really rather not swap to that given that I have a number of custom plugins written for kaleidoscope that I'd miss dearly.