keyboardio/Kaleidoscope #825: Nintendo Switch compatibility regression

indirect, 2020-03-27:
The Nintendo Switch dock has 3 USB type A ports, and connecting any keyboard to those ports allows you to provide typed input. The easiest way to test out a keyboard is:

1. Take a screenshot of your home screen using the ⏺ screenshot joycon button
2. Open the Albums app in the row of circular system apps along the bottom
3. Select the picture by pressing "A", then "A" again to choose "Editing and Posting"
4. Navigate down to "Add Text", and press "A".
5. Type some text!

You can leave your Switch on that screen while connecting and disconnecting keyboards, and that's the main way that I was able to bisect the issue I found.

Now the issue: I've tried using my Model01 and Atreus2 with my Switch, and discovered that older firmwares work normally and as expected, but newer firmwares... don't work as well. The symptoms I observed are space/backspace getting stuck and repeating after being typed, individual letters not being able to be typed more than once, and no further output after typing 5-10 letters.

A `git bisect` on the Kaleidoscope-Bundle-Keyboardio repository found a specific commit that caused the regression: https://github.com/keyboardio/Kaleidoscope-Bundle-Keyboardio/commit/340de890e80ba52684405b7827a9853eec5800c9.

That seems to track, since the issues with the Switch seem to be HID issues. Happy to provide additional testing or bisecting if that would help narrow it down!

indirect, 2020-03-29:
Today, I managed to get my Atreus2 working with my Switch via some silly git machinations:

1. Check out https://github.com/keyboardio/Kaleidoscope-Bundle-Keyboardio/commit/55065cec598906c3ef336e6cf5e28e0118c4c63f, which is the last commit before the regression.
2. Update all submodules.
3. Go and update _just_ the `avr/libraries/Kaleidoscope` submodule to the latest master (which is currently https://github.com/keyboardio/Kaleidoscope/commit/919d8736d5be80aff5f4c2dbeb404d33f223d7b3), to get the flashing tools needed for A2.
4. Go back to my personal A2 firmware at https://github.com/indirect/Atreus2-Firmware/commit/920eaba43621cdb66e95eba0034132e0d7cb7475, and run my `bin/flash` wrapper around `make flash`.
5. A2 now works perfectly on Switch!

indirect, 2020-03-30:
...and of course, today, I cannot get my A2 working with my Switch even by following my own instructions. 🤔🤨

[30 minutes of frantic testing]

Aha! I found a super important missing step: you have to _build_ the firmware with the _old_ Kaleidoscope, but then _install_ it with the _new_ Kaleidoscope (which can skip the build using `maybe_build`, and has the tooling to talk to an Atreus 2). Phew. I'll update my steps.

nevd, 2020-04-26:
thanks @indirect these instructions also work users of OSX+filevault where you cannot stop the key repeat either

obra, 2020-04-27:
@algernon provided a bit of insight:

"we have multiple HID symbols
that... confuses things" It's documented a bit in https://github.com/keyboardio/KeyboardioHID/issues/59 

The solution that needs to happen is to remove duplicated HID constructors, but he's not currently sure -why- they're getting duped.

indirect, 2020-04-27:
Thanks for the update! Looking forward to a less wonky HID situation. 🙂

obra, 2020-04-27:
I believe this is fixed by
https://github.com/keyboardio/Kaleidoscope/pull/833

closed 2020-04-27 by obra

indirect, 2020-04-28:
I can confirm this is fixed by #833. Atreus2 + Switch working great now. 🎉

commit 37a3d44f (2020-04-27): Boot Keyboard keys were never being released.
Fixes #825
PR #833 (2020-04-27): Boot Keyboard keys were never being released.
Fixes #825

Due to a presumed refactoring bug, keys on boot protocol keyboards were never being released. This can't possibly work
commit 9caad1d8 (2020-04-27): Boot Keyboard keys were never being released.
Fixes #825