keyboardio/Kaleidoscope #830: "Modifier/Layer shift when held" feature of Chrysalis results in non-functioning keys

numist, 2020-04-17:
Running 03bee4b874 on a Model 01, tried to use Chrysalis 0.7.4 to configure `pgup` to be `shift` when held and `pgdn` to layer shift to `num` when held; the actual outcome is that neither key does anything until I turn the "Modifier/Layer shift when held" feature back off.

gedankenexperimenter, 2020-04-24:
Is your firmware compiled with the Qukeys plugin? Without that, those keys won't work. Alternatively, Qukeys might be disabled.

gedankenexperimenter, 2020-04-24:
Sorry for not reading more carefully before replying; I glossed over the link you provided to the sketch. Indeed, it doesn't include Qukeys, which is why those keys don't work.

numist, 2020-05-12:
Yep, that was it.

Seems like that should be built into the default firmware if its functionality is going to be showcased in Chrysalis' interface?

algernon, 2020-05-12:
Chrysalis can't - currently - auto-detect which plugins are enabled on the firmware side, so it shows pretty much all possibilities. Until we can detect the plugins used by the firmware, I believe this is still the better option (the alternative being not showing this - and other plugin-provided - features at all).

obra, 2020-05-13:
perhaps it’s time to make the default firmware  the one with the features.

gedankenexperimenter, 2021-06-09:
@obra — I think this issue can be closed now.  It certainly isn't an actual Kaleidoscope issue, at least.

closed 2022-06-08 by algernon