keyboardio/Kaleidoscope #832: Qukeys - LT does not work with  Key_LEDEffectNext

sboesebeck, 2020-04-25:
When trying to use qukeys in with ` Key_LEDEffectNext`, the key does not work anymore. No layer toggle, nor the LEDEffect is changed.

gedankenexperimenter, 2020-04-27:
It is possible to get the behaviour you want with Qukeys, but not with the `LT()` macro. That macro only works with normal keyboard (8-bit) keycodes. `Key_LEDEffectNext` has a different value of the `flags` byte than a Qukeys `LT()` key does, so it's just not possible to combine the two.

What you need to do is define a full Qukey object. Keep `Key_LEDEffectNext` in your keymap, and add the following to your sketch:
```
QUKEYS(
    kaleidoscope::plugin::Qukey(0, KeyAddr(0, 6), ShiftToLayer(FUNCTION))
)
```
For more information, see the [Qukeys documentation](https://github.com/keyboardio/Kaleidoscope/blob/master/docs/plugins/Qukeys.md)

sboesebeck, 2020-04-27:
Thanks, that worked fine!

strange still, that the plugin does not work - is that not possible to 
fix?

closed 2020-04-27 by obra