keyboardio/Kaleidoscope #840: Possible rollover race condition in SystemControl keys

obra, 2020-05-04:
@gedankenexperimenter noticed a possible race condition issue with the SystemControl fixes in  #834 

Looking at the code, I believe there's still a bug. If the user rolls over from one system control key to another, the second one will get released as soon as the first one gets released.

Key A toggles on -> A report sent
Key B toggles on -> B report sent, clobbering A (correctly)
Key A toggles off -> empty report sent (incorrectly indicating that B was released)
Key B toggles off -> another empty report sent
When a system control key is released, it shouldn't unconditionally send the empty report. Instead, it should check to verify that the released key is the same as the last one that was sent. To do this right, it's necessary to store that state, probably in KeyboardioHID.

gedankenexperimenter, 2020-09-03:
I believe the best way to solve this is by storing the `KeyAddr` of the system control key that was last pressed. Then, on release, we can compare that to the `KeyAddr` of the one that toggled off, and only send the release (empty report) if the same key was released. This still doesn't solve the problem in every possible case, because of injected keys, but it's pretty close. The trouble here is that we would need to add some ugly code to `key_events.cpp` (probably passing an additional `KeyAddr` parameter to `handleSyntheticKeyswitchEvent()` and using a static variable inside that function), or making a change to the HID interface.

The other option is to make a change to KeyboardioHID which would also require altering the public interface, because the `release()` method doesn't take any parameters, and it would need to be able to compare the released key's keycode to the last one pressed.

gedankenexperimenter, 2020-09-03:
It looks like I could make a change to `kaleidoscope::driver::hid::base::Keyboard` that wouldn't alter the signature of any functions. It looks like that's the best option. It adds 14 bytes of program memory and one byte of RAM, and fixes the problem except when rolling over from one System Control key to a second one that has the same value, which is a case that is extraordinarily unlikely to ever occur in the wild.

obra, 2020-09-03:
It occurs to me that since the HID System keys that are supported by hosts
are all(?) OSC one-shot HID controls, there may not be a real rollover
issue in practice. But perhaps I'm missing docs to the contrary?

gedankenexperimenter, 2020-09-03:
In at least one case we've seen (#834), an OS required the System Control key to be held for a minimum duration, or it would be ignored. It's not likely to come up in practice, and any user who does encounter this particular failure is likely to just try a second time and succeed (because they probably won't roll over again). I just like to make the behaviour as correct and standards-compliant as possible.

I've made the changes that should fix this issue, but haven't had time to test them out yet, so I haven't submitted a PR for it. If you've changed your mind about wanting this to be fixed, or think that a fix isn't worth the extra 14 bytes of PROGMEM and 1 byte of RAM, feel free to close the issue, and I won't bother. The cost isn't very high, but the problem is, as we've said before exceedingly unlikely to occur, so while I would choose to fix it, I don't think it's unreasonable to take the other position.

obra, 2020-09-03:
Correct and standards compliant is good.
That amount of overhead is not necessarily a deal breaker.

I had indeed forgotten the issue that spawned this ticket. Thanks for the reminder on that front.

In theory, I don't think that rollover between two keys mapped to the same OSC functionality -really- needs to be treated as a release and a new press.

What's the shape of your fix? (It's totally ok to make draft PRs from untested changes - that's often a great way to start discussion of an actual implementation.)

gedankenexperimenter, 2020-09-03:
Clarification: the System Control HID report data consists of a single keycode byte. When a key is pressed, a report is sent, and when it's released, an "empty" report is sent. If the user rolls over from one key to another, there's no "release" report sent; there's just a new report with the new keycode, which the OS should interpret as a release and press of a new key. That's not the problem; it's that the release of the first key causes the empty report to get sent, even though the second key is still being held.

I'm pretty confident that my fix is correct (with the ridiculous exception of two of the same System Control key in one keymap involved in rollover). I just need to find the time to go down to my basement, make a keymap, and test it. That won't take long, but I do have three children…

obra, 2020-09-03:
> Clarification: the System Control HID report data consists of a single
> keycode byte. When a key is pressed, a report is sent, and when it's
> released, an "empty" report is sent. If the user rolls [...]

That makes sense. And it sounds like a solution similar to the ChromeOS
workaround we have in the NKRO Keyboard driver would work.

"If the key being sent does not match with the previous report and the
previous report is not empty, force an empty report before sending the new
one"

This -would- require us to start tracking the previous SystemControl report
inside KeyboardioHID::SystemControl, which doesn't seem wrong. Based on the
code and data size you describe, I bet that's what you're doing ;)

> I'm pretty confident that my fix is correct (with the ridiculous exception
> of two of the same System Control key in one keymap involved in rollover).
> I just need to find the time to go down to my basement, make a keymap, and
> test it. That won't take long, but I do have three children…

Totally get it. I've only got the one. But still.

With luck, we'll soon have the testing framework in and be able to actually
verify that this stuff works the way we want it to a bit easier. (But this
doesn't need to block on that.)

gedankenexperimenter, 2020-09-03:
> This -would- require us to start tracking the previous SystemControl report inside KeyboardioHID::SystemControl, which doesn't seem wrong. Based on the code and data size you describe, I bet that's what you're doing ;)

That would be better than what I'm doing (arguably), but it would require a change to the KeyboardioHID API, because the `release()` method doesn't have any parameters, so there's no way to compare the released keycode to the one in the previous report. It just now occurred to me that we could add a new `release(keycode)` method, and deprecate the old one, rather than tracking it in `kaleidoscope::driver::hid::base::Keyboard`, which is what I've done. Maybe I'll try that, too, and see if it's a smaller increase in PROGMEM.

obra, 2020-09-03:
Is the failure case you're thinking of:

Cycle 1:  SysCtlKeyA pressed
 -> SystemControl.press(A);

Cycle 2: SysCtlKeyA released + SysCtlKeyB pressed
-> SystemControl.press(B);
->SystemControl.release();

Cycle 3: SysCtlKeyB released
SystemControl.release();

With the main issue coming in because the "press" of B is sent down before
the "release" of A is sent, which results in no "release" of A -and- a
"stuttering" press of B if the two events during Cycle 2 happen in a "bad"
order?

(Have I mentioned that I can not wait for the ability to write a test for
this.)

Conveniently, releaseSystemControl already expects to be passed the key
being released, so extending the API for SystemControl to know what key's
being released seems plausible.

It still gets a little messy to be massaging things in cycle 2 for the
"right" thing to happen, but it's not totally implausible.

gedankenexperimenter, 2020-09-03:
I don't think there's any issue with the lack of a "release" report for <kbd>Sysctl_A</kbd>. The problem is that there are four events:

1. <kbd>Sysctl_A</kbd> press: report `A` sent, no problem
2. <kbd>Sysctl_B</kbd> press: report `B` sent, OS interprets as a release of `A` & press of `B`, no problem
3. <kbd>Sysctl_A</kbd> release: null report sent, OS interprets as a release of `B`, even though it's still being held
4. <kbd>Sysctl_B</kbd> release: redundant null report sent, probably not a problem

The problem is in event 3, where there shouldn't be a report sent at all, but instead we're signalling that `B` was released (when it was `A` that was released, in fact).

`releaseSystemControl()` does get passed the `Key` value, but that doesn't propagate all the way down the chain to KeyboardioHID. I just tried that out, and it costs an extra 20 bytes of PROGMEM (+34 vs +14).

PR #892 (2020-09-15): Fix System Control rollover bug
Previously, rolling over from one System Control key to another would cause the second one to be released as soon as the first one was released, because the empty release report would be sent unconditionally on release of any System Control key.

This change stores the value of the last System Control key pressed. When a System Control key is released, it first checks to see that the released key's keycode matches the last one pressed before sending the empty report.

Fixes #840
closed 2020-09-15 by obra