keyboardio/Kaleidoscope #845: Map Shit+LED to previous LED effect

xerus2000, 2020-05-15:
As fas as I'm aware, Shift+LED currently does nothing, and I don't see a way to map it from Kaleidoscope. Couldn't this be mapped to go to the previous LED effect by default?

obra, 2020-05-15:
That shouldn’t it be possible to do

alfalfasprossen, 2020-06-03:
Technically `LEDControl` could check the state of the shift modifiers when interpreting the `Key_LEDEffectNext` / `Key_LEDEffectPrevious` key press. But that may not work reliably when plugins are involved that handle shift modifiers like `OneShot` - and having LEDControl access those plugins is unlikely to be manageable or desirable.

You can achieve this in your sketch using a Macro though. If you use OneShot, you may need to check the state of shift with it as well, but I didn't try it.

```c++
#include "Kaleidoscope.h"
#include "Kaleidoscope-Macros.h"
#include "Kaleidoscope-LEDControl.h"
#include "Kaleidoscope-LEDEffect-SolidColor.h"

enum { MACRO_NEXT_PREV_LED_MODE };

// *INDENT-OFF*
KEYMAPS(
  [0] = KEYMAP_STACKED
  (Key_NoKey, Key_1, Key_2, Key_3, Key_4, Key_5, M(MACRO_NEXT_PREV_LED_MODE),
   Key_Backtick,      Key_Q, Key_W, Key_E, Key_R, Key_T, Key_Tab,
   Key_PageUp,        Key_A, Key_S, Key_D, Key_F, Key_G,
   Key_PageDown,      Key_Z, Key_X, Key_C, Key_V, Key_B, Key_Escape,

   Key_LeftControl, Key_Backspace, Key_LeftGui, Key_LeftShift,
   XXX,

   Key_skip,  Key_6, Key_7, Key_8,     Key_9,      Key_0,         Key_skip,
   Key_Enter, Key_Y, Key_U, Key_I,     Key_O,      Key_P,         Key_Equals,
              Key_H, Key_J, Key_K,     Key_L,      Key_Semicolon, Key_Quote,
   Key_skip,  Key_N, Key_M, Key_Comma, Key_Period, Key_Slash,     Key_Minus,

   Key_RightShift, Key_RightAlt, Key_Spacebar, Key_RightControl,
   XXX),
)
// *INDENT-ON*

static void nextPrevLEDModeMacro(uint8_t keyState) {
  if (keyToggledOn(keyState)) {
    if (kaleidoscope::Runtime.hid().keyboard().isModifierKeyActive(Key_LeftShift) ||
        kaleidoscope::Runtime.hid().keyboard().isModifierKeyActive(Key_RightShift))
    {
      LEDControl.prev_mode();
    } else {
      LEDControl.next_mode();
    }
  }
}

const macro_t *macroAction(uint8_t macroIndex, uint8_t keyState) {
  switch (macroIndex) {

  case MACRO_NEXT_PREV_LED_MODE:
    nextPrevLEDModeMacro(keyState);
    break;

  }
  return MACRO_NONE;
}

static kaleidoscope::plugin::LEDSolidColor solidYellow(130, 100, 0);
static kaleidoscope::plugin::LEDSolidColor solidGreen(0, 160, 0);
static kaleidoscope::plugin::LEDSolidColor solidViolet(130, 0, 120);

KALEIDOSCOPE_INIT_PLUGINS(
  LEDControl,
  solidYellow, solidGreen, solidViolet,
  Macros
);

void setup() {
  Kaleidoscope.setup();
}

void loop() {
  Kaleidoscope.loop();
}

```

obra, 2020-06-03:
It looks like I blew it on my response. I was either going for “That shouldn’t be too hard” or “That should be possible”

I think that adding functionality inside the plugin to check shift state and go in reverse would be a lovely feature to have. I’d be thrilled to look at a PR.

gedankenexperimenter, 2020-06-04:
You might want to consider fixing the typo in the title of this issue…

obra, 2020-06-04:
Hadn’t even noticed. fixed

xerus2000, 2020-06-04:
lol same, I had to look really hard to find that...
Also @obra yes that response also confused me ^^

@alfalfasprossen are you saying this isn't reliably possible without custom firmware? I thought there could be some way to do translate this directly on the keyboard rather than checking the resulting keypress.

alfalfasprossen, 2020-06-08:
@xerus2000 Not sure what you mean - you have to check the resulting keypress, either via a macro in your own sketch or at the respective event handlers in the LEDControl plugin.

What I mean with "not reliable" is that there are ways to produce a shift state that some keys are not compatible with - and the LEDControl keys are of such nature. They must match a specific bit-pattern to be recognized as the keys they are and setting the bit for `SHIFT_HELD` will produce an "invalid" key. -- But it may just be OK to ignore such cases. Like who would use `LSHIFT(Key_LEDEffectNext)` 😄  The expected default behaviour - using a normal shift key - should work. The results of using plugins like OneShot though depend on the order they are initialized. Using OneShot before Macros for example, will cause shift to correctly be registered when interpreting a Macro, while initializing it after Macros will not. For such cases OneShot has its own function to check if a OneShot modifier is supposed to be active - and handling such cases in a different plugin like LEDControl would create undesirable dependencies and convoluted code.

That being said, I experimented with checking the shift modifier state in the LEDControl plugin and it works for the normal cases (plain shift and OneShot initialized before LEDControl) - I will create a PR for it 😉

gedankenexperimenter, 2020-09-03:
I think this issue can be closed now that #849 has been merged.

obra, 2020-09-03:
Thanks, @gedankenexperimenter. Closed.

closed 2020-09-03 by obra