keyboardio/Kaleidoscope #857: Layers, transparency, ordering

algernon, 2020-06-18:
I have seen a number of people getting confused about how our layer system works - it can, indeed, be quite confusing at times. So in this issue, I attempt to both describe how it works now, why it might be confusing, and how we could improve it.

Our current system has layers numbered from 0 to 31, they can stack, and the higher the index of the layer, the more prominent it is. In other words, a higher numbered layer will always shadow lower layers. Active layers, that is. If we have layer 0 and 2 active, 2 will shadow 0, but since layer 1 is inactive, it has no effect. What happens under the hood, is that we look at the keycode on the topmost active layer, if it is non-transparent, we use it. Otherwise we find the next lower-indexed, but still active layer, and do the same check: if non-transparent, use it, otherwise walk downwards. If we reach the lowest active layer, and still didn't find a non-transparent key, we'll treat it as no action. We do this for every key we need to resolve.

Of course, there are a number of tricks involved to make this performant: whenever layer state changes, we build a cache of all keys on the layer, fully resolving each. So we only do this once, not every time we press a key within the same state. Another trick we do is that if we have a key _held_ while layers change, we keep that key emitting the same code it was emitting before the layer change, until released. This is to avoid surprising behaviour and timing mistakes when a key release is registered after a layer change, even if it was released sooner.

The confusion mostly stems from a combination of transparency and index-based ordering. The tricks we use don't really play a role here, so we'll focus on transparency and ordering.

One confusing thing about transparency is that many people seem to think of layers as distinct, unrelated layers. As in, the expectation is that when you activate one, that becomes the definitive layer, and no other layers are in play. Many expect it to be a switch, not something that's layered on top of others. With this thinking, transparency makes little sense, and one may even arrive to the conclusion that the firmware may look at any layer below (even inactive ones), and that makes even less sense.

So, to reiterate: layers are layered on top of each other. There can be any number of layers active at the same time. The current implementation will walk from top to bottom, and find the first non-transparent key, when it needs to resolve what keycode a given key corresponds to.

Another part of the confusion is the ordering. It seems logical that the last layer activated would have the highest priority - but that is not the case in our current implementation. The highest indexed layer is the highest priority, even if that was activated last. This may seem surprising to people new to the system.

To avoid this confusion, we could have a system where - as one would expect - activation order matters. If we're on layer 0, and activate layer 2, then layer 1, then keys would be looked up in 1->2->0 order instead of 2->1->0 like in the current index-based ordering scheme. If we release layer 2, then we'd look up 1->0. If we release layer 1 first, then 2->0.

Now, my question is this: should we implement an option to use activation-order instead of index-order? If so, which should be the default? And do we want to keep index-order to begin with?

algernon, 2020-06-18:
As far as implementation goes, we have a number of plugins that assume index-based ordering, those are reasonably easy to fix to work with either system: just introduce `Layer.getEffectiveLayer()` which returns either the highest or the most recently activated layer. We could even bake this into `Layer.top()`, but I think that name doesn't fit activation-order all too well.

Other than this, most plugins that deal with layers shouldn't need to know which ordering we use. They just want to know the effective layer, and they want to be able to resolve a key. Some may want to walk all the active layers in order, and we can provide an iterator for that, too. All of this can be neatly encapsulated in `Layer`, with a small knob to switch between ordering methods (either at run-time, or at compile time). I do think there's merit in allowing run-time switching: so users can try and experiment with either, without compiling new firmware.

Activation-order does have one drawback however: it requires a bit more resources. We can't juse use a 32-bit bitfield to keep track of active layers, because that's not enough to store ordering. We'll need a list of 32 bytes instead. But that makes the bitfield redundant, so we should drop that to save some space. We need to drop `Layer.getLayerState()` too then, but that's not very widely used, and we can do a rough compatibility shim too: walk the list, turn on the respective bits, done. It wouldn't be able to keep the order, but it would allow to have a quick look at the active layers, regardless of their order. Mind you, an iterator sounds like a better option in this case, but that'd be a new API, not a compatibility shim.

algernon, 2020-06-18:
One more thing: layer keys. We currently have the following layer keys available:

- `ShiftToLayer(n)`: Shifts to layer `n`, while the key is held. When the layer key is released, the layer deactivates.
- `LockLayer(n)`/`UnlockLayer(n)`: Toggles layer `n` on and off. The layer remains active until unlocked - which is the same keycode as lock, due to technicalities - another possible source of confusion!
- `MoveToLayer(n)`: Moves to layer `n`, activating it, and deactivating _all_ other layers.
- `OSL(n)`: OneShot to layer `n`, activating it for the next keypress, or until held. Or if double-tapped, until `OSL` is deactivated by a third tap.
- `LT(n, key)`: DualUse key, activating layer `n` while held, like `ShiftToLayer(n)`, but emitting `key` if simply tapped.

The interactions between all these, and transparency, can lead to very confusing situations, too.

obra, 2020-06-19:
@algernon isn't kidding when he says things can be confusing. 

-I- misunderstood our layer ordering. On the Model 01, we don't have enough space for layers and the default layers are ordered such that this would rarely come up, but our current behavior isn't friendly or expected for most users. 

I think transparent keymaps can be a valuable feature and that a lot of the confusion around them could probably be alleviated with better UI in Chrysalis. Our internal implementation would certainly be easier if we got rid of transparency, but I'm not excited to do that.

In a future comment, I'll talk about what I think we should be doing for layer ordering.

obra, 2020-06-19:
In terms of layer ordering: my -suspicion- is that almost nobody is depending on index ordering rather than activation ordering. The only "desirable" use case I can think of is to change what happens with transparent keys when toggling on an alternate intermediate layer. And when I say "desirable", those quotation marks should be read to indicate that I'm skeptical.

I feel like index ordering was an expedient choice for a simple implementation that matched what other firmware did at the time, but that it's not actually a good choice.

@algernon - You've previously mentioned you had a personal use case that was easier with index ordering. I'd love to know more about that.

I've been pondering this for the past day and my belief is that Kaleidoscope should, by default, use activation ordering for layers. I don't think we get a lot of value out of also preserving index ordering.

> As far as implementation goes, we have a number of plugins that assume index-based ordering, those are reasonably easy to fix to work with either system: just introduce Layer.getEffectiveLayer() which returns either the highest or the most recently activated layer. 

It depends what they're trying to find out.  I'd -probably- call it getActiveLayer() if they're trying to find out what the more recently activated layer is.

> We could even bake this into Layer.top(), but I think that name doesn't fit activation-order all too well.

Layer.top() barely gets called. deprecation is not necessarily wrong.

>  I do think there's merit in allowing run-time switching [between ordering systems]: so users can try and experiment with either, without compiling new firmware.

Without some concrete examples of when index ordering gets us something we can't do with activation ordering, I don't think this flexibility feels like a useful tradeoff. We don't have a ton of system resources.

I don't -think- we need to replace our layer activation/deactivation primitives (layer keys) at the same time as making the activation ordering change, though I bet we would benefit from a clean and clear set of names. (LT and OSL don't fit nicely with the other names and I know that some folks have found the lock and shift names confusing.)

algernon, 2020-06-19:
> @algernon - You've previously mentioned you had a personal use case that was easier with index ordering. I'd love to know more about that.

I no longer have that use-case, I'm afraid.

I used to have a very complex, 6-layer setup on my KBD4x. I often set up scenarios there where I had 2-4 layers active at a time, overriding parts of the base layer. The layers were designed to be used in index-order, doing the same thing with activation order would have required two more layers, as far as I recall. I don't remember the details off the top of my head, sadly. But like I said, this was a very complex case, and noone in their right mind would ever set this up this way. I eventually redesigned the whole thing, and now it can work with activation order too (it's only 4 layers now, too).

I _can_ dig up the previous setup, but at this point, I do not think it would be very useful, because it's a very, very far edge case, not something we should actively support, really.

gedankenexperimenter, 2020-06-24:
I think most of the confusion new users experience would be alleviated by renaming the existing macros (`ShiftToLayer()` & (`Un`)`LockLayer()`) to better reflect their actual behaviour, and introducing new layer `Key` values that do the things that they expect. I tried to do this a while back, with a "shift *to* layer" behaviour that effectively deactivates higher layers while the key is held.

On another note, I can think of a use case where index order helps. If the user simultaneously presses two temporary ("shift") layer keys, one of which is partially transparent, there would be a race condition determining the effective keymap. I think this would be a good reason to keep index ordering, unless you only want to allow one layer-shift at a time.

gedankenexperimenter, 2020-09-07:
> * `LockLayer(n)`/`UnlockLayer(n)`: Toggles layer `n` on and off. The layer remains active until unlocked - which is the same keycode as lock, due to technicalities - another possible source of confusion!

If the switch to activation order happens, I think it would make more sense to have `LockLayer(N)` (unless there's a better name for it) have a slightly different behaviour: instead of simply toggling layer `N`, move it to the top of the active layer stack, unless it's already the top active layer, in which case, it should get deactivated instead.

My reasoning is based on the supposition that complete or nearly-complete layers are the norm, in which case a layer that is not at the top of the stack is unlikely to be visible to the user. If I can't tell whether or not a layer is active, then pressing a key to toggle it might or might not have the effect I desire. I do think it's fair to assume that the user knows which layer is on top, however.

PR #867 (2020-10-10): Switch to activation-order for Layers
Previously, we used index-ordering for layers, meaning, we looked keys up based on the index of active layers. This turned out to be confusing, and in many cases, limiting, since we couldn't easily shift to a lower layer from a higher one. As such, index-ordering required careful planning of one's layers, and a deeper understanding of the system.

This patch switches us to activation-ordering: the layer subsystem now keeps track of the order in which layers are activated, and uses that order to look keys up, instead of the index of layers. This makes it ea [... cut]
commit ab3b661c (2020-10-10): Switch to activation-order for Layers

Previously, we used index-ordering for layers, meaning, we looked keys up based
on the index of active layers. This turned out to be confusing, and in many
cases, limiting, since we couldn't easily shift to a lower layer from a higher
one. As such, index-ordering required careful planning of one's layers, and a
deeper understanding of the system.

This patch switches us to activation-ordering: the layer subsystem now keeps
track of the order in which layers are activated, and uses that order to look
keys up, instead of the index of layers. This makes it e [... cut]
closed 2020-10-10 by obra